espressif/esp-idf #5783: TASK Problem with Memory ESP32 - Unavaliable Memory - Corruption

filzek, 2020-08-24:
Lastest Master as of 8/24/2020 has a serious problem with memory, it is using a lot of memory and we cant use with tasks.

please advise that is a serious faulty

xPortGetFreeHeapSize: 4244228 before_count:12 before_heap_32:4260724 before_heap_8:4246532

Heap summary for capabilities 0x00000004:
  At 0x3ffc8de4 len 32767 free 2304 allocated 30408 min_free 2304
    largest_free_block 2304 alloc_blocks 4 free_blocks 1 total_blocks 5
  At 0x3f8037b0 len 4180047 free 4122000 allocated 57808 min_free 4092260
    largest_free_block 4121984 alloc_blocks 49 free_blocks 2 total_blocks 51
  At 0x3ffaff10 len 240 free 0 allocated 180 min_free 0
    largest_free_block 0 alloc_blocks 7 free_blocks 0 total_blocks 7
  At 0x3ffb6388 len 7288 free 0 allocated 6968 min_free 0
    largest_free_block 0 alloc_blocks 72 free_blocks 0 total_blocks 72
  At 0x3ffb9a20 len 16648 free 4 allocated 15556 min_free 0
    largest_free_block 4 alloc_blocks 263 free_blocks 1 total_blocks 264
  At 0x3ffc8dc8 len 94776 free 0 allocated 94336 min_free 0
    largest_free_block 0 alloc_blocks 102 free_blocks 0 total_blocks 102
  At 0x3ffe0440 len 15072 free 8420 allocated 5972 min_free 8420
    largest_free_block 8420 alloc_blocks 161 free_blocks 1 total_blocks 162
  At 0x3ffe4350 len 113840 free 113804 allocated 0 min_free 88200
    largest_free_block 113804 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 4246532 allocated 211228 min_free 4191184 largest_free_block 4121984

Heap summary for capabilities 0x00000002:
  At 0x3ffc8de4 len 32767 free 2304 allocated 30408 min_free 2304
    largest_free_block 2304 alloc_blocks 4 free_blocks 1 total_blocks 5
  At 0x3f8037b0 len 4180047 free 4122000 allocated 57808 min_free 4092260
    largest_free_block 4121984 alloc_blocks 49 free_blocks 2 total_blocks 51
  At 0x3ffaff10 len 240 free 0 allocated 180 min_free 0
    largest_free_block 0 alloc_blocks 7 free_blocks 0 total_blocks 7
  At 0x3ffb6388 len 7288 free 0 allocated 6968 min_free 0
    largest_free_block 0 alloc_blocks 72 free_blocks 0 total_blocks 72
  At 0x3ffb9a20 len 16648 free 4 allocated 15556 min_free 0
    largest_free_block 4 alloc_blocks 263 free_blocks 1 total_blocks 264
  At 0x3ffc8dc8 len 94776 free 0 allocated 94336 min_free 0
    largest_free_block 0 alloc_blocks 102 free_blocks 0 total_blocks 102
  At 0x3ffe0440 len 15072 free 8420 allocated 5972 min_free 8420
    largest_free_block 8420 alloc_blocks 161 free_blocks 1 total_blocks 162
  At 0x3ffe4350 len 113840 free 113804 allocated 0 min_free 88200
    largest_free_block 113804 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x4009c86c len 14228 free 14192 allocated 0 min_free 14184
    largest_free_block 14192 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 4260724 allocated 211228 min_free 4205368 largest_free_block 4121984
Task Name       Status  Prio    HWM     Task#   Affinity
Sentinel       R       15      5216    15      0
IDLE1           R       0       984     7       1
IDLE0           R       0       988     6       0
tiT             B       18      2596    12      -1
MultiCast   B       10      18524   21      -1
DigitalTRX B       5       3584    22      -1
Tmr Svc         B       1       1564    8       0
ipc1            B       24      416     2       1
sys_evt         B       20      1568    13      0
wifi            B       23      1868    14      0
esp_timer       B       22      3396    3       0
ipc0            B       24      492     1       0

so when trying to create any other task now the system just halt.

Please advise;

chegewara, 2020-08-25:
Add caps internal to functions checking free heap. Task stack cant be allocated in spiram.

projectgus, 2020-08-25:
Hi @filzek,

As Chegewara mentions, task stacks have to be allocated in internal memory. The heap dumps you have posted are for all 8-bit capable memory (0x04) and all 32-bit capable memory (0x02). Both these include PSRAM so there is a lot of memory free which is not available for use as a task stack.

Can you please try generate the same output for `MALLOC_CAP_8BIT|MALLOC_CAP_INTERNAL`? Also, suggest running `idf.py size` or `make size` to see a summary of how much static DRAM & IRAM is in use by the app.

You can try making more small memory allocations from PSRAM first by reducing the value of this config item:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-spiram-malloc-alwaysinternal

My colleague has also written some more tips for reducing memory usage here:
https://medium.com/the-esp-journal/esp32-memory-analysis-case-study-eacc75fe5431

filzek, 2020-08-25:
Hi, @projectgus 

I have made what you request and it shows:

Task Name       Status  Prio    HWM     Task#   Affinity
Sentinel   R       15      1504    15      0
IDLE1           R       0       532     7       1
IDLE0           R       0       632     6       0
tiT             B       18      2220    12      -1
Digital      B       5       3580    25      -1
Tmr Svc         B       1       1528    8       0
hciT            B       22      1644    23      0
sys_evt         B       20      900     13      0
btuT            B       20      2728    24      0
BTC_TASK        B       19      28596   22      0
btController    B       23      2048    21      0
wifi            B       23      4088    14      0
esp_timer       B       22      3320    3       0
ipc1            B       24      1520    2       1
ipc0            B       24      1528    1       0
Task Create Failed, trying to go to recreate task again, erro returned:[-1]
xPortGetFreeHeapSize: 4129024 before_count:15 before_heap_32:4137548 before_heap_8:4129200 before_heap_INTERNAL:18952

So I just have 2 tasks and its something eating all the memory and I cant find out what is using it.

Is there any code that show exactly what is running and using memory on the SDK? 
As, right now this is the more agressive problem.

Canary bytes doesnt show a heap corruption, char *task_list_buffer = malloc(uxTaskGetNumberOfTasks() * bytes_per_task); cant show the full usage, so still lost to understand memory usage,

chegewara, 2020-08-26:
`BTC_TASK B 19 28596 22 0` 
this task is having way too big stack size. You can easy decrease it by 25000. It is bluetooth task, so you increased it stack size in menuconfig i think.

projectgus, 2020-08-26:
Well spotted, @chegewara . The default size for this task is 4096 so [setting the config item back to this should help greatly](https://docs.espressif.com/projects/esp-idf/en/release-v4.0/api-reference/kconfig.html#config-bt-btu-task-stack-size).

> Is there any code that show exactly what is running and using memory on the SDK?

In addition to the functions you're already using, there is the [idf.py size, idf.py size-components, and idf.py size-files](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#advanced-commands) commands which print some information about the static memory allocations in an ESP-IDF project.

Also please check the other things linked in my previous reply, in particular changing the ratio of PSRAM to internal memory allocated by malloc can free up some internal memory.

But you may not need to do any of this after the BTC_TASK stack size is reduced.

filzek, 2020-08-28:
We have move some code off the btc_task area as if the callback executes  any function it shall need memory allocated to it  so, we have pushed it to another task and create a queue pool area so share among calls beside process within the callback, it shrink the btc_task but moves the code to the new task, so result is the same.

Also we found out that BLE, WIFI and HTTP sockets are consuming a lot of memory to be executed.

We are trying to identify why this huge blocks of memory are being pulled out from the heap as well.

chegewara, 2020-08-28:
Maybe you need to switch to nimble.
It is "normal" that bluedroid is using a lot of heap, HTTP, especially HTTPS also need some memory to make TLS, and wifi is also not very light module, but you should be able to run all 3 easy and stay with 50kB free heap.

dima0905, 2020-09-02:
fizer, are you using .cpp and enable CONFIG_COMPILER_CXX_EXCEPTIONS in your program? It takes more then 1M RAM

filzek, 2020-09-10:
We are not using it, but it seens some unexpected behavior for some default tasks, as we use some functions to write to fat inside the BLE call we need to increase the task size as it needs heap space to run, so we cant reduce that much, but seens that memory still a huge problem in the esp world.

dima0905, 2020-09-10:
Some components (e.g. nvs_flash) are written on CPP and in conjunction with CONFIG_COMPILER_CXX_EXCEPTIONS I've lost a huge heap of RAM. This features or bag appeared only in latest version. I had to roll back.

projectgus, 2020-09-14:
Hi @dima0905 ,

> are you using .cpp and enable CONFIG_COMPILER_CXX_EXCEPTIONS in your program? It takes more then 1M RAM

This sounds like a bug. Could you please open a new issue with the requested details including exact ESP-IDF version, etc?

As this option is disabled by default it sounds like it's unrelated to the issue reported here, at least.

Hi @filzek ,

>  as we use some functions to write to fat inside the BLE call we need to increase the task size as it needs heap space to run, so we cant reduce that much, but seens that memory still a huge problem in the esp world.

A FATFS operation shouldn't require an extra 24KB of stack overhead (4KB stack size increased to 28KB stack size). Is it possible you're doing something here which is inadvertently blowing out the stack requirement? Is there any code you can share?

> We have move some code off the btc_task area as if the callback executes any function it shall need memory allocated to it so, we have pushed it to another task and create a queue pool area so share among calls beside process within the callback

If you're doing relatively heavy operations (like filesystem operations) in the BTC callbacks then moving them off to a queue + worker task design like this is still probably a good idea. Depending on what else your firmware does, you may be able to consolidate some other operations here as well.

filzek, 2020-10-16:
So, can you guys advise what is the real memory available?

MALLOC_CAP_8BIT:
Heap summary for capabilities 0x00000004:
  At 0x3ffb2730 len 15448 free 0 allocated 15080 min_free 0
    largest_free_block 0 alloc_blocks 20 free_blocks 1 total_blocks 21
  At 0x3ffca5dc len 32767 free 356 allocated 31976 min_free 16
    largest_free_block 356 alloc_blocks 24 free_blocks 1 total_blocks 25
  At 0x3f804ad4 len 4175147 free 3258184 allocated 913908 min_free 3246496
    largest_free_block 3248820 alloc_blocks 185 free_blocks 12 total_blocks 197
  At 0x3ffaff10 len 240 free 0 allocated 96 min_free 0
    largest_free_block 0 alloc_blocks 7 free_blocks 0 total_blocks 7
  At 0x3ffb6388 len 7288 free 0 allocated 6988 min_free 0
    largest_free_block 0 alloc_blocks 16 free_blocks 1 total_blocks 17
  At 0x3ffb9a20 len 16648 free 40 allocated 14580 min_free 0
    largest_free_block 0 alloc_blocks 122 free_blocks 8 total_blocks 130
  At 0x3ffca5b8 len 88648 free 0 allocated 86944 min_free 0
    largest_free_block 0 alloc_blocks 104 free_blocks 1 total_blocks 105
  At 0x3ffe0440 len 15072 free 0 allocated 13360 min_free 0
    largest_free_block 0 alloc_blocks 105 free_blocks 0 total_blocks 105
  At 0x3ffe4350 len 113840 free 2716 allocated 109488 min_free 100
    largest_free_block 2136 alloc_blocks 98 free_blocks 6 total_blocks 104
  Totals:
    free 3261296 allocated 1192420 min_free 3246612 largest_free_block 3248820

 MALLOC_CAP_32BIT:
Heap summary for capabilities 0x00000002:
  At 0x3ffb2730 len 15448 free 0 allocated 15080 min_free 0
    largest_free_block 0 alloc_blocks 20 free_blocks 1 total_blocks 21
  At 0x3ffca5dc len 32767 free 356 allocated 31976 min_free 16
    largest_free_block 356 alloc_blocks 24 free_blocks 1 total_blocks 25
  At 0x3f804ad4 len 4175147 free 3258184 allocated 913908 min_free 3246496
    largest_free_block 3248820 alloc_blocks 185 free_blocks 12 total_blocks 197
  At 0x3ffaff10 len 240 free 0 allocated 96 min_free 0
    largest_free_block 0 alloc_blocks 7 free_blocks 0 total_blocks 7
  At 0x3ffb6388 len 7288 free 0 allocated 6988 min_free 0
    largest_free_block 0 alloc_blocks 16 free_blocks 1 total_blocks 17
  At 0x3ffb9a20 len 16648 free 40 allocated 14580 min_free 0
    largest_free_block 0 alloc_blocks 122 free_blocks 8 total_blocks 130
  At 0x3ffca5b8 len 88648 free 0 allocated 86944 min_free 0
    largest_free_block 0 alloc_blocks 104 free_blocks 1 total_blocks 105
  At 0x3ffe0440 len 15072 free 0 allocated 13360 min_free 0
    largest_free_block 0 alloc_blocks 105 free_blocks 0 total_blocks 105
  At 0x3ffe4350 len 113840 free 2716 allocated 109488 min_free 100
    largest_free_block 2136 alloc_blocks 98 free_blocks 6 total_blocks 104
  At 0x4009d100 len 12032 free 11984 allocated 0 min_free 11964
    largest_free_block 11984 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 3273280 allocated 1192420 min_free 3258576 largest_free_block 3248820

 MALLOC_CAP_INTERNAL:
Heap summary for capabilities 0x00000800:
  At 0x3ffb2730 len 15448 free 0 allocated 15080 min_free 0
    largest_free_block 0 alloc_blocks 20 free_blocks 1 total_blocks 21
  At 0x3ffca5dc len 32767 free 356 allocated 31976 min_free 16
    largest_free_block 356 alloc_blocks 24 free_blocks 1 total_blocks 25
  At 0x3ffaff10 len 240 free 0 allocated 96 min_free 0
    largest_free_block 0 alloc_blocks 7 free_blocks 0 total_blocks 7
  At 0x3ffb6388 len 7288 free 0 allocated 6988 min_free 0
    largest_free_block 0 alloc_blocks 16 free_blocks 1 total_blocks 17
  At 0x3ffb9a20 len 16648 free 40 allocated 14580 min_free 0
    largest_free_block 0 alloc_blocks 122 free_blocks 8 total_blocks 130
  At 0x3ffca5b8 len 88648 free 0 allocated 86944 min_free 0
    largest_free_block 0 alloc_blocks 104 free_blocks 1 total_blocks 105
  At 0x3ffe0440 len 15072 free 0 allocated 13360 min_free 0
    largest_free_block 0 alloc_blocks 105 free_blocks 0 total_blocks 105
  At 0x3ffe4350 len 113840 free 2716 allocated 109488 min_free 100
    largest_free_block 2136 alloc_blocks 98 free_blocks 6 total_blocks 104
  At 0x4009d100 len 12032 free 11984 allocated 0 min_free 11964
    largest_free_block 11984 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 15096 allocated 278512 min_free 12080 largest_free_block 11984

 MALLOC_CAP_8BIT|MALLOC_CAP_INTERNAL:
Heap summary for capabilities 0x00000804:
  At 0x3ffb2730 len 15448 free 0 allocated 15080 min_free 0
    largest_free_block 0 alloc_blocks 20 free_blocks 1 total_blocks 21
  At 0x3ffca5dc len 32767 free 356 allocated 31976 min_free 16
    largest_free_block 356 alloc_blocks 24 free_blocks 1 total_blocks 25
  At 0x3ffaff10 len 240 free 0 allocated 96 min_free 0
    largest_free_block 0 alloc_blocks 7 free_blocks 0 total_blocks 7
  At 0x3ffb6388 len 7288 free 0 allocated 6988 min_free 0
    largest_free_block 0 alloc_blocks 16 free_blocks 1 total_blocks 17
  At 0x3ffb9a20 len 16648 free 40 allocated 14580 min_free 0
    largest_free_block 0 alloc_blocks 122 free_blocks 8 total_blocks 130
  At 0x3ffca5b8 len 88648 free 0 allocated 86944 min_free 0
    largest_free_block 0 alloc_blocks 104 free_blocks 1 total_blocks 105
  At 0x3ffe0440 len 15072 free 0 allocated 13360 min_free 0
    largest_free_block 0 alloc_blocks 105 free_blocks 0 total_blocks 105
  At 0x3ffe4350 len 113840 free 2716 allocated 109488 min_free 100
    largest_free_block 2136 alloc_blocks 98 free_blocks 6 total_blocks 104
  Totals:
    free 3112 allocated 278512 min_free 116 largest_free_block 2136

**Is Just 3112??**

**or is it 15096??**

projectgus, 2020-11-12:
Hi @filzek ,

Sorry for the slow reply. The internal DRAM is `MALLOC_CAP_INTERNAL_MALLOC_CAP_8BIT` and the available RAM is 3112 bytes. 

If you count all of the internal IRAM & DRAM together then you have 15096 bytes. But the internal IRAM can only be allocated with `MALLOC_CAP_INTERNAL|MALLOC_CAP_32BIT` and this memory can only be read/written with 32-bit aligned reads/writes, so you can't use this as a general purpose RAM only for specific purposes with custom written code.

To try and move more of the internal DRAM allocations into PSRAM and free up some internal DRAM then you can tweak the [things mentioned here](https://github.com/espressif/esp-idf/issues/5783#issuecomment-679579933).

projectgus, 2021-01-04:
Closing this as it seems like the issue is to do with task internal memory usage in the firmware rather than something we can fix in ESP-IDF.

@filzek if you have notice some excessive memory use by ESP-IDF itself then please a new issue with the details of how we can reproduce this.

closed 2021-01-04 by projectgus