espressif/esp-idf #5785: Tool for creating FatFS images on host

JakubRakus, 2020-08-25:
ESP-IDF has great tools for creating SPIFFS filesystem images (`spiffsgen.py` and `mkspiffs`), but there is any official tool for creating images of FatFS partitions with support for wear leveling.

It would be great to see such a tool, officially supported by Espressif and fully compatible with latest 4.x branches. Tool would be very useful especially for mass production when You have to load a few, sometimes "big" (a few hundred of kB) files, individually generated for every piece of hardware - for example certificates and keys used for secure connections between ESP and cloud services, images and scripts for built-in web pages, app-specific config files etc.

There are some similar tools around the web, but firstly they looks unmaintained by their authors: (https://github.com/lllucius/esp32_fatfsimage - last commit 3 years ago, https://github.com/jkearins/ESP32_mkfatfs - also, last commit 3 years ago, https://github.com/TobleMiner/mk_esp32fat - last commit year ago) and secondly - just don't work with latest esp-idf versions (compilation errors, clunky configuration via files).

igrr, 2020-08-25:
Hi @JakubRakus, thanks for the good suggestion. Unfortunately I can't give you a timeline for implementing this feature, but we will consider this feature when planning next releases.

sio13, 2021-11-08:
Hi @JakubRakus, we have just implemented the requested feature. See the script [fatfsgen.py](https://github.com/espressif/esp-idf/blob/master/components/fatfs/fatfsgen.py). It is also integrated into the build system and you could easily include the feature in your project using the CMake function `fatfs_create_partition_image` (Beware that the feature is still in progress and there is a chance that the CMake function will be renamed). See also [an example](https://github.com/espressif/esp-idf/tree/master/examples/storage/fatfsgen) of how to use the feature.

The current solution supports FAT12, short file names only, and only read-only mounting. The reason for read-only access is that wear levelling is still not supported, however, it is the objective of the current work and will be provided soon.

Thanks again for the great suggestion and for your patience.

dobairoland, 2021-11-08:
Just for reference, this is the commit which adds this feature: 3c4034d36e91109cd8d0cd4f64672f5ee9a8bd29

JakubRakus, 2021-11-08:
Wow, absolutely fantastic! Great that the script is written in Python. I will definitely try this on my next project! Good work ESP team!

closed 2021-11-08 by JakubRakus