espressif/esp-idf #5786: Generating signing key for Secure Boot V2 throws an error

achraf-boussaada, 2020-08-25:
## Environment

- Development Kit:    none
- Module or chip used:  ESP32-WROOM-32E
- IDF version : ESP-IDF v4.3-dev-907-g6c17e3a64
- Build System:         idf.py
- Compiler version : (crosstool-NG esp-2020r2) 8.2.0
- Operating System:    Linux
- Using an IDE?: yes, visual code studio
- Power Supply:        USB

## Problem Description

the problem here is simple and straightforward, executing:
```
espsecure.py generate_signing_key secure_boot_signing_key.pem --version 2
```
throws the following error:
```
espsecure.py v3.0-dev
Traceback (most recent call last):
  File "/workspace/esp/esp-idf/components/esptool_py/esptool/espsecure.py", line 826, in <module>
    _main()
  File "/workspace/esp/esp-idf/components/esptool_py/esptool/espsecure.py", line 819, in _main
    main()
  File "/workspace/esp/esp-idf/components/esptool_py/esptool/espsecure.py", line 814, in main
    operation_func(args)
  File "/workspace/esp/esp-idf/components/esptool_py/esptool/espsecure.py", line 161, in generate_signing_key
    f.write(private_key)
TypeError: a bytes-like object is required, not 'str'
```
**NOTES:** 

- Executing the same command with the `--version 1` flag works without a problem

- Despite throwing an error with `--version 2`, the file was created but with 0 bytes as size.

hashyaha, 2020-08-26:
I changed f.write(private_key) into f.write(private_key.encode()) in line 161 of file espsecure.py and it works.

KonstantinKondrashov, 2020-08-26:
Hi @achraf-boussaada @hashyaha .This issue occurs under python3, but pytnon2 works well. We will fix this.

KonstantinKondrashov, 2020-08-26:
The fix is already in the master of esptool repo but was not updated in ESP-IDF. https://github.com/espressif/esptool/commit/c67e07943b2b87aead7d4ad3264951a269868442. We will update it ASAP.

Alvin1Zhang, 2020-10-21:
Thanks for reporting, and sorry for slow turnaround. The fix is available at https://github.com/espressif/esp-idf/commit/46b3a6d88e90c294a52d65398568df9da4d3f078. Feel free to reopen.

closed 2020-10-21 by Alvin1Zhang

pedros89, 2021-11-05:
> Hi @achraf-boussaada @hashyaha .This issue occurs under python3, but pytnon2 works well. We will fix this.

Hello, yes use python2 to always launch `espsecure.py`
You will have to install some modules, it is all exmplained [here ](https://github.com/espressif/arduino-esp32/issues/5645#issuecomment-961191953)