espressif/esp-idf #5787: idf.py uses the wrong python environment after a VS Code onboarding install

istokm, 2020-08-25:
- IDF version (run ``git describe --tags`` to find it): **v4.3-dev-907-g6c17e3a64**
- Build System: **idf.py**
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): **xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2020r2) 8.2.0**
- Operating System: **Windows**
- Environment type: **ESP Command Prompt** | **Powershell**
- Using an IDE?: **Visual Studio Code w/ ESP-IDF extension**

## Problem Description

After a clean installation through the ESP-IDF's onboarding the builtin ESP terminal can't run anything related to `idf.py` due to a package error:
```**PS D:\Infotech\Projects\ESP32-IDF\Projects\flowrack-mechanical> idf.py**
The following Python requirements are not satisfied:
gdbgui>=0.13.2.0
To install the missing packages, please run "D:\Infotech\Projects\ESP32-IDF\esp-idf\install.bat"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: D:\Infotech\Projects\ESP32-IDF\.espressif\python_env\idf4.3_py3.8_env
    Python interpreter used: D:\Infotech\Projects\ESP32-IDF\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe

ESP-IDF v4.3-dev-907-g6c17e3a64
**PS D:\Infotech\Projects\ESP32-IDF\Projects\flowrack-mechanical> pip show gdbgui**
Name: gdbgui
Version: 0.14.0.0
Summary: Browser-based frontend to gdb. Debug C, C++, Go, or Rust.
Home-page: https://github.com/cs01/gdbgui
Author: Chad Smith
Author-email: grassfedcode@gmail.com
License: License :: GNU GPLv3
Location: d:\infotech\projects\esp32-idf\.espressif\python_env\idf4.3_py3.8_env\lib\site-packages
Requires: Flask-SocketIO, eventlet, pygdbmi, gevent-websocket, Flask-Compress, Pygments, gevent, Flask
Required-by:
**PS D:\Infotech\Projects\ESP32-IDF\Projects\flowrack-mechanical> pip show pygdbmi**
Name: pygdbmi
Version: 0.9.0.2
Summary: Parse gdb machine interface output with Python
Home-page: https://github.com/cs01/pygdbmi
Author: Chad Smith
Author-email: grassfedcode@gmail.com
License: MIT
Location: d:\infotech\projects\esp32-idf\.espressif\python_env\idf4.3_py3.8_env\lib\site-packages
Requires: 
Required-by: gdbgui
```

This is having powershell set as default VS Code terminal, switching to cmd.exe yields that same result.

These are the paths that get added to the PS terminal when you launch it:

```
$Env:IDF_PATH = "D:\Infotech\Projects\ESP32-IDF\esp-idf"

$Env:Path = "D:\Infotech\Projects\ESP32-IDF\.espressif\python_env\idf4.3_py3.8_env\Scripts;D:\Infotech\Projects\ESP32-IDF\esp-idf\components\esptool_py\esptool;D:\Infotech\Projects\ESP32-IDF\esp-idf\components\espcoredump;D:\Infotech\Projects\ESP32-IDF\esp-idf\components\partition_table;D:\Infotech\Projects\ESP32-IDF\.espressif\python_env\idf4.3_py3.8_env\Scripts;C:/Users/istok/AppData/Local/Programs/Python/Python38-32;D:\Infotech\Projects\ESP32-IDF\.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\bin;D:\Infotech\Projects\ESP32-IDF\.espressif\tools\xtensa-esp32s2-elf\esp-2020r2-8.2.0\xtensa-esp32s2-elf\bin;D:\Infotech\Projects\ESP32-IDF\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;D:\Infotech\Projects\ESP32-IDF\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;D:\Infotech\Projects\ESP32-IDF\.espressif\tools\cmake\3.16.4\bin;D:\Infotech\Projects\ESP32-IDF\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;D:\Infotech\Projects\ESP32-IDF\.espressif\tools\ninja\1.10.0;D:\Infotech\Projects\ESP32-IDF\.espressif\tools\idf-exe\1.0.1;D:\Infotech\Projects\ESP32-IDF\.espressif\tools\ccache\3.7;D:\Infotech\Projects\ESP32-IDF\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;D:\Infotech\Projects\ESP32-IDF\.espressif\python_env\idf4.3_py3.8_env;D:\Infotech\Projects\ESP32-IDF\esp-idf\tools;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\Git\cmd;C:\Users\istok\AppData\Local\Microsoft\WindowsApps;C:\Users\istok\AppData\Local\Programs\Python\Python38-32\Scripts\;C:\Users\istok\AppData\Local\Programs\Python\Python38-32\;C:\Users\istok\AppData\Local\Programs\Microsoft VS Code\bin"
```

Running the manual installation scripts `install.bat` & `export.bat` yield weird incomplete output:
```
D:\Infotech\Projects\ESP32-IDF\esp-idf>install.bat
Installing ESP-IDF tools

D:\Infotech\Projects\ESP32-IDF\esp-idf>export.bat
Setting IDF_PATH: D:\Infotech\Projects\ESP32-IDF\esp-idf

Adding ESP-IDF tools to PATH...
```

I haven't been able to fix this in any way. No matter how many environment variables I remove or add, change the directory from which I call `idf.py`, nothing helps.

### Expected Behavior

The onboarding installation sets up everything required to run idf.py.

[... 31 lines cut ...]
Installing collected packages: pygdbmi
  Attempting uninstall: pygdbmi
    Found existing installation: pygdbmi 0.10.0.0
    Uninstalling pygdbmi-0.10.0.0:
      Successfully uninstalled pygdbmi-0.10.0.0
Successfully installed pygdbmi-0.9.0.2
WARNING: You are using pip version 20.1.1; however, version 20.2.2 is available.
You should consider upgrading via the 'c:\users\istok\appdata\local\programs\python\python38-32\python.exe -m pip install --upgrade pip' command.
```

### Steps to reproduce

This is my process, I couldn't be bothered trying all kinds of combinations as the process is quite long.

1. Clean install of VS Code
2. Install ESP-IDF extension
3. Go through the configuration, select existing esp-idf, download new tools to a non-standard directory. Use '/' when specifying python path due to issue 5510.. which is still a thing for some reason
4. Add IDF_TOOLS_PATH env. variable to the tools folder
5. Run install.bat & export.bat from a normal cmd window opened in that folder
6. Open ESP-IDF terminal (CTRL + E, T), type `idf.py`, to humour the idea that the package is actually missing, type `pip show gdbgui`

Here's a very basic video reproducing the issue. I reinstalled visual studio code, the esp-idf extension and cloned the latest esp-idf repo. I seems like it remembered a few settings, but otherwise nothing was set up on purpose.
https://youtu.be/W0KZqcSCnqk

_my python packages are an absolute mess thanks to this issue_

brianignacio5, 2020-08-25:
Thanks for posting this, in fact our debug adapter's requirements.txt has a conflict with current ESP-IDF master requirements.txt. We will update a fix as soon as possible. For now, you can fix in your current extension directory esp_debug_adapter/requirements.txt to match with ESP-IDF requirements.txt pygdbmi version.

You can also post an issue to the vscode extension repo directly in https://github.com/espressif/vscode-esp-idf-extension

istokm, 2020-08-25:
I see the version conflict, but even after matching the version numbers the installation ends with an error, and `install.bat` still doesn't do anything. I've also added gdbgui into the `esp_debug_adapter/requirements.txt` to be sure, but with the same outcome.

```
The following Python requirements are not satisfied:
gdbgui>=0.13.2.0
To install the missing packages, please run "D:\Infotech\Projects\ESP32-IDF\esp-idf\install.bat"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: D:\Infotech\Projects\ESP32-IDF\.espressif\python_env\idf4.3_py3.8_env
    Python interpreter used: D:\Infotech\Projects\ESP32-IDF\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe
Command failed: "D:\Infotech\Projects\ESP32-IDF\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe" "D:\Infotech\Projects\ESP32-IDF\esp-idf\tools\check_python_dependencies.py" -r "D:\Infotech\Projects\ESP32-IDF\esp-idf\requirements.txt"
```

Manually going into the python environment reports that these packages are properly installed
```
D:\Infotech\Projects\ESP32-IDF\.espressif\python_env\idf4.3_py3.8_env\Scripts>activate

(idf4.3_py3.8_env) D:\Infotech\Projects\ESP32-IDF\.espressif\python_env\idf4.3_py3.8_env\Scripts>pip show gdbgui
Name: gdbgui
Version: 0.14.0.0
Summary: Browser-based frontend to gdb. Debug C, C++, Go, or Rust.
Home-page: https://github.com/cs01/gdbgui
Author: Chad Smith
Author-email: grassfedcode@gmail.com
License: License :: GNU GPLv3
Location: d:\infotech\projects\esp32-idf\.espressif\python_env\idf4.3_py3.8_env\lib\site-packages
Requires: pygdbmi, Flask-SocketIO, Flask, eventlet, gevent, Pygments, gevent-websocket, Flask-Compress
Required-by:

(idf4.3_py3.8_env) D:\Infotech\Projects\ESP32-IDF\.espressif\python_env\idf4.3_py3.8_env\Scripts>pip show pygdbmi
Name: pygdbmi
Version: 0.9.0.2
Summary: Parse gdb machine interface output with Python
Home-page: https://github.com/cs01/pygdbmi
Author: Chad Smith
Author-email: grassfedcode@gmail.com
License: MIT
Location: d:\infotech\projects\esp32-idf\.espressif\python_env\idf4.3_py3.8_env\lib\site-packages
Requires:
Required-by: gdbgui
```

istokm, 2020-08-25:
https://www.gdbgui.com/installation/#windows-dependencies
> Note that windows is only supported for gdbgui versions less than 0.14.

Changing requirements to gdbgui<=0.13.2.0 (I also manually reinstalled it just in case) solved the issue.
When manually typing in `idf.py` into the ESP-IDF terminal, I still get a popup of _how do you want to open this file?_, but alteast the VSCode buttons work.
Don't know since when this issue exists, because this is the first clean install I've done in a while, but I'm surprised noone has come across this before.

closed 2020-09-16 by espressif-bot