espressif/esp-idf #5790: Is RSSI really working on BLE Mesh in the new release?

Thecave3, 2020-08-25:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32
- IDF version v4.3-dev-907-g6c17e3a64
- Build System:         [idf.py]
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:    Linux
- Using an IDE?: Yes Clion
- Power Supply:  USB

## Problem Description

From the release notes of latest version it says that RSSI has been enabled on BLE Mesh networks, but when I tried to log the values it does not show anything more than 0.

### Expected Behavior

Correct rssi being shown.

### Actual Behavior

rssi field equals to 0

### Steps to reproduce

1. Open the server_onoff_example
2. Add a log of the rssi field into the generic_callback registered on the server
3. flash and launch
4. provision with the Android nRF Mesh app
5. try to change value on the model from app
6. the packets logs by the following code should show the RSSI

### Code to reproduce this issue

```cpp

static void ble_mesh_generic_server_cb(esp_ble_mesh_generic_server_cb_event_t event,
                                       esp_ble_mesh_generic_server_cb_param_t *param) {
    ESP_LOGI(TAG,
             "net_idx 0x%04x, app_idx 0x%04x, src 0x%04x, dest 0x%04x, rcv_rssi 0x%02x, recv_ttl 0x%02x, send_rel 0x%02x, send_ttl 0x%02x, opcode 0x%08x, srv_send %s",
             param->ctx.net_idx, param->ctx.app_idx, param->ctx.addr, param->ctx.recv_dst, param->ctx.recv_rssi,
             param->ctx.recv_ttl, param->ctx.send_rel, param->ctx.send_ttl,
             param->ctx.recv_op, param->ctx.srv_send ? "true" : "false");
    }
}

esp_err_t ble_mesh_init(void) {
/* *** */ 
    esp_ble_mesh_register_generic_server_callback(ble_mesh_generic_server_cb);

/* *** */
}

```

## Debug Logs

Please watch carefully the BenchMark TAG which is the output produced by the code above.

```
I (134618) BenchMark: net_idx 0x0000, app_idx 0x0000, src 0x0001, dest 0x0002, rcv_rssi 0x00, recv_ttl 0x05, send_rel 0x00, send_ttl 0xff, opcode 0x00008206, srv_send false
I (134638) SERVER: ESP_BLE_MESH_GENERIC_SERVER_STATE_CHANGE_EVT
I (135948) BLE_MESH: recv, app_idx 0x0000 src 0x0001 dst 0x0002
I (135948) BLE_MESH: recv, len 2: 8205
I (135948) BLE_MESH: send, app_idx 0x0000 src 0x0002 dst 0x0001
I (135958) BLE_MESH: send, len 4: 820828dc
I (136418) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (146418) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (149678) BLE_MESH: recv, app_idx 0x0000 src 0x0001 dst 0x0002
I (149678) BLE_MESH: recv, len 7: 820699d9c40000
I (149678) BLE_MESH: send, app_idx 0x0000 src 0x0002 dst 0x0001
I (149688) BLE_MESH: send, len 4: 820899d9
I (149698) SERVER: event 0x00, opcode 0x8206, src 0x0001, dst 0x0002
I (149698) BenchMark: net_idx 0x0000, app_idx 0x0000, src 0x0001, dest 0x0002, rcv_rssi 0x00, recv_ttl 0x05, send_rel 0x00, send_ttl 0xff, opcode 0x00008206, srv_send false
I (149708) SERVER: ESP_BLE_MESH_GENERIC_SERVER_STATE_CHANGE_EVT
I (156428) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
```

## Other items if possible

- [x] [sdkconfig file](https://github.com/espressif/esp-idf/files/5123889/sdkconfig.txt)

Campou, 2020-08-26:
Hi @Thecave3 

Sorry for the misunderstanding caused.

In the release note, it means when a Provisioner scans the advertising packets from the unprovisioned devices, the RSSI of these unprovisioned advertising packets will be notified to the application layer.

For notifing the RSSI of other mesh messages, the corresponding MR has not been merged internally. Later it will be synchronized to the `release/v4.1` branch firstly, and will be released in `idf v4.1.1`.

Thanks.

Thecave3, 2020-08-26:
> In the release note, it means when a Provisioner scans the advertising packets from the unprovisioned devices, the RSSI of these unprovisioned advertising packets will be notified to the application layer.

Ok, thanks for the clarification.

> For notifing the RSSI of other mesh messages, the corresponding MR has not been merged internally. Later it will be synchronized to the release/v4.1 branch firstly, and will be released in idf v4.1.1.

Is there a way right now to perform this merge in local and make them work? More generally, is there a way to access and log rssi even in the lower levels of BLE Mesh stack?

Thecave3, 2020-08-28:
Regarding this issue I have now realized that actually the rssi sometimes works without problems and some other times it is just set to 0. I don't really know why, what I can state surely is that this expected behaviour happened when I correctly set up a SIG Generic Level BLE Mesh network with 1 Generic Level Server and 1 Generic Level Client.

I'll post more updates or pieces of code if you think they can be relevant for your debug.

Campou, 2020-09-01:
Hi @Thecave3 

Please have a try with this patch [0001-update-recv_rssi-for-received-mesh-message.patch.zip](https://github.com/espressif/esp-idf/files/5154915/0001-update-recv_rssi-for-received-mesh-message.patch.zip), it will update the `recv_rssi` of the received mesh messages before notifying to the application layer.

Thanks.

Thecave3, 2020-09-01:
Can you please also give me the instructions to apply the patch?

Thecave3, 2020-09-09:
To whoever needs [the aforementioned instructions](https://stackoverflow.com/questions/2249852/how-to-apply-a-patch-generated-with-git-format-patch).

In any case, it seems that in the actual latest release it is not a problem since it is already implemented.
The only issue right now seems to happen exclusively during the provisioning phase with the nRF app, but it runs smooth on classical client server connections.

closed 2020-09-09 by Thecave3