espressif/esp-idf #5791: esp-idf/examples/bluetooth/bluedroid/ble/ble_hid_device_demo. error

dragonpilot, 2020-08-25:
ESP32-WROOM-32D
release/v4.2

code   esp-idf/examples/bluetooth/bluedroid/ble/ble_hid_device_demo

error:
MONITOR
--- WARNING: Serial ports accessed as /dev/tty.* will hang gdb if launched.
--- Using /dev/cu.usbserial-01CC5902 instead...
--- idf_monitor on /dev/cu.usbserial-01CC5902 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6924
load:0x40078000,len:14340
load:0x40080400,len:3980
entry 0x40080680
I (29) boot: ESP-IDF v4.2-dev-2086-g2807a0397-dirty 2nd stage bootloader
I (29) boot: compile time 21:32:27
I (30) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) bable:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1c5d8 (116184) map
I (157) esp_image: segment 1: paddr=0x0002c600 vaddr=0x3ffbdb60 size=0x037e0 ( 14304) load
I (163) esp_image: segment 2: paddr=0x0002fde8 vaddr=0x40080000 size=0x00230 (   560) load
0x40080000: _WindowOverflow4 at /Users/cafe/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (164) esp_image: segment 3: paddr=0x00030020 vaddr=0x400d0020 size=0x787bc (493500) map
0x400d0020: _stext at ??:?

I (359) esp_image: segment 4: paddr=0x000a87e4 vaddr=0x40080230 size=0x001d0 (   464) load
0x40080230: _Level4Vector at ??:?

I (360) esp_image: segment 5: paddr=0x000a89bc vaddr=0x40080400 size=0x13808 ( 79880) load
I (411) boot: Loaded app from partition at offset 0x10000
I (411) boot: Disabling RNG early entropy source...
I (412) cpu_start: Pro cpu up.
I (415) cpu_start: Application information:
I (420) cpu_start: Project name:     hidd_demos
I (425) cpu_start: App version:      v4.2-dev-2086-g2807a0397-dirty
I (432) cpu_start: Compile time:     Aug 25 2020 21:32:36
I (438) cpu_start: ELF file SHA256:  926d0b7a9501c99c...
I (444) cpu_start: ESP-IDF:          v4.2-dev-2086-g2807a0397-dirty
I (451) cpu_start: Starting app cpu, entry point is 0x40081208
0x40081208: call_start_cpu1 at /Users/cafe/esp-idf/components/esp32/cpu_start.c:282

I (0) cpu_start: App cpu up.
I (462) heap_init: Initializing. RAM available for dynamic allocation:
I (469) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (475) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (481) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (487) heap_init: At 3FFC8F90 len 00017070 (92 KiB): DRAM
I (493) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (499) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (506) heap_init: At 40093C08 len 0000C3F8 (48 KiB): IRAM
I (512) cpu_start: Pro cpu start user code
I (530) spi_flash: detected chip: generic
I (531) spi_flash: flash io: dio
I (531) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (615) BTDM_INIT: BT controller compile version [c293075]
E (615) HID_DEMO: app_main initialize controller failed

Alvin1Zhang, 2020-08-26:
Thanks for reporting, we will look into.

me-no-dev, 2020-08-28:
Hi @dragonpilot ! This is an old example and will be removed. Please refer to https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_hid_device instead.

closed 2020-09-09 by dragonpilot