espressif/esp-idf #5792: BLUFI is BREAK -> Error - Custom Data and Others - Example

filzek, 2020-08-25:
Hi,

The BluFi example is broken, cant work.

I (0) cpu_start: App cpu up.
I (572) cpu_start: Pro cpu start user code
I (572) cpu_start: Application information:
I (572) cpu_start: Project name:     blufi_demo
I (577) cpu_start: App version:      v4.3-dev-907-g6c17e3a64-dirty
I (584) cpu_start: Compile time:     Aug 25 2020 16:43:05
I (590) cpu_start: ELF file SHA256:  362b9a16f2835f76...
I (596) cpu_start: ESP-IDF:          v4.3-dev-907-g6c17e3a64-dirty
I (603) heap_init: Initializing. RAM available for dynamic allocation:
I (610) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (616) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (622) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (629) heap_init: At 3FFC9280 len 00016D80 (91 KiB): DRAM
I (635) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (641) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (647) heap_init: At 4009D024 len 00002FDC (11 KiB): IRAM
W (655) flash_encrypt: Flash encryption mode is DEVELOPMENT (not secure)
I (661) spi_flash: detected chip: gd
I (665) spi_flash: flash io: dio
W (669) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (684) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (788) wifi:wifi driver task: 3ffcd670, prio:23, stack:6656, core=0
I (788) system_api: Base MAC address is not set
I (788) system_api: read default base MAC address from EFUSE
I (808) wifi:wifi firmware version: 8d94a49
I (808) wifi:wifi certification version: v7.0
I (808) wifi:config NVS flash: enabled
I (808) wifi:config nano formating: disabled
I (818) wifi:Init data frame dynamic rx buffer num: 32
I (818) wifi:Init management frame dynamic rx buffer num: 32
I (828) wifi:Init management short buffer num: 32
I (828) wifi:Init dynamic tx buffer num: 32
I (838) wifi:Init static rx buffer size: 1600
I (838) wifi:Init static rx buffer num: 10
I (848) wifi:Init dynamic rx buffer num: 32
I (848) wifi_init: rx ba win: 6
I (848) wifi_init: tcpip mbox: 32
I (858) wifi_init: udp mbox: 6
I (858) wifi_init: tcp mbox: 6
I (858) wifi_init: tcp tx win: 5744
I (868) wifi_init: tcp rx win: 5744
I (868) wifi_init: tcp mss: 1440
I (878) wifi_init: WiFi IRAM OP enabled
I (878) wifi_init: WiFi RX IRAM OP enabled
I (978) phy: phy_version: 4370, 4e803b3, Aug 11 2020, 14:18:07, 0, 0
I (978) wifi:mode : sta (4c:11:ae:db:1d:b0)
I (978) BTDM_INIT: BT controller compile version [270f54d]
I (1308) BLUFI_EXAMPLE: BD ADDR: 4c:11:ae:db:1d:b2

I (1308) BLUFI_EXAMPLE: BLUFI VERSION 0102

I (1308) BLUFI_EXAMPLE: BLUFI init finish

trying to send Custom Data return eternal loop of

E (136678) BT_BTC: btc_blufi_recv_handler seq 117 is not expect 1

E (136678) BLUFI_EXAMPLE: BLUFI report error, error code 0

E (136688) BT_L2CAP: l2ble_update_att_acl_pkt_num not found p_tcb
W (136688) BT_BTC: btc_blufi_send_encap wait to send blufi custom data

E (136708) BT_L2CAP: l2ble_update_att_acl_pkt_num not found p_tcb
W (136708) BT_BTC: btc_blufi_send_encap wait to send blufi custom data

E (136718) BT_L2CAP: l2ble_update_att_acl_pkt_num not found p_tcb
W (136718) BT_BTC: btc_blufi_send_encap wait to send blufi custom data

E (136738) BT_L2CAP: l2ble_update_att_acl_pkt_num not found p_tcb
W (136738) BT_BTC: btc_blufi_send_encap wait to send blufi custom data

**If trying to request Version returns infinite loop of:**

E (30858) BT_L2CAP: l2ble_update_att_acl_pkt_num not found p_tcb
W (30858) BT_BTC: btc_blufi_send_encap wait to send blufi custom data

**If trying to request ScanWifi returns infinite loop of:**

E (49298) BT_L2CAP: l2ble_update_att_acl_pkt_num not found p_tcb
W (49298) BT_BTC: btc_blufi_send_encap wait to send blufi custom data

**If trying to send CUSTOM DATA it returns infinite loop of:**

E (106428) BT_BTC: btc_blufi_recv_handler seq 32 is not expect 1

E (106428) BLUFI_EXAMPLE: BLUFI report error, error code 0

Please advise as it was working correct before update sdk.

Alvin1Zhang, 2020-08-26:
Thanks for reporting and sorry for the inconvenience, we will look into.

filzek, 2020-08-28:
@Alvin1Zhang  We have found some ways to trigger the issue, if the callbacks are outside the main file it will return unexpectedly behaviors.

We have test to build in v4.1 and it works but functions must be in the same file, in v4.3 doesnt matter it doesnt work.

xiewenxiang, 2020-08-28:
@filzek 
Can you describe in more detail how to reproduce this problem? It would be even better to have example code.

filzek, 2020-09-04:
> Hello @Alvin1Zhang <https://github.com/Alvin1Zhang> & @xiewenxiang
> <https://github.com/xiewenxiang>, I am hitting the same issue.
>
> I have bisected it to fa729f3

The example doesn't work as intent, we have found out some errors but we
cant recreate where they happen, but if the callbacks are not in the same
file the error occurs, just move the callbacks and calls to the same file
and it will work. But custom data dont.

xiewenxiang, 2020-09-05:
@filzek 

How long is the custom data you are sending? Can you provide a simple demo that can be reproduced? If not, could you  describe the operating steps, and we will try to reproduce this scenario

hb-kaertech, 2020-09-07:
@xiewenxiang : I have spend some time to make a reproducer:

- Use `esp-idf/examples/bluetooth/bluedroid/ble/blufi` as is (I am using master branch)
- Patch EspBlufiForAndroid with the following: (this adds random long custom data just after wifi provisioning)
```
diff --git a/app/src/main/java/com/espressif/espblufi/ui/BlufiActivity.java b/app/src/main/java/com/espressif/espblufi/ui/BlufiActivity.java
index bc7b052..0ef760b 100644
--- a/app/src/main/java/com/espressif/espblufi/ui/BlufiActivity.java
+++ b/app/src/main/java/com/espressif/espblufi/ui/BlufiActivity.java
@@ -144,6 +144,32 @@ public class BlufiActivity extends BaseActivity {
                 BlufiConfigureParams params =
                         (BlufiConfigureParams) data.getSerializableExtra(BlufiConstants.KEY_CONFIGURE_PARAM);
                 configure(params);
+
+                String dataLongStr = "AAAAAAAAAABBBBBBBBBB" +
+                                     "CCCCCCCCCCDDDDDDDDDD" +
+                                     "EEEEEEEEEEFFFFFFFFFF" +
+                                     "CCCCCCCCCCDDDDDDDDDD" +
+                                     "EEEEEEEEEEFFFFFFFFFF" +
+                                     "CCCCCCCCCCDDDDDDDDDD" +
+                                     "EEEEEEEEEEFFFFFFFFFF" +
+                                     "CCCCCCCCCCDDDDDDDDDD" +
+                                     "EEEEEEEEEEFFFFFFFFFF" +
+                                     "CCCCCCCCCCDDDDDDDDDD" +
+                                     "EEEEEEEEEEFFFFFFFFFF" +
+                                     "CCCCCCCCCCDDDDDDDDDD" +
+                                     "EEEEEEEEEEFFFFFFFFFF" +
+                                     "CCCCCCCCCCDDDDDDDDDD" +
+                                     "EEEEEEEEEEFFFFFFFFFF" +
+                                     "CCCCCCCCCCDDDDDDDDDD" +
+                                     "EEEEEEEEEEFFFFFFFFFF" +
+                                     "CCCCCCCCCCDDDDDDDDDD" +
+                                     "EEEEEEEEEEFFFFFFFFFF" +
+                                     "CCCCCCCCCCDDDDDDDDDD" +
+                                     "EEEEEEEEEEFFFFFFFFFF" +
+                                     "CCCCCCCCCCDDDDDDDDDD" +
+                                     "EEEEEEEEEEFFFFFFFFFF" +
+                                      "";
+                mBlufiClient.postCustomData(dataLongStr.getBytes());
             }
 
             return;

```
- Build & flash everything
- On the Android app, connect, then Configure Wifi.
- Boom, I hit the issue

My guess about why https://github.com/espressif/esp-idf/commit/fa729f3805433ff1aa262978515a49a17f1246df is bad: `btc_blufi_send_encap` is being called from the `btuT` (aka bluedroid) task. This exact same task is the one able to process the queue. Thus calling `vTaskDelay` won't allow `btuT` to run and process the queue.

persan666, 2020-09-14:
Hi, I'm facing the same issue when sending custom data from ESP32 to Android APP.
Endless looping the following two lines:

E (10789) BT_L2CAP: l2ble_update_att_acl_pkt_num not found p_tcb
W (10794) BT_BTC: btc_blufi_send_encap wait to send blufi custom data

jimwhitfield, 2020-09-14:
I just checked out 4.1 
C:\esp\esp-idf\examples\protocols\mqtt\tcp>git status
On branch release/v4.1
Your branch is up to date with 'origin/release/v4.1'.

It dies like this with the original example (and I'm using a WROVER board)

hb-kaertech, 2020-09-22:
@xiewenxiang : When do you expect to look into this? Thank you!

jimwhitfield, 2020-09-22:
Just to note: when using the code at SHA bd72a9ab2 I had other issues. 
I decided I'd rather put efforts into trying release 4.2, and that appears to be all good.  
In fact, I'm fully functional with all the expected code working.
Espressif folk, you might want to see if you can use that as a basis to fix 4.1 or just decree 4.2 ready sooner rather than later.

hb-kaertech, 2020-09-22:
@jimwhitfield : I am currently using the master branch (so v4.3) and I still see this issue. I have a local patch in esp-idf to circumvent this.

filzek, 2020-09-22:
We have fix it already with a lot of pain, making some modifications we
make it to work on v4.1 release again, but solution is complete weird and
makes no logic at all.

But I can say that blufi continue break.

xiewenxiang, 2020-09-23:
@hb-kaertech @jimwhitfield 

ESP32 sends customer data to the APP is through notify.  If the MTU size doesn't exchange, it will uses the default 23 bytes, at this point, if sends a lot of data to the phone, such as a few thousand bytes, or more, the queues will soon fill up, So a mechanism has been added here to detect the number of remaining sendable packets.

I retest it, send 2k or 5k cuntomer data to APP, did not encounter such a problem, so I want to know, Do I need to do anything other than send customer data to APP?

hb-kaertech, 2020-09-23:
@xiewenxiang : Did you follow my test plan in https://github.com/espressif/esp-idf/issues/5792#issuecomment-688252081 ?

xiewenxiang, 2020-09-24:
@hb-kaertech  @filzek @jimwhitfield 

If it is urgent, can you make such modification temporarily? We are doing further analysis.

`components/bt/host/bluedroid/btc/profile/esp/blufi/blufi_prf.c`

`esp_ble_get_cur_sendable_packets_num(blufi_env.conn_id)`  -> `esp_ble_get_sendable_packets_num()`

hb-kaertech, 2020-09-24:
@xiewenxiang : I have reverted the offending commit and I'm quite fine with it right now.

However, as we are many people affected by this, we would like to make sure you can reproduce the issue for it will be fixed one day.

marciopamplona, 2020-10-30:
Same issue here. 
ESP-IDF master branch (blufi example from there)

persan666, 2020-11-09:
I wrote on the Android Git repository that the problem could be fixed by reverting the commit for the file blufi_prf.c but the problem persists using a Samsung Galaxy S10+ phone. 
Any news about this, someone?

persan666, 2020-11-09:
I have confirmed that it is working in esp-idf v3.3.1-133-g817efe8aa

persan666, 2020-11-09:
Just to clarify, the problem is in the 4.0 branch. I have verified if it works in 4.1/4.2/master

Stefano-Orsolini, 2020-12-12:
I traced the problem to function gatt_get_tcb_by_idx in components/bt/host/bluedroid/stack/gatt/gatt_utils.c

When called  returns a NULL pointer for p_tcb when the input argument tcb_idx is 3.
For clariity tcb_idx is the same as blufi_env.conn_id in the code indicated by @filzek.

the if statement is not entered because there is no connection id 3 allocated in gatt_cb (in my case only connection id 0 is allocated) hence NULL is returned.

I have this problem using the BLUFI APP from Android and iOS and esp-idf 4.1/4.2/4.3.

Chandrajadeja, 2020-12-23:
Hey, 
I also have the same issue with ESP-IDFv4.2.
The issue is 100% reproducible by sending custom data from Blufi IOS app version 1.3.1 and BluFi Library version  2.1.0

I have tried the fix  suggested by @xiewenxiang 
`
components/bt/host/bluedroid/btc/profile/esp/blufi/blufi_prf.c
esp_ble_get_cur_sendable_packets_num(blufi_env.conn_id) -> esp_ble_get_sendable_packets_num()`

Thanks it worked for me :)

xiewenxiang, 2020-12-24:
This issue has been fixed and it is recommended to update the IDF.

Chandrajadeja, 2020-12-24:
Can you share the commit#?

Stefano-Orsolini, 2020-12-24:
> This issue has been fixed and it is recommended to update the IDF.

confirmed, thanks!

on a general note when running menuconfig after updating idf check that BT_BLE_BLUFI_ENABLE=y for unknown reasons mine switched to 'n'.

anslex, 2021-01-12:
@xiewenxiang upgrade to wich version ?
An issue is still observed on v4.2:
Refs: [release/v4.2], {origin/release/v4.2}, v4.2-47-g2532ddd

filzek, 2021-01-22:
RELEASE V4.2 still break, cant work blufi!

daichengqian, 2021-02-22:
RELEASE V4.0.2 already have this problem , only RELEASA V4.0 can work well

xiewenxiang, 2021-02-22:
@filzek  @daichengqian 

Please use the latest release/4.2 and release/v4.0 branch. If it still does not work, please provide the accurate commit.

daichengqian, 2021-03-22:
> @filzek @daichengqian
> 
> Please use the latest release/4.2 and release/v4.0 branch. If it still does not work, please provide the accurate commit.

yeah，i use latest  release/4.2 and my app version is 1.3.1 still have problem

xulongzhuang, 2021-03-22:
@daichengqian Please upgrade the app version. I tested it with the app version of Android 1.5.3 and the latest release/v4.2, and did not encounter this problem again

daichengqian, 2021-03-22:
> @daichengqian Please upgrade the app version. I tested it with the app version of Android 1.5.3 and the latest release/v4.2, and did not encounter this problem again

 I can t see IOS new version , did you upgrade?

xulongzhuang, 2021-03-22:
At present, the latest IOS version on the official website is 1.2.0, I think you used 1.3.1 before, isn’t it an Android app?

daichengqian, 2021-03-22:
iOS version is 1.3.1 in appstore upgraded three months ago.  you should check it.

发自我的iPhone

xulongzhuang, 2021-03-22:
sorry,I changed an iPhone, the version is indeed 1.3.1, but the latest release/v4.2 SDK used still does not show the problem described in the title, so, can you describe the phenomenon of your problem, whether it is Must appear?

yangxinzhe90, 2021-04-22:
Hello I have test v4.3 with Android. App v1.6.0 the demo blefi is not works (with error E (136708) BT_L2CAP: l2ble_update_att_acl_pkt_num not found p_tcb
W (136708) BT_BTC: btc_blufi_send_encap wait to send blufi custom data)we want to use this demo for esp32-c3 and esp32 do you have a temporary patch for fixed this pb? Thanks

vulko, 2021-05-19:
> @hb-kaertech @filzek @jimwhitfield
> 
> If it is urgent, can you make such modification temporarily? We are doing further analysis.
> 
> `components/bt/host/bluedroid/btc/profile/esp/blufi/blufi_prf.c`
> 
> `esp_ble_get_cur_sendable_packets_num(blufi_env.conn_id)` -> `esp_ble_get_sendable_packets_num()`

This fixed issue for me! **But please add this fix to your next release!!!**

xulongzhuang, 2021-05-24:
@yangxinzhe90 Did you test it on release/v4.3 and use the esp32-c3 chip? Can you provide the commit_id, we will find the problem and solve it as soon as possible

xulongzhuang, 2021-06-01:
> 
> iOS version is 1.3.1 in appstore upgraded three months ago. you should check it. 发自我的iPhone
> […](#)

@daichengqian , Will you still encounter this problem in the current IDF release/4.2 version or other IDF versions? What is the iPhone iOS version number you are using?

filzek, 2022-04-29:
Finally fixed in v4.4 and newer

Alvin1Zhang, 2022-04-29:
Thanks for sharing updates.

Songthinker, 2023-04-07:
The esp-idf I use: V4.4.2, I also occasionally have such problems today
ESP-ROM: esp32c3-api1-20210207
Build: Feb 7 2021
rst:0x1 (POWERON),boot:0xf (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode: DIO, clock div: 1
load:0x3fcd6100,len:0x16bc
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2d28
entry 0x403ce000
[0;32mI (30) boot: ESP-IDF v4.4.2-dirty 2nd stage bootloader [0m
[0;32mI (30) boot: compile time 11:43:18 [0m
[0;32mI (30) boot: chip revision: 3 [0m
[0;32mI (32) boot.esp32c3: SPI Speed : 80MHz [0m
[0;32mI (37) boot.esp32c3: SPI Mode : DIO [0m
[0;32mI (42) boot.esp32c3: SPI Flash Size : 4MB [0m
Occurs during the reconnection process after the network is disconnected..., the serial port keeps printing the following information:
[0;33mW (3488689) BT_BTC: esp_blufi_send_encap wait to send blufi custom data
[0;33mW (3488699) BT_BTC: esp_blufi_send_encap wait to send blufi custom data
[0;33mW (3488709) BT_BTC: esp_blufi_send_encap wait to send blufi custom data
[0;33mW (3488719) BT_BTC: esp_blufi_send_encap wait to send blufi custom data
[0;33mW (3488729) BT_BTC: esp_blufi_send_encap wait to send blufi custom data
[0;33mW (3488739) BT_BTC: esp_blufi_send_encap wait to send blufi custom data
[0;33mW (3488749) BT_BTC: esp_blufi_send_encap wait to send blufi custom data
[0;33mW (3488759) BT_BTC: esp_blufi_send_encap wait to send blufi custom data
[0;33mW (3488769) BT_BTC: esp_blufi_send_encap wait to send blufi custom data
[0;33mW (3488779) BT_BTC: esp_blufi_send_encap wait to send blufi custom data
[0;33mW (3488789) BT_BTC: esp_blufi_send_encap wait to send blufi custom data
[0;33mW (3488799) BT_BTC: esp_blufi_send_encap wait to send blufi custom data
[0;33mW (3488809) BT_BTC: esp_blufi_send_encap wait to send blufi custom data
[0;33mW (3488819) BT_BTC: esp_blufi_send_encap wait to send blufi custom data
[0;33mW (3488829) BT_BTC: esp_blufi_send_encap wait to send blufi custom data
[0;33mW (3488839) BT_BTC: esp_blufi_send_encap wait to send blufi custom data
[0;33mW (3488849) BT_BTC: esp_blufi_send_encap wait to send blufi custom data
......
Of course, my callback function and calling function are not in the same file, is it caused by this problem? But this problem is very difficult to reproduce, it just happened for the first time today