espressif/esp-idf #5797: wifi:bcn_timout,ap_probe_send_start while OTA update or uploading file on server.

ROYMURLI, 2020-08-26:
OS- Windows (msys2)
IDFv- 4.1-beta2

While updating the firmware OTA sometimes I am getting these errors and device stuck at this moment but sometimes it updated successfully. Same for posting any file to my server sometimes having this error and sometimes it is successful.

sdconfig is attached here:-
[sdkconfig - Copy.txt](https://github.com/espressif/esp-idf/files/5130707/sdkconfig.-.Copy.txt)

liuzfesp, 2020-08-27:
Hi @ROYMURLI ,  need your help to provide following info:

- What's the frequency of the problem?
- Does the problem exist with idf/examples/system/ota example?
- Does the problem exist with a different AP or hotspot?

ROYMURLI, 2020-08-27:
Hi @liuzfesp , thanks for your reply,
-> on OTA update the frequency is more. Let say 8 out of 10 for this problem. I have to restart my device to again start the OTA update task. On uploading/http post the frequency is low as 1 out of 10.
-> No, I have writing firmware for my product. The OTA task is attached below. It was working fine with 3.1.5 version. I have upgraded the IDF version and trying to check all the funtionality ang getting these error.
-> problem is with AP.

[fota.txt](https://github.com/espressif/esp-idf/files/5134774/fota.txt)

ROYMURLI, 2020-09-02:
any updates on this @liuzfesp ?

YouDONG-ESP, 2020-09-03:
Hi @ROYMURLI, could u provide a packet captured for this issue, I can't see much from your code?

ROYMURLI, 2020-09-05:
Hi @YouDONG-ESP  , the ota update code is attached below.
[fota.txt](https://github.com/espressif/esp-idf/files/5177834/fota.txt)

ROYMURLI, 2020-09-08:
Any updates? @YouDONG-ESP  and @liuzfesp

YouDONG-ESP, 2020-09-09:
> Hi @YouDONG-ESP , the OTA update code is attached below.
> [fota.txt](https://github.com/espressif/esp-idf/files/5177834/fota.txt)
The code you provided couldn't be run directly because It lacks many variables, I rewrote it with esp_http_client API and the bcn_timout didn't reproduce. Can you give me an example with bcn_timeout that can be run directly?

ROYMURLI, 2020-09-12:
Hi @YouDONG-ESP 
I don't have any other example for this issue. It is coming only in OTA updates in my device. Earlier I was using IDFv 3.1.5 and I was not getting this error. I moved to now v4.1-beta2 and also included bluedroid gattc and having this problem. Please find the attached sdk config for 4.1-beta2
[sdkconfig1 (2).txt](https://github.com/espressif/esp-idf/files/5212316/sdkconfig1.2.txt)

YouDONG-ESP, 2020-12-11:
@ROYMURLI Sorry for late reply, this issue can't be reproduce in our side, could you try to update IDF and see if it still happens? 
Could you tell me which AP did you used?

ROYMURLI, 2020-12-11:
Yes, I have updated to IDFv- v4.3-dev-1561-g357a27760
Now, I am getting this issue rarely when the wifi rssi is weak. This is not happening in AP mode. It is happening in STATION mode. But, in this IDF I am getting some other issue which is mentioned in this https://github.com/espressif/esp-idf/issues/6073

Alvin1Zhang, 2020-12-15:
@ROYMURLI Thanks for reporting and updates, for the other issue, let us track in https://github.com/espressif/esp-idf/issues/6073, feel free to reopen this ticket.

closed 2020-12-15 by Alvin1Zhang

AxelLin, 2021-09-01:
> Yes, I have updated to IDFv- v4.3-dev-1561-g357a27760
> Now, I am getting this issue rarely when the wifi rssi is weak. This is not happening in AP mode. It is happening in STATION mode. 

I'm using esp-idf v4.3-356-g48ae2309fd9c and my device is very closed to the AP.
Still got bcn_timout,ap_probe_send_start sometimes (rarely happen).
My device still alive and networking seems fine, just FYI.

I (3886) wifi:new:<11,2>, old:<1,0>, ap:<255,255>, sta:<11,2>, prof:1
I (4367) wifi:state: init -> auth (b0)
I (4373) wifi:state: auth -> assoc (0)
I (4384) wifi:state: assoc -> run (10)
I (4399) wifi:connected with IGS-MESH, aid = 1, channel 11, 40D, bssid = 18:a6:f7:ab:ca:66
I (4400) wifi:security: WPA2-PSK, phy: bgn, rssi: -35
I (4402) wifi:pm start, type: 1

I (4490) wifi:AP's beacon interval = 102400 us, DTIM period = 1

I (5450) wifi:Set ps type: 1

W (5527) wifi:<ba-add>idx:0 (ifx:0, 18:a6:f7:ab:ca:66), tid:0, ssn:0, winSize:64
I (74615270) wifi:bcn_timout,ap_probe_send_start
I (74615270) app:WiFi event_id:21
I (74627691) wifi:bcn_timout,ap_probe_send_start
I (74627691) app:WiFi event_id:21
I (74643768) wifi:bcn_timout,ap_probe_send_start
I (74643768) app:WiFi event_id:21

monkeytronics, 2023-04-24:
Was this issue every resolved? I'm using 4.3.1 & still seeing this exact issue in OTA.

AxelLin, 2023-04-24:
> Was this issue every resolved? I'm using 4.3.1 & still seeing this exact issue in OTA.

Please check if upgrading to v4.3.5 helps. I do not observe this issue for a while.