espressif/esp-idf #5798: (vQueueDelete)- assert failed!  - Error on FFAT mount and format on non formated storage

filzek, 2020-08-26:
SDK Release V4.1 

Error on Fat Format inside a bluetooth callback to a funciton.

Direct call a function or even create a task of it will produce the same result.

esp-idf/components/freertos/queue.c:1801 (vQueueDelete)- assert failed!
abort() was called at PC 0x400965ec on core 0
0x400965ec: vQueueDelete at C:/Users/DEV/Desktop/esp-idf/components/freertos/queue.c:1801 (discriminator 1)

code:

const esp_vfs_fat_mount_config_t mount_config = {
 				.max_files = 4,
 				.format_if_mount_failed = true,
 				.allocation_unit_size = CONFIG_WL_SECTOR_SIZE
};

esp_err_t err = esp_vfs_fat_spiflash_mount(base_path, "storage", &mount_config, &s_wl_handle);

this deploy the error.

ELF file SHA256: 001fbd36fe2903fe

Backtrace: 0x40093674:0x3fff2810 0x40093e59:0x3fff2830 0x400965ec:0x3fff2850 0x400eba45:0x3fff2870 0x400e9abf:0x3fff2890 0x400ebb02:0x3fff28d0 0x400e2133:0x3fff2900 0x400e2425:0x3fff2930 0x400e2b61:0x3fff29a0 0x400d6b69:0x3fff2d90 0x400f1df1:0x3fff2db0 0x40103bdb:0x3fff2df0 0x40104dbb:0x3fff2e10
0x40093674: invoke_abort at C:/Users/DEV/Desktop/esp-idf/components/esp32/panic.c:159

0x40093e59: abort at C:/Users/DEV/Desktop/esp-idf/components/esp32/panic.c:174

0x400965ec: vQueueDelete at C:/Users/DEV/Desktop/esp-idf/components/freertos/queue.c:1801 (discriminator 1)

0x400eba45: ff_del_syncobj at C:/Users/DEV/Desktop/esp-idf/components/fatfs/port/freertos/ffsystem.c:75

0x400e9abf: f_mount at C:/Users/DEV/Desktop/esp-idf/components/fatfs/src/ff.c:3521

0x400ebb02: esp_vfs_fat_spiflash_mount at C:/Users/DEV/Desktop/esp-idf/components/fatfs/vfs/vfs_fat_spiflash.c:76

0x400e2133: format_factory at c:\users\dev\desktop\esp-idf\dev\producao\build/../main/include/cmd_ble_parser.c:745

0x400e2425: parser_char_ble_cmd at c:\users\dev\desktop\esp-idf\dev\producao\build/../main/include/cmd_ble_parser.c:947

0x400e2b61: gatts_profile_a_event_handler at c:\users\dev\desktop\esp-idf\dev\producao\build/../main/include/ble_spp_server.c:622

0x400d6b69: gatts_event_handler at c:\users\dev\desktop\esp-idf\dev\producao\build/../main/include/ble_spp_server.c:860

0x400f1df1: btc_gatts_cb_to_app at C:/Users/DEV/Desktop/esp-idf/components/bt/host/bluedroid/btc/profile/std/gatt/btc_gatts.c:46
 (inlined by) btc_gatts_cb_handler at C:/Users/DEV/Desktop/esp-idf/components/bt/host/bluedroid/btc/profile/std/gatt/btc_gatts.c:805

0x40103bdb: btc_thread_handler at C:/Users/DEV/Desktop/esp-idf/components/bt/common/btc/core/btc_task.c:158

0x40104dbb: osi_thread_run at C:/Users/DEV/Desktop/esp-idf/components/bt/common/osi/thread.c:68

So, in Release v4.1 trying to format a storage using Fat will result in a assert call! and program abort / esp reset!

filzek, 2021-01-05:
we fixed this problem assing a 9 seconds delay after enable the bluetooth to perform the erase on flash, it works, but it was not supose to have problem doind this without waiting 9 seconds.

filzek, 2022-04-29:
Fixed in sdk 4.4 and newer

Alvin1Zhang, 2022-04-29:
Thanks for sharing updates.

filzek, 2023-01-26:
SDK st

filzek, 2023-01-26:
SAME PROBLEM AGAIN in the new SDK 4.4.3

igrr, 2023-04-24:
I've looked at the code mentioned in the backtrace, and I think there is one way to explain this crash:

- FatFS `f_mount` is called first time, it initializes a filesystem object and call `ff_cre_syncobj` to create a mutex, https://github.com/espressif/esp-idf/blob/349385cb645432b17a4f2148bdd0e863136324d1/components/fatfs/src/ff.c#L3529
- That fails due to lack of heap memory, returning `FR_INT_ERR`
- On the next call to `f_mount`, FatFS tries to delete the mutex (which hasn't been created last time, so is NULL), triggering this assert, https://github.com/espressif/esp-idf/blob/349385cb645432b17a4f2148bdd0e863136324d1/components/fatfs/src/ff.c#L3521

(I haven't found any other way how `ff_del_syncobj(cfs->sobj)` would be called with a NULL pointer argument.)

Considering that we'd rather not modify upstream FatFS code too much, we could adjust `ff_del_syncobj` to do nothing if a NULL pointer argument is given, and return successfully.

closed 2023-10-04 by filzek

garudaonekh, 2024-09-09:
Additionally,  If I enable "Bluedroid Memory Debug" in Menuconfig->Component->Bluetooth->Bluedroid, this crash will surface.