espressif/esp-idf #5799: release/v3.3 gpio.c build warning

maolin-china, 2020-08-27:
C:/Users/liumaolin/Desktop/MSYS32/esp-idf-v3.3/components/driver/gpio.c:267:13: warning: variable 'pd_en' set but not used [-Wunused-but-set-variable]
     uint8_t pd_en = 0;
             ^
C:/Users/liumaolin/Desktop/MSYS32/esp-idf-v3.3/components/driver/gpio.c:266:13: warning: variable 'pu_en' set but not used [-Wunused-but-set-variable]
     uint8_t pu_en = 0;
             ^
C:/Users/liumaolin/Desktop/MSYS32/esp-idf-v3.3/components/driver/gpio.c:265:13: warning: variable 'od_en' set but not used [-Wunused-but-set-variable]
     uint8_t od_en = 0;
             ^
C:/Users/liumaolin/Desktop/MSYS32/esp-idf-v3.3/components/driver/gpio.c:264:13: warning: variable 'output_en' set but not used [-Wunused-but-set-variable]
     uint8_t output_en = 0;
             ^
C:/Users/liumaolin/Desktop/MSYS32/esp-idf-v3.3/components/driver/gpio.c:263:13: warning: variable 'input_en' set but not used [-Wunused-but-set-variable]
     uint8_t input_en = 0;

Alvin1Zhang, 2020-08-27:
@maolin-china Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

maolin-china, 2020-08-27:
@Alvin1Zhang  
## Environment
- IDF version 
    v3.3.2-270-gf4333c8e3
- Build System: 
   idf.py
- Compiler version
   xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-96-g28523988) 5.2.0
- Operating System:
   windows
## Problem Description
Just build example/hello-world. 
In gpio.c, the function gpio_config declared variable pd_en/pu_en/od_en/output_en/input_en but not be used.

## Build Logs

```
C:/Users/liumaolin/Desktop/MSYS32/esp-idf-v3.3/components/driver/gpio.c:267:13: warning: variable 'pd_en' set but not used [-Wunused-but-set-variable]
uint8_t pd_en = 0;
^
C:/Users/liumaolin/Desktop/MSYS32/esp-idf-v3.3/components/driver/gpio.c:266:13: warning: variable 'pu_en' set but not used [-Wunused-but-set-variable]
uint8_t pu_en = 0;
^
C:/Users/liumaolin/Desktop/MSYS32/esp-idf-v3.3/components/driver/gpio.c:265:13: warning: variable 'od_en' set but not used [-Wunused-but-set-variable]
uint8_t od_en = 0;
^
C:/Users/liumaolin/Desktop/MSYS32/esp-idf-v3.3/components/driver/gpio.c:264:13: warning: variable 'output_en' set but not used [-Wunused-but-set-variable]
uint8_t output_en = 0;
^
C:/Users/liumaolin/Desktop/MSYS32/esp-idf-v3.3/components/driver/gpio.c:263:13: warning: variable 'input_en' set but not used [-Wunused-but-set-variable]
uint8_t input_en = 0;
```

Ethan-CYQ, 2020-09-03:
Hi, @maolin-china 
Can you check your `menuconfig` whether you set the log level to `Info`?
(the log level setting is on `Component config`->`Log output`->`Default log verbosity`)

maolin-china, 2020-09-03:
> 
> 
> Hi, @maolin-china
> Can you check your `menuconfig` whether you set the log level to `Info`?
> (the log level setting is on `Component config`->`Log output`->`Default log verbosity`)

The log level was set to 'Info'.

Ethan-CYQ, 2020-09-15:
Hi, @maolin-china 
It seem that it runs well in my local test. Can you provide your `sdkconfig`(under the build warning) so that we can reproduce your problem?

maolin-china, 2020-09-25:
I update the idf，it ok now.

closed 2020-09-25 by maolin-china