espressif/esp-idf #5800: problem about  esp_ble_gap_start_scanning

niexuzhong, 2020-08-27:
I hope to restart ble scan after it is stopped. I use the example of gatt_client as to test it. I set the duration as 10 seconds in  esp_ble_gap_start_scanning, then the scan is over after 10 second. Then I try to restart scan using esp_ble_gap_start_scanning. but the erros message is display as following:
E (23345) BT_BTM: BTM_BleScan scan already active
W (23345) BT_APPL:  bta_dm_ble_scan start scan failed. status=0x6

E (23345) GATTC_DEMO: scan start failed, error status = 1
E (23545) BT_BTM: BTM_BleScan scan already active
How can I restart the ble scan?

xiewenxiang, 2020-08-28:
@niexuzhong 
Please provide the version of ESP-IDF which did you used. 
A suggestion， Call the API to stop the scan and then re-start the scan.

niexuzhong, 2020-08-28:
the version is v4.2-dev-1905-g625bd5eb1
The scan should be stopped in fact. because:
1.The event ESP_GAP_SEARCH_INQ_CMPL_EVT had been triggered. So the scan should be over
2.I try use esp_ble_gap_stop_scanning() to stop before I restart the scanning. the error message is displayed as "scan is unactive"

xiewenxiang, 2020-08-28:
I repeated the test for a while in accordance with the steps you described, but failed to reproduce. 
Does it easy for you to repeat the problem? How long will it take?
In addition, What is the state of wi-fi at this point?

niexuzhong, 2020-08-29:
Thank for your nice help.It is so surprise that the scan become normal today. but The test environment is different. There are many Bluetooth devices in my office,but there are less device in my family, maybe some Bluetooth device can cause such errors? I don't know.  I will test it in Monday again in my office Thank you!

niexuzhong, 2020-09-02:
I test scan result in my office. the result is OK. Thank you!

closed 2020-09-02 by niexuzhong