espressif/esp-idf #5801: BLE mesh -- support for node subscribe/publish

AlpinProgrammer, 2020-08-27:
Hello all, 

We are currently using esp-idf for BLE mesh with bluedroid. 
So far, we always managed to use releases which are more stable.

We now need the support to subscribe/publish on groups. I know there exists _esp_ble_mesh_model_subscribe_group_addr()_.
However, this feature is only available on the master branch. It is said to be available on release/v4.0 (less stable than releases, but more than master) but it does not produce the expected effect contrary to master branch.

Would you have any knowledge on when we should expect a release version supporting this feature to be issued ?

Regards.

Campou, 2020-08-27:
Hi @AlpinProgrammer 

The functions [esp_ble_mesh_model_subscribe_group_addr](https://github.com/espressif/esp-idf/blob/e376e8f989547ffbd87d8746785470f710e71cd5/components/bt/esp_ble_mesh/api/core/include/esp_ble_mesh_local_data_operation_api.h#L126) and [esp_ble_mesh_model_unsubscribe_group_addr](https://github.com/espressif/esp-idf/blob/e376e8f989547ffbd87d8746785470f710e71cd5/components/bt/esp_ble_mesh/api/core/include/esp_ble_mesh_local_data_operation_api.h#L142) are now available on `release/v4.0` branch.

The next release verison which contains these functions will be idf v4.0.2, which may be released in September.

BTW, about the description ` it does not produce the expected effect contrary to master branch.`, if convenient, could you please give more information about this?

Thanks.

AlpinProgrammer, 2020-08-27:
Hi @Campou 
Thanks for the very fast answer !

When using release/v4.0 branch, _esp_ble_mesh_model_subscribe_group_addr_ returns ESP_OK.
But unlike with master branch, the ESP32 does not receive the messages publish on the group we subscribed for.

```
esp_err_t ret = esp_ble_mesh_model_subscribe_group_addr(addr, BLE_MESH_CID_NVAL, root_models[2].model_id, group_addr);

    if (ret == ESP_OK)
    {
        ESP_LOGI(TAG, "In %s, subscribing to group address succeeded !", __func__);
    }
    else
    {
        ESP_LOGE(TAG, "In %s, subscribing to group address failed !", __func__);
    }
```

Campou, 2020-09-01:
Hi @AlpinProgrammer 

Please change the log level of the mesh stack to `INFO`, this will provide more logs for debugging this issue.

And it may help to check whether the message is received by the mesh stack.

Thanks.

Alvin1Zhang, 2021-08-23:
Thanks for reporting, feel free to reopen.

closed 2021-08-23 by Alvin1Zhang

djbs291, 2022-09-12:
Hi @AlpinProgrammer, did you solve this issue? If yes, how did you solve it?