espressif/esp-idf #5802: Cannot connect socket Client to Server

federicoscata, 2020-08-27:
I'm having a problem to connect my ESP-W-Room 32d to a socket Server. I'm sure the problem is on Client, because with another Client in C# it works.

Here my "tcp_client_task" that starts when the event_handler check if ESP is connected to the AP.

```
case SYSTEM_EVENT_STA_CONNECTED:
		xTaskCreate(tcp_client_task, "tcp_client", 4096, NULL, 5, NULL);
```

"socket" is created correctly. When the code execute the function "**connect(sock, (struct sockaddr *)&dest_addr, sizeof(dest_addr))**", it returns "**ESP_LOGE(TAG, "Socket unable to connect: errno %d", errno)**"

```
#define HOST_IP_ADDR "127.0.0.1"
#define PORT 11000
```
```
static void tcp_client_task(void *pvParameters)
{
	char rx_buffer[128];
	char addr_str[128];
	int addr_family;
	int ip_protocol;

	while (1) {

		struct sockaddr_in dest_addr;
		dest_addr.sin_addr.s_addr = inet_addr(HOST_IP_ADDR);
		dest_addr.sin_family = AF_INET;
		dest_addr.sin_port = htons(PORT);
		addr_family = AF_INET;
		ip_protocol = IPPROTO_IP;

		
		int sock =  socket(addr_family, SOCK_STREAM, ip_protocol);
		if (sock < 0) {
			ESP_LOGE(TAG, "Unable to create socket: errno %d", errno);
			break;
		}
		ESP_LOGI(TAG, "Socket created, connecting to %s:%d", HOST_IP_ADDR, PORT);

		int err = connect(sock, (struct sockaddr *)&dest_addr, sizeof(dest_addr));
		if (err != 0) {
			ESP_LOGE(TAG, "Socket unable to connect: errno %d", errno);
			break;
		}
		ESP_LOGI(TAG, "Successfully connected");

		while (1) {
			int err = send(sock, payload, strlen(payload), 0);
			if (err < 0) {
				ESP_LOGE(TAG, "Error occurred during sending: errno %d", errno);
				break;
			}

			int len = recv(sock, rx_buffer, sizeof(rx_buffer) - 1, 0);
			// Error occurred during receiving
			if(len < 0) {
				ESP_LOGE(TAG, "recv failed: errno %d", errno);
				break;
			}
			// Data received
			else {
				rx_buffer[len] = 0;  // Null-terminate whatever we received and treat like a string
				ESP_LOGI(TAG, "Received %d bytes from %s:", len, addr_str);
				ESP_LOGI(TAG, "%s", rx_buffer);
			}

			vTaskDelay(2000 / portTICK_PERIOD_MS);
		}

		if (sock != -1) {
			ESP_LOGE(TAG, "Shutting down socket and restarting...");
			shutdown(sock, 0);
			close(sock);
		}
	}
	vTaskDelete(NULL);
}
```

Alvin1Zhang, 2020-08-28:
Thanks for updating the details, we will look into.

YouDONG-ESP, 2020-09-01:
Hi federicoscata, I tried your code and it works. Only thing I changed is inet_addr(), i used esp_ip4addr_aton(). You can try to replace this function.

#include "esp_netif.h"
#define HOST_IP_ADDR "192.168.10.161"
#define PORT 5001

static const char *payload = "Message from ESP32 ";

static void tcp_client_task(void *pvParameters)
{
	char rx_buffer[128];
	char addr_str[128];
	int addr_family;
	int ip_protocol;

	while (1) {
        
		struct sockaddr_in dest_addr;
		dest_addr.sin_addr.s_addr = esp_ip4addr_aton(HOST_IP_ADDR);
		dest_addr.sin_family = AF_INET;
		dest_addr.sin_port = htons(PORT);
		addr_family = AF_INET;
		ip_protocol = 0;
		
		int sockfd =  socket(AF_INET, SOCK_STREAM, IPPROTO_IP);
		if (sockfd < 0) {
			ESP_LOGE(TAG, "Unable to create socket: errno %d", errno);
			break;
		}
		ESP_LOGI(TAG, "Socket created, connecting to %s:%d", HOST_IP_ADDR, PORT);

		int err = connect(sockfd, (struct sockaddr*)&dest_addr, sizeof(dest_addr));
		if (err != 0) {
			ESP_LOGE(TAG, "Socket unable to connect: errno %d", errno);
		}
		ESP_LOGI(TAG, "Successfully connected");

		while (1) {
			int err = send(sockfd, payload, strlen(payload), 0);
			if (err < 0) {
				ESP_LOGE(TAG, "Error occurred during sending: errno %d", errno);
				break;
			}

			int len = recv(sockfd, rx_buffer, sizeof(rx_buffer) - 1, 0);
			// Error occurred during receiving
			if(len < 0) {
				ESP_LOGE(TAG, "recv failed: errno %d", errno);
				break;
			}
			// Data received
			else {
				rx_buffer[len] = 0;  // Null-terminate whatever we received and treat like a string
				ESP_LOGI(TAG, "Received %d bytes from %s:", len, addr_str);
				ESP_LOGI(TAG, "%s", rx_buffer);
			}

			vTaskDelay(2000 / portTICK_PERIOD_MS);
		}

		if (sockfd != -1) {
			ESP_LOGE(TAG, "Shutting down socket and restarting...");
			shutdown(sockfd, 0);
			close(sockfd);
		}

        vTaskDelay(500/portTICK_PERIOD_MS);
	}
	vTaskDelete(NULL);
}

		

void test(void)
{
    printf("test\n");
    xTaskCreate(tcp_client_task, "tcp_client", 4096, NULL, 5, NULL);
    printf("task done \n");
    while(1){

    }
}

LOG:
I (9049) tag: Socket created, connecting to 192.168.10.161:5001
I (9067) tag: Successfully connected
I (9080) tag: Received 28 bytes from �����������������@(@0:
I (9081) tag:

YouDONG-ESP, 2020-09-01:
Considering you didn't give the esp-idf commit, I used latesst version of master.

liuzfesp, 2020-09-07:
HI @federicoscata does the suggestion helpful for you?

federicoscata, 2020-09-10:
Yes, my code works correctly, I found out that there was a problem with the server ip.

Thanks.

closed 2020-09-10 by federicoscata