espressif/esp-idf #5803: ESP-IDF v4.3 unlike behavior when using MQTT over wss connection

fdworld, 2020-08-27:
Hello. I'm using ESP-IDF v4.3-dev-907-g6c17e3a64-dirty and got this unlike behavior when using MQTT over wss connection.
When using MQTT in a classical way all work fine.

Some one knows the best way to fix this?
Thank you.

V (69248) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (70248) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (71248) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
D (71938) esp_netif_lwip: esp_netif_dhcpc_cb lwip-netif:0x3ffd2ecc
D (71938) esp_netif_lwip: if0x3ffd2e4c ip unchanged
V (72248) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (73248) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (74248) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (75248) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (76248) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (77248) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (78248) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (79248) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (80248) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (81248) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (82248) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (83248) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (84248) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (85248) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (86248) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (87248) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (88248) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (89248) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (90248) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (91248) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
D (91248) RANDOM: getrandom(buf=0x3ffea212, buflen=4, flags=0)
D (91248) RANDOM: getrandom returns 4
D (91258) MQTT_CLIENT: Sent PING successful
D (91258) MQTT_CLIENT: PING sent
E (91448) TRANSPORT_WS: Error read data
E (91448) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=0
E (91448) MQTT_CLIENT: mqtt_process_receive: mqtt_message_receive() returned -1
D (91458) MQTT_CLIENT: Reconnect after 10000 ms

MQTT_EVENT_DISCONNECTEDD (101938) esp_netif_lwip: esp_netif_dhcpc_cb lwip-netif:0x3ffd2ecc
D (101938) esp_netif_lwip: if0x3ffd2e4c ip unchanged
D (106458) MQTT_CLIENT: Reconnecting...

Other event id:7D (106458) esp-tls: host:xxxxxxx: strlen 18
D (106658) esp-tls: handshake in progress...
D (108368) RANDOM: getrandom(buf=0x3ffea1d0, buflen=16, flags=0)
D (108368) RANDOM: getrandom returns 16
D (108368) TRANSPORT_WS: sub_protocol: mqtt
D (108368) TRANSPORT_WS: Write upgrade request
GET /mqtt HTTP/1.1
Connection: Upgrade
Host: xxxxxr:443
User-Agent: ESP32 Websocket Client
Upgrade: websocket
Sec-WebSocket-Version: 13
Sec-WebSocket-Key: 1FPa/X0FoOT/Iq3nVPTF2A==
Sec-WebSocket-Protocol: mqtt

D (108658) TRANSPORT_WS: Read header chunk 227, current header size: 227
D (108658) TRANSPORT_WS: server key=K1AEiaxuV3sa80SYoyLdf3iBXsM=, send_key=1FPa/X0FoOT/Iq3nVPTF2A==, expected_server_key=K1AEiaxuV3sa80SYoyLdf3iBXsM=
D (108668) MQTT_CLIENT: Transport connected to wss://xxxxxx:443
D (108678) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
D (108688) RANDOM: getrandom(buf=0x3ffea232, buflen=4, flags=0)
D (108688) RANDOM: getrandom returns 4
D (109168) TRANSPORT_WS: Opcode: 2, mask: 0, len: 4

D (109168) TRANSPORT_WS: Actual data to receive (4) are longer than ws buffer (1)
D (109168) MQTT_CLIENT: mqtt_message_receive: first byte: 0x20
D (109168) TRANSPORT_WS: Actual data to receive (3) are longer than ws buffer (1)
D (109178) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x2
D (109188) MQTT_CLIENT: mqtt_message_receive: total message length: 4 (already read: 2)
D (109198) MQTT_CLIENT: mqtt_message_receive: read_len=2
D (109198) MQTT_CLIENT: mqtt_message_receive: transport_read():4 4
D (109208) MQTT_CLIENT: Connected

MQTT_EVENT_CONNECTEDD (109208) MQTT_CLIENT: mqtt_enqueue id: 2552, type=8 successful
D (109218) OUTBOX: ENQUEUE msgid=2552, msg_type=8, len=39, size=39

Alvin1Zhang, 2020-08-28:
Thanks for reporting, we will look into.

david-cermak, 2020-08-30:
Hi @fdworld 

Seeing this message and as the `errno=Success` it seems like the server, for some reason terminated the connecting sending a `FIN` message
```
E (91448) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=0
```
Do you see any error messages on the broker side?
Is the same issue reproducible on for `ws` only transport? If yes, could you please share a packet capture?

fdworld, 2020-09-01:
Hi @david-cermak. The issue was solved. I only set a custom keepalive value and done. Some way a wss connection have a different behavior than an simple mqtt secure for ping requests. The default 120 seconds was not enough for broker when in wss.
Thank you for attention.

closed 2020-09-01 by fdworld