espressif/esp-idf #5805: can't connect ap with wpa2-enterprise when PMF=capable

kaedewang, 2020-08-28:
Environment

    Development Kit: ESP32-DevKitC
    Module or chip used: ESP32-WROOM-32D
    IDF version (run git describe --tags to find it):
    // v4.3-dev-771-gc77c4ccf6
    Build System: idf.py
    Compiler version (run xtensa-esp32-elf-gcc --version to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
    Operating System: wsl
    Using an IDE?: No
    Power Supply: USB

Problem Description

I used example station and wpa2_enterprise to test all auth types.
But ESP32 can't connect ap with wpa2-enterprise when PMF=capable.
Auth                                PMF             Connect       Example
WPA3-PSK                require        Yes                station
WPA2-PSK                capable       Yes                station
WPA2-PSK                              require        Yes                station
WPA2-EAP-TLS         capable       Yes                wpa2-enterprise
WPA2-EAP-TLS                                 requrire       No                wpa2-enterprise
WPA2-EAP-PEAP      capable       Yes                wpa2-enterprise
 WPA2-EAP-PEAP                                require         No               wpa2-enterprise
I added the following code to example wpa2-enterprise but it is unuseful.
.pmf_cfg = {
                .capable = true,
                .required = false
            },

Alvin1Zhang, 2020-08-28:
Thanks for reporting, we will look into.

negativekelvin, 2020-08-29:
Your results show "Yes".  Is that the expected result or actual result?

**Auth**|**PMF**|**Connect**|**Example**
:-----:|:-----:|:-----:|:-----:
WPA3-SAE|require|Yes|station
WPA2-PSK|capable|Yes|station
WPA2-PSK|require|Yes|station
WPA2-EAP-TLS|capable|Yes|wpa2-enterprise
WPA2-EAP-TLS|require|No|wpa2-enterprise
WPA2-EAP-PEAP|capable|Yes|wpa2-enterprise
WPA2-EAP-PEAP|require|No|wpa2-enterprise

kaedewang, 2020-08-31:
They are actual results.
I think all expected ressults should be Yes.

kaedewang, 2020-08-31:
BTW, PMF column is the setting in AP.
And ESP32 always uses the following setting.
.pmf_cfg = {
.capable = true,
.required = false
},

kapilkedawat, 2020-09-05:
Hi @kaedewang , We checked on the latest master with PMF required and 802.1x enabled. We were able to connect.

Is it possible to share more details about the issue to debug further?

Please help to provide:

1. Sniffer capture
2. AP details
3. Setup details(hostapd/freeradius version with os details).
4. Current codebase tip which you are using.
5. sdkconfig

Also, Please enable logging in WLAN to narrow down the issue. Logs can be enabled using menuconfig(component config -> Wi-Fi -> Enable Wi-Fi debug log  Debug log level = verbose, log submodule = all)

kaedewang, 2020-09-10:
Hi @kapilkedawat, I pull the lastest master branch. It doesn't work, either.
  1.Sniffer capture
    [capture.zip](https://github.com/espressif/esp-idf/files/5198529/capture.zip)
  2.AP details
    ASUS RT-AX56U，FW：ASUSwrt meilin 384.18，install freeradius3 by entware
  3.Setup details(hostapd/freeradius version with os details).
    Check https://github.com/espressif/esp-idf/issues/5698 for details
  4.Current codebase tip which you are using.
    [log.txt](https://github.com/espressif/esp-idf/files/5198537/log.txt)
  5.sdkconfig
    Almost wpa2-enterprise default config

kapilkedawat, 2020-09-10:
Hi @kaedewang , thanks for the logs. AP is using 'WPA2-Enterprise using SHA256' AKM which is not supported by the device at the moment and causes the device to not connect. You can use regular 'WPA2-Enterprise' AKM to connect.

Please note we are in the process of adding this AKM support and will update the issue once the changes are merged for this support.

sagb2015, 2020-10-08:
This is fixed by https://github.com/espressif/esp-idf/commit/ebc0b42c75a58ea456c064abd5ffd6c85006b5ec Feel free to reopen of the issue persists.

AxelLin, 2020-10-08:
> This is fixed by [ebc0b42](https://github.com/espressif/esp-idf/commit/ebc0b42c75a58ea456c064abd5ffd6c85006b5ec) Feel free to reopen of the issue persists.

But current master tree (v4.3-dev-1197-g8bc19ba893e5) does not include this fix. not yet.

sagb2015, 2020-10-08:
@AxelLin  Thanks for pointing out. Will check this.

closed 2020-10-14 by espressif-bot