espressif/esp-idf #5806: [question]

chegewara, 2020-08-28:
@Campou is there a way to stop, deinit, pause bluedroid ble mesh somehow? It would be useful in case of OTA update, which now takes about 10 minutes, which normally would be less than 1 minute.

Thanks

Campou, 2020-09-01:
Hi @chegewara 

Currently BLE Mesh provides a function `esp_ble_mesh_deinit` for de-initializing the whole mesh stack, and the parameter of this function can be used to decide whether erasing the mesh information from flash.

Thanks.

chegewara, 2020-09-02:
Yes, i know @Campou thanks, but this does not solved problem with stopping transfer every few seconds. It is better, but OTA still takes 5+ minutes, which usually takes about 1 minute.

AxelLin, 2020-09-02:
Is it possible to improve the title?

Campou, 2020-09-04:
Hi @chegewara 

So even if the mesh stack is de-initialized, there is still no big improvement for the OTA speed.

Just one more question here, I didn't understand `stopping transfer every few seconds` very clearly, did you mean the BLE transfer is just stopped every few seconds?

Thanks.

chegewara, 2020-09-06:
Ok, seems it is some other problem with transfer being paused every few seconds. Its strange, because i didnt have it with the same code in other apps using the same or similar ionic app.
I did some test and i disabled ble mesh completely and still OTA update is stopping every few seconds.
I think i can close issue for now, since i dont know where is the real problem, esp32 or android ionic app.

Thanks

closed 2020-09-06 by chegewara