espressif/esp-idf #5807: ipv6 not working in SoftAP mode

fariouche, 2020-08-28:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version : "v4.3-dev-907-g6c17e3a64" (git master 6c17e3a64)
- Build System:         [Make]
- Compiler version:  xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description
I'm building a ipv6 & ipv4 softap.
Using the simple softap example, I can run the softap, connect my smartphone and get an ip address. Good!
My issue is with ipv6... 
My code waits for the WIFI_EVENT_AP_START event, and when I get it, I call esp_netif_create_ip6_linklocal()
I see the netif debug trace saying "netif: IPv6 address 0 of interface  set to FE80::200:FF:FE00:0/0x08"
But I never receive the IP_EVENT_GOT_IP6.
If I try to call esp_netif_get_ip6_linklocal() after WIFI_EVENT_AP_START, I get an error -1.

Further investigations shows that the failure reason is because of the ip6 address not being PREFERED.

IPv6 works fine in station mode.

I'm not sure where to look at, and I unfortunately really need ipv6.

### Expected Behavior

### Actual Behavior

### Steps to reproduce
Just compile the softap_example_main.c example with the following additions:
ESP_ERROR_CHECK(esp_event_handler_instance_register(IP_EVENT,
                                                        ESP_EVENT_ANY_ID,
                                                        &ip_event_handler,
                                                        NULL,
                                                        NULL));
just after the already existing call to esp_event_handler_instance_register.

The ip_event_handler function:

static void ip_event_handler(void* arg, esp_event_base_t event_base,
                                    int32_t event_id, void* event_data)
{
 
    switch (event_id) 
    {
    case IP_EVENT_STA_GOT_IP:
        ESP_LOGI(GATTC_TAG, "got ipv4");
    break;
        
    case IP_EVENT_GOT_IP6:
    {
        ESP_LOGI(GATTC_TAG, "got ipv6"); 
    }
    break;

    default:
        break;
    }
}

and inside the wifi_event_handler:
if(event_id == WIFI_EVENT_AP_START)
{
        esp_err_t err;
        ESP_LOGI(GATTC_TAG, "WIFI_EVENT_AP_START");
       
        if(!esp_netif_is_netif_up(g_sta_netif))
[... 120 lines cut ...]
D (1852) BT_GATT: gatt_find_the_connected_bda start_idx=0
I (1852) wifi_init: tcp tx win: 5744
I (1862) wifi_init: tcp rx win: 5744
I (1872) wifi_init: tcp mss: 1440
I (1882) wifi_init: WiFi IRAM OP enabled
I (1892) wifi_init: WiFi RX IRAM OP enabled
---- creating softAP ------
netif: netmask of interface  set to 255.255.255.0
netif: GW address of interface  set to 192.168.4.1
netif_set_ipaddr: netif address being changed
netif: setting default interface 
netif_ip6_addr_set_state: netif address state being changed
netif: IPv6 address 0 of interface  set to FE80::200:FF:FE00:0/0x08
I (1952) wifi:mode : softAP (24:62:ab:b5:a8:89)
I (1952) wifi:Total power save buffer number: 16
I (1952) wifi:Init max length of beacon: 752/752
I (1952) wifi:Init max length of beacon: 752/752
WIFI_EVENT_AP_START
netif: netmask of interface  set to 255.255.255.0
netif: GW address of interface  set to 192.168.4.1
E (1972) main server: get IPv6 address failed -1
netif_set_ipaddr: netif address being changed
netif: added interface ap IP addr 192.168.4.1 netmask 255.255.255.0 gw 192.168.4.1
netif: setting default interface ap
E (3992) main server: get IPv6 address failed -1

david-cermak, 2020-08-30:
Hi @fariouche 

I can see that the LL address state is 0x08 immediately after calling the `esp_netif_create_ip6_linklocal()` (which is `IP6_ADDR_TENTATIVE`, looking for possibly duplicates on the network).
Could you please check what state the LL address is after 2-5 seconds after the `esp_netif_create_ip6_linklocal()`?
Please insert the below debugging message
```
    ESP_LOGI(TAG, "%s netif_ip6_addr_state=%d", __func__, netif_ip6_addr_state(p_netif, 0));
```
into [`esp_netif_get_ip6_linklocal()`](https://github.com/espressif/esp-idf/blob/master/components/esp_netif/lwip/esp_netif_lwip.c#L1534) and call it multiple times after creating the IPv6 LL?

fariouche, 2020-08-31:
Hi

I've already done the test, and it was always returning an error. I've redone the test with the debug trace you asked to be sure, calling the function every 2s. Still in TENTATIVE after 20s...

By the way, I tried also to set the lwip flag to not do the tentative retry (LWIP_IPV6_DUP_DETECT_ATTEMPTS), and the same error: (the only difference is that now the trace that displays the ipv6 address shows /0x30 (which means IP6_ADDR_PREFERRED I think)

netif: netmask of interface  set to 255.255.255.0
netif: GW address of interface  set to 192.168.4.1
netif_set_ipaddr: netif address being changed
netif: setting default interface 
netif_ip6_addr_set_state: netif address state being changed
netif: IPv6 address 0 of interface  set to FE80::200:FF:FE00:0/0x30
I (1742) wifi:mode : softAP (24:62:ab:b5:a8:89)
I (1742) wifi:Total power save buffer number: 16
I (1742) wifi:Init max length of beacon: 752/752
I (1742) wifi:Init max length of beacon: 752/752
V () careos core init iotivity
I (1742) CAREOS_BLE: WIFI_EVENT_AP_START
V () careos core init done
netif: netmask of interface  set to 255.255.255.0
netif: GW address of interface  set to 192.168.4.1
I (1762) main server: got IPv4 addr:192.168.4.1
netif_set_ipaddr: netif address being changed
netif: added interface ap IP addr 192.168.4.1 netmask 255.255.255.0 gw 192.168.4.1
I (1782) esp_netif_lwip: esp_netif_get_ip6_linklocal netif_ip6_addr_state=0
netif: setting default interface ap
E (1792) main server: get IPv6 address failed -1
I (3802) esp_netif_lwip: esp_netif_get_ip6_linklocal netif_ip6_addr_state=0
E (3802) main server: get IPv6 address failed -1
I (5802) esp_netif_lwip: esp_netif_get_ip6_linklocal netif_ip6_addr_state=0
E (5802) main server: get IPv6 address failed -1
I (7802) esp_netif_lwip: esp_netif_get_ip6_linklocal netif_ip6_addr_state=0
E (7802) main server: get IPv6 address failed -1
I (9802) esp_netif_lwip: esp_netif_get_ip6_linklocal netif_ip6_addr_state=0
E (9802) main server: get IPv6 address failed -1
I (11802) esp_netif_lwip: esp_netif_get_ip6_linklocal netif_ip6_addr_state=0
E (11802) main server: get IPv6 address failed -1
I (13802) esp_netif_lwip: esp_netif_get_ip6_linklocal netif_ip6_addr_state=0
E (13802) main server: get IPv6 address failed -1
I (15802) esp_netif_lwip: esp_netif_get_ip6_linklocal netif_ip6_addr_state=0

david-cermak, 2020-08-31:
Thanks for the update. Unfortunately I am not able to reproduce the issue, could you please also share your `sdkconfig`? (Could think of some configuration options, but AFAIK this state machine should work for all possible settings...)

> I tried also to set the lwip flag to not do the tentative retry

In that case you could use the assigned LL address as is, it should be valid and should work. But the `IP_EVENT_GOT_IP6` wouldn't come as the [`nd6 callback `](https://github.com/espressif/esp-lwip/blob/2.1.2-esp/src/core/ipv6/nd6.c#L1146) gets called only on status change and in this configuration it's directly set to 0x30 (`IP6_ADDR_PREFERRED`)

fariouche, 2020-08-31:
I see, this is good news if you do not reproduce it. It will be easier to pinpoint the issue.
Do you think you can share the test code you used?
This is the config file I'm using:
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5151990/sdkconfig.txt)

david-cermak, 2020-09-02:
@fariouche Thanks for the update, will check your configuration.

I simply placed these lines at the end of the `main()` function:
``` C
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_GOT_IP6, &on_got_ipv6, NULL));
    vTaskDelay(10000/ portTICK_PERIOD_MS);
    esp_err_t err = esp_netif_create_ip6_linklocal(g_sta_netif);
    ESP_LOGI(TAG, "esp_netif_create_ip6_linklocal returned %d", err);
```
(just assumed the AP to start prior to these 10s, of course adding an event flag, or calling it directly from the handler is a better option)

Plus an IPv6 handler which just prints out the address:
``` c
static void on_got_ipv6(void *arg, esp_event_base_t event_base, int32_t event_id, void *event_data)
{
    ip_event_got_ip6_t *event = (ip_event_got_ip6_t *)event_data;
    ESP_LOGI(TAG, "Got IPv6 event: Interface \"%s\" address: " IPV6STR, esp_netif_get_desc(event->esp_netif),IPV62STR(event->ip6_info.ip));
}
```

david-cermak, 2020-09-08:
@fariouche I've tested your configuration with the IDF version of 6c17e3a and it works okay for me:
```
netif_ip6_addr_set_state: netif address state being changed
netif: IPv6 address 0 of interface ap set to FE80:3XXXX9/0x08
I (10862) wifi softAP: esp_netif_create_ip6_linklocal returned 0
netif_ip6_addr_set_state: netif address state being changed
netif: IPv6 address 0 of interface ap set to FE80::3XXXX9/0x09
netif_ip6_addr_set_state: netif address state being changed
netif: IPv6 address 0 of interface ap set to FE80::3XXXX9/0x30
I (12602) wifi softAP: Got IPv6 event: Interface "ap" address: fe80:0000:0000:0000:3xxx:xxx:xxx:xxx9
```

Any news from your end? Have you tried the snippets above?

fariouche, 2020-09-08:
Hello,

Sorry, I'm facing other issues currently that I need to sort out before doing again the investigation. Hopefully soon

fariouche, 2020-12-09:
Sorry, closed by mistake!
I'm back on the investigation of this issue

I tried your 4 lines of code in the main and guess what, it worked.

The create_linklocal code in my case was done inside the AP_START event and in that case it does not work.

fariouche, 2020-12-09:
The question now if where should I put the esp_netif_create_ip6_linklocal() call if I cannot in AP_START event?

fariouche, 2020-12-09:
For information, doing the create_ip6_linklocal 2s after the AP_START event works (the 2s must be asynchronous, so no task_delay inside the AP_START!) I suspect that something is done after the AP_START event returns.

david-cermak, 2020-12-10:
@fariouche I'd suggest waiting until the related netif is up:
``` c
    ESP_LOGI(TAG, "ESP_WIFI_MODE_AP");
    wifi_init_softap();
    while (!esp_netif_is_netif_up(g_ap_netif)) {
        vTaskDelay(100/ portTICK_PERIOD_MS);
    }
    esp_err_t err = esp_netif_create_ip6_linklocal(g_ap_netif);
    ESP_LOGI(TAG, "esp_netif_create_ip6_linklocal returned %d", err);
```
Unfortunately there is no event to notify about certain netif going up/down.

I might work in the handler of the `WIFI_EVENT_AP_START` event, too, but need to register the handler *after* wifi system handler gets registered in `esp_netif_create_default_wifi_ap()`. Then when the event comes it first starts the netif, queues it into the lwip task, and after that we create local link address again enqueueing into lwip, then the lwip task pre-empts the main task setting the netif up first and then creating local link IPv6 address.
I'd still suggest the wait loop above in the main task as it's I think easier, more readable, robust and less error prone.

fariouche, 2020-12-10:
Thank you

Blocking the WIFI_EVENT_AP_START will not work (if I do the vtask_delay inside, the netif is never up)
So I suspect that the netif is set up in the same task that handle the AP_START event.

The wait loop you suggest may work, but in my case I will have to create a task just for that to not block the app.

It would have been nice to have a NETIF_UP event (there is one for the ethernet interface)

david-cermak, 2020-12-10:
> So I suspect that the netif is set up in the same task that handle the AP_START event.
It's not executed in the same task, but queues both action to the same time, which has the same effect. Here however, we could control the order of the handlers, so having `esp_netif_create_default_wifi_ap()` run before `esp_event_handler_instance_register(..., esp_event_handler_instance_register, ..)` should work, even with adding the `vTaskDelay()` directly into the handler (before creating local ink address). This would be the same as the wait loop, but would effectively **block** the entire system not only the app.

> It would have been nice to have a NETIF_UP event (there is one for the ethernet interface)
Do you mean `ETHERNET_EVENT_CONNECTED` event?

I general i think it would be nice to have such an event.

fariouche, 2020-12-11:
For the vtask_delay in the event handler, I've actually already tried it (put 10s) and it didn't work.
For now I'm using my timer task to queue a timed event to myself when I receive the event AP_START (1s seems to work)

For the ETHER_EVENT_CONNECTED, for netif, a possible solution is maybe fire up the AP_START only after the netif is also up.

This will solve the issue I believe (and maybe raise regressions?)

Alvin1Zhang, 2021-08-23:
Thanks for reporting and sharing updates, will close the ticket now, feel free to reopen.

closed 2021-08-23 by Alvin1Zhang