espressif/esp-idf #5808: mesh: WiFi stack is freezed (on root node) for a while when child node is disconnected

dshil, 2020-08-29:
## Environment

- Module or chip used:  ESP32-WROVER-B
- IDF version: v4.0.1
- Build System: idf.py (cmake - the same)
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Linux
- Using an IDE: No
- Power Supply: USB

----

## Problem Description

I played a little bit with various examples provided under `examples/mesh` folder and found a strange behavior which firstly occurred in my real application. The issue is that each time any child node is disconnected from the mesh network I have some kind of `stop the world` situation on root node. During this "stop the world" phase all tasks in the system are frozen, there is zero log output. Then after some amount of time WiFi stack will alive.

I found that the duration of this phase is near about the time I set in:

```cpp
    ESP_ERROR_CHECK(esp_mesh_set_ap_assoc_expire(60));
```

I am curious in 2 things:

- Why it is happened?
- Is it possible to avoid such stop the world phase since I need to more quickly react on connect/disconnect events?

----

Moreover, another annoying thing is that when WiFi stack is frozen my connection to the MQTT broker is aborted. Reconnect to MQTT broker is toop expensive operation and I want to avoid it as mush as possible. I prepare the simple solution for MQTT library, introduced in https://github.com/espressif/esp-mqtt/pull/166 (hope @david-cermak will have time to take a look).

cc: @Alvin1Zhang @Jiangyafeng

Alvin1Zhang, 2020-08-31:
Thanks for reporting, we will look into.

dshil, 2020-10-23:
@Alvin1Zhang any updates? 

It is very critical for us because now it is only possible to detect node disconnect after 10sec but for us it is too late. It would be perfect to be able to detect network changes more frequently (1 seconds or event ms granularity).

Probably I missed some configuration but I think this issue is because of: ```esp_mesh_set_ap_assoc_expire``` which is by default 10 seconds. I tried to set it less then 10sec but it did not solve the problem, even after it I will be notified about node connection/disconnection only after 10sec.

Probably somebody from team that is responsible for mesh component could elaborate on it?

dshil, 2021-03-29:
Since there is no update for the issue more then half of the year I think it is better to close it.
The issue is still with us and from time to time we can have the random freeze when a node disconnect is happened.

Ref: https://github.com/espressif/esp-idf/issues/3472

closed 2021-03-29 by dshil

shenjun7, 2021-05-11:
We add an API esp_mesh_send_block_time() to fix the issue that root node is frozen for a while when child node is disconnected. We have to merge it to master first and then backport to the release/v4.0.

VitorBFreitas, 2023-04-20:
Can someone explain to me what exactly this API esp_mesh_send_block_time() mean? It's define a timeout waiting some kind of ACK, blocking caller during this time?
Why this block is necessary?
I couldn't find a good explanation anywhere.
Thanks in advance.