espressif/esp-idf #5809: mesh: allow to use external memory

dshil, 2020-08-29:
**Is your feature request related to a problem? Please describe.**

Since RAM capacity is quite limited having a lot of nodes in mesh network using only internal memory is quite challenging. I try to use internal memory only for components that can't work without it and move all non-critical components to external memory. 

**Describe the solution you'd like**

NimBLE, mbedTLS and others are allowed to easily replace the memory allocation policy by a single configuration option. I would be happy to have the possibility to use external memory in mesh if it is possible from the implementation point of view.

NimBLE uses the following workflow (as far as I remember mbedTLS uses the same):

```c
IRAM_ATTR void *nimble_platform_mem_malloc(size_t size)
{
#ifdef CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_INTERNAL
    return heap_caps_malloc(size, MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT);
#elif CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL
    return heap_caps_malloc(size, MALLOC_CAP_SPIRAM|MALLOC_CAP_8BIT);
#elif CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_IRAM_8BIT
    return heap_caps_malloc_prefer(size, 2, MALLOC_CAP_INTERNAL|MALLOC_CAP_IRAM_8BIT, MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT);
#else
    return malloc(size);
#endif
}

IRAM_ATTR void *nimble_platform_mem_calloc(size_t n, size_t size)
{
#ifdef CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_INTERNAL
    return heap_caps_calloc(n, size, MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT);
#elif CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL
    return heap_caps_calloc(n, size, MALLOC_CAP_SPIRAM|MALLOC_CAP_8BIT);
#elif CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_IRAM_8BIT
    return heap_caps_calloc_prefer(n, size, 2, MALLOC_CAP_INTERNAL|MALLOC_CAP_IRAM_8BIT, MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT);
#else
    return calloc(n, size);
#endif
}

IRAM_ATTR void nimble_platform_mem_free(void *ptr)
{
    heap_caps_free(ptr);
}
```

Alvin1Zhang, 2020-08-31:
Thanks for raising the feature request.

dshil, 2020-11-12:
Hi @Alvin1Zhang, any progress?

dshil, 2021-03-29:
Since there is no update for the issue I think it is better to close it.

closed 2021-03-29 by dshil