espressif/esp-idf #5810: TCP & ICMP Fail, UDP and WiFi still connected

gadget-man, 2020-08-29:
- Module or chip used:  ESP32-WROVER-IB
- IDF version v3.4-r5
- Build System:         Mongoose-OS 
- Operating System:     macOS
- Using an IDE?: Yes - Mongoose-OS using VSCode
- Power Supply:         External 5V

Problem Description:
I have approx. 200 devices deployed in production, all with identical (custom) PCB using the Wrover-IB module. Of these, 5-8 devices consistently have an issue where they very regularly disconnect from MQTT / TCP connections - usually at a pretty regular interval.  The other 190ish devices are very stable and disconnect once every 2-3 days (probably due to the local environment)

What is particularly strange about the problem is that when the device disconnects, I continue to be able to see full logging over UDP (including the ESP_LOG wifi: debug logging). However ICMP ping to the local router or a DNS server starts to timeout, and shortly thereafter an MQTT ping also fails. The only way to recover is to force a reboot of the device (via an MQTT watchdog timer), and on reboot the device will immediately reconnect.

Here is an example output from a graphical dashboard showing the frequency of disconnects for this particular device:
[image]

Here is an output from the log showing the disconnect event until my timeout eventually forces a reboot - note there are no wifi: debugs logs shown here, but I know the wifi: logging is working as at 19:08 I can see a `wifi:obss scan is disabled` in my logs.
https://pastebin.com/qUGqta5a

What could be causing my device to drop ICMP and TCP connection even though UDP is still active (it's a remote device so my only way to get the logs is via remote UDP logging)?

Alvin1Zhang, 2020-08-31:
Thanks for reporting, and sorry for the inconvenience, we will look into.

david-cermak, 2020-09-09:
Hi @gadget-man 

What I can see from the logs is that the UDP packets are delivered inside the local network, but the ICMP to `8.8.8.8` timeouts and resolving of a AWS host name fails, which just seems like the connection to your local network works okay, but fails to route any further. I think you can easily track the packets from ESP32 on the `192.168.1.254` (your local gateway I assume) to see if the pings to 8.8.8.8 get delivered to your local network at lease and if ICMP replies to those packet are available (the same for DNS queries)

gadget-man, 2020-09-09:
Thanks. As a test, I also tried changing the ping from 8.8.8.8 to the IP address of the local gateway, the same issue occurs, so the ping will also fail when trying to ping the local router.

david-cermak, 2020-09-09:
Ah, okay. In that case, could you please check for the packets and the replies on the local network? And at the same time check if the replies are delivered to the ESP32. You can enable IP and ICMP debugging in the lwip by setting:
``` c
#define IP_DEBUG                        LWIP_DBG_ON
#define ICMP_DEBUG                      LWIP_DBG_ON
```

You could see the logs similar to the below:
```
ip_input: iphdr->dest 0x100000a netif->ip_addr 0x100000a (0xa, 0xa, 0x1000000)0a-00-00-02-0a-00-00-01-08-00-f7-fd-00-01-00-01
ip4_input: packet accepted on interface st
ip4_input:
IP header:
+-------------------------------+
| 4 | 5 |  0x00 |        28     | (v, hl, tos, len)
+-------------------------------+
|        0      |000|       0   | (id, flags, offset)
+-------------------------------+
|  255  |    1  |    0xa7de     | (ttl, proto, chksum)
+-------------------------------+
|   10  |    0  |    0  |    2  | (src)
+-------------------------------+
|   10  |    0  |    0  |    1  | (dest)
+-------------------------------+
ip4_input: p->len 28 p->tot_len 28
icmp_input: ping
```
Which might help indicate on which level the packet was dropped (if received)

The only guess at this point on what might cause dropping some packets while processing others (plain UDP) would be shortage of heap? Could you please check for that as well?

Also could you please share the version of IDF you use, the ` v3.4-r5` is not a recognized version.

gadget-man, 2020-09-09:
OK thanks. I'll try and turn on the additional debugging and also do some logging of available heap to see if it gives any clues.

Regards the IDF - I'm using mongoose-os which uses a seperate fork of the ESP-IDF here: https://github.com/cesanta/esp-idf. Last commit was https://github.com/cesanta/esp-idf/commit/fb7ba1baa068a2267ef54b7105cf3c8a004fe4eb

gadget-man, 2020-09-09:
Question - bearing in mind I'm using Wrover-IB with 4mb SPIRAM, should I be reporting on any particular memory capability (e.g. does the ICMP call use only internal ram?)

david-cermak, 2020-09-09:
> should I be reporting on any particular memory capability

This depends on the configuration. You can quickly look at `components/lwip/lwip/src/core/mem.c` for a definition of `mem_malloc()`, this depends on config values such as `CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST` (may be called differently in different idf versions)

gadget-man, 2020-09-09:
Thanks. I am indeed using `CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST`. I'll report on both `MALLOC_CAP_INTERNAL` and `MALLOC_CAP_SPIRAM` whenever I get a ICMP ping timeout and see whether it gives any clues.

gadget-man, 2020-09-10:
I've had clarification from the team at mongoose-os, they are using https://github.com/mongoose-os/esp-idf/commits/3.3-r6 - it's based on espressif/esp-idf@bf02206 with patches on top

gadget-man, 2020-09-11:
@david-cermak I haven't yet managed to get the ICMP debug turned on as it causes a crash loop when the mongoose-os UDP logging starts up before wifi is connected, however I have managed to check the heap when the ping timeout starts:

`[Sep 11 09:18:57.889] iParcelBox-30AEA4C6F1C8 3091 12260.507 2 2|pn_ping.c:44            PING From 8.8.8.8 icmp_seq=1226 timeout
[Sep 11 09:18:57.900] iParcelBox-30AEA4C6F1C8 3092 12260.525 2 2|pn_ping.c:47            MemReport: intenal_free: 157784, internal_min_free: 129112, internal_free_block: 49544, psram_free: 4157328, psram_min_free: 4126476, psram_free_block: 4124560

[Sep 11 09:19:05.949] iParcelBox-30AEA4C6F1C8 3093 12268.516 2 3|mg_mqtt.c:192           Send PINGREQ
[Sep 11 09:19:07.871] iParcelBox-30AEA4C6F1C8 3094 12270.507 2 2|pn_ping.c:44            PING From 8.8.8.8 icmp_seq=1227 timeout
[Sep 11 09:19:07.884] iParcelBox-30AEA4C6F1C8 3095 12270.525 2 2|pn_ping.c:47            MemReport: intenal_free: 157784, internal_min_free: 129112, internal_free_block: 49544, psram_free: 4155764, psram_min_free: 4126476, psram_free_block: 4124560`

I'm capturing both `MALLOC_CAP_INTERNAL` and `MALLOC_CAP_SPIRAM`.  To me, this looks as though it isn't a heap problem?

david-cermak, 2020-09-17:
@gadget-man Agree, that it looks like this is not a heap problem.

gadget-man, 2020-10-16:
@david-cermak  is there anything else we can do to try to get to the bottom of this problem - it is now occurring on further devices as more customers connect their products to the network.

gadget-man, 2020-10-17:
Here's a further log from another device - log captured remotely over UDP, showing similar behaviour:

`[Oct 17 08:46:44.790] iParcelBox-30AEA4C6F1C8 1557 705.332 2 2|pn_ping.c:36            PING 64 bytes from 192.168.0.1 icmp_seq=69 ttl=64 time=1 ms
[Oct 17 08:46:51.780] iParcelBox-30AEA4C6F1C8 1558 712.327 2 3|mg_mqtt.c:192           Send PINGREQ
[Oct 17 08:46:51.780] iParcelBox-30AEA4C6F1C8 1559 712.366 2 3|mg_mqtt.c:174           Recv PINGRESP
[Oct 17 08:46:51.780] iParcelBox-30AEA4C6F1C8 1560 712.372 2 3|mgos_mqtt_conn.c:219    MQTT0 event: 213
[Oct 17 08:46:54.751] iParcelBox-30AEA4C6F1C8 1561 715.332 2 2|pn_ping.c:36            PING 64 bytes from 192.168.0.1 icmp_seq=70 ttl=64 time=2 ms
[Oct 17 08:47:04.783] iParcelBox-30AEA4C6F1C8 1562 725.334 2 2|pn_ping.c:36            PING 64 bytes from 192.168.0.1 icmp_seq=71 ttl=64 time=4 ms
[Oct 17 08:47:06.016] iParcelBox-30AEA4C6F1C8 1563 726.297 2 2|D (726452) wifi:obss scan is disabled
[Oct 17 08:47:06.016] iParcelBox-30AEA4C6F1C8 1564 726.302 2 2|D (726452) wifi:start obss scan: obss scan is stopped
[Oct 17 08:47:14.818] iParcelBox-30AEA4C6F1C8 1565 735.333 2 2|pn_ping.c:36            PING 64 bytes from 192.168.0.1 icmp_seq=72 ttl=64 time=3 ms
[Oct 17 08:47:25.774] iParcelBox-30AEA4C6F1C8 1566 746.327 2 2|pn_ping.c:45            PING From 192.168.0.1 icmp_seq=73 timeout
[Oct 17 08:47:25.774] iParcelBox-30AEA4C6F1C8 1567 746.346 2 2|pn_ping.c:48            MemReport: intenal_free: 150056, internal_min_free: 115312, internal_free_block: 38100, psram_free: 4156540, psram_min_free: 4126236, psram_free_block: 4127996
[Oct 17 08:47:35.718] iParcelBox-30AEA4C6F1C8 1568 756.327 2 2|pn_ping.c:45            PING From 192.168.0.1 icmp_seq=74 timeout
[Oct 17 08:47:35.735] iParcelBox-30AEA4C6F1C8 1569 756.345 2 2|pn_ping.c:48            MemReport: intenal_free: 150056, internal_min_free: 115312, internal_free_block: 38100, psram_free: 4156540, psram_min_free: 4126236, psram_free_block: 4127996
[Oct 17 08:47:45.847] iParcelBox-30AEA4C6F1C8 1570 766.327 2 2|pn_ping.c:45            PING From 192.168.0.1 icmp_seq=75 timeout
[Oct 17 08:47:45.847] iParcelBox-30AEA4C6F1C8 1571 766.346 2 2|pn_ping.c:48            MemReport: intenal_free: 150056, internal_min_free: 115312, internal_free_block: 38100, psram_free: 4156540, psram_min_free: 4126236, psram_free_block: 4127996
[Oct 17 08:47:51.784] iParcelBox-30AEA4C6F1C8 1572 772.337 2 3|mg_mqtt.c:192           Send PINGREQ
[Oct 17 08:47:55.780] iParcelBox-30AEA4C6F1C8 1573 776.327 2 2|pn_ping.c:45            PING From 192.168.0.1 icmp_seq=76 timeout
[Oct 17 08:47:55.780] iParcelBox-30AEA4C6F1C8 1574 776.346 2 2|pn_ping.c:48            MemReport: intenal_free: 150056, internal_min_free: 115312, internal_free_block: 38100, psram_free: 4154980, psram_min_free: 4126236, psram_free_block: 4127996
[Oct 17 08:48:05.814] iParcelBox-30AEA4C6F1C8 1575 786.327 2 2|pn_ping.c:45            PING From 192.168.0.1 icmp_seq=77 timeout
[Oct 17 08:48:05.814] iParcelBox-30AEA4C6F1C8 1576 786.346 2 2|pn_ping.c:48            MemReport: intenal_free: 150056, internal_min_free: 115312, internal_free_block: 38100, psram_free: 4154980, psram_min_free: 4126236, psram_free_block: 4127996
[Oct 17 08:48:15.721] iParcelBox-30AEA4C6F1C8 1577 796.327 2 2|pn_ping.c:45            PING From 192.168.0.1 icmp_seq=78 timeout
[Oct 17 08:48:15.736] iParcelBox-30AEA4C6F1C8 1578 796.345 2 2|pn_ping.c:48            MemReport: intenal_free: 150056, internal_min_free: 115312, internal_free_block: 38100, psram_free: 4154980, psram_min_free: 4126236, psram_free_block: 4127996
[Oct 17 08:48:25.785] iParcelBox-30AEA4C6F1C8 1579 806.327 2 2|pn_ping.c:45            PING From 192.168.0.1 icmp_seq=79 timeout
[Oct 17 08:48:25.986] iParcelBox-30AEA4C6F1C8 1580 806.347 2 2|pn_ping.c:48            MemReport: intenal_free: 150056, internal_min_free: 115312, internal_free_block: 38100, psram_free: 4154980, psram_min_free: 4126236, psram_free_block: 4127996
[Oct 17 08:48:35.816] iParcelBox-30AEA4C6F1C8 1581 816.327 2 2|pn_ping.c:45            PING From 192.168.0.1 icmp_seq=80 timeout
[Oct 17 08:48:35.817] iParcelBox-30AEA4C6F1C8 1582 816.346 2 2|pn_ping.c:48            MemReport: intenal_free: 150056, internal_min_free: 115312, internal_free_block: 38100, psram_free: 4154980, psram_min_free: 4126236, psram_free_block: 4127996
[Oct 17 08:48:45.750] iParcelBox-30AEA4C6F1C8 1583 826.327 2 2|pn_ping.c:45            PING From 192.168.0.1 icmp_seq=81 timeout
[Oct 17 08:48:45.750] iParcelBox-30AEA4C6F1C8 1584 826.346 2 2|pn_ping.c:48            MemReport: intenal_free: 150056, internal_min_free: 115312, internal_free_block: 38100, psram_free: 4154980, psram_min_free: 4126236, psram_free_block: 4127996
[Oct 17 08:48:52.103] iParcelBox-30AEA4C6F1C8 1585 832.612 2 0|mg_ssl_if_mbedtls.c:30  0x3ffe7b44 mbedtls_ssl_flush_output() returned -78 (-0x004e)
[Oct 17 08:48:52.103] iParcelBox-30AEA4C6F1C8 1586 832.621 2 0|mg_ssl_if_mbedtls.c:30  0x3ffe7b44 mbedtls_ssl_write_record() returned -78 (-0x004e)
[Oct 17 08:48:52.103] iParcelBox-30AEA4C6F1C8 1587 832.630 2 0|mg_ssl_if_mbedtls.c:30  0x3ffe7b44 mbedtls_ssl_send_alert_message() returned -78 (-0x004e)
[Oct 17 08:48:52.104] iParcelBox-30AEA4C6F1C8 1588 832.635 2 2|mgos_mqtt_conn.c:253    MQTT0 Disconnect
[Oct 17 08:48:52.104] iParcelBox-30AEA4C6F1C8 1589 832.641 2 3|mgos_event.c:135        ev MOS5 triggered 1 handlers
[Oct 17 08:48:52.104] iParcelBox-30AEA4C6F1C8 1590 832.647 2 3|mgos_event.c:135        ev MOS5 triggered 1 handlers
[... cut ...]
[Oct 17 08:50:05.828] iParcelBox-30AEA4C6F1C8 1751 906.345 2 2|pn_ping.c:48            MemReport: intenal_free: 146528, internal_min_free: 115312, internal_free_block: 38100, psram_free: 4156668, psram_min_free: 4126236, psram_free_block: 4127996
[Oct 17 08:50:08.796] iParcelBox-30AEA4C6F1C8 1752 909.331 2 0|mg_resolv.c:208         Failed to resolve 'a24buxh69lukw9-ats.iot.eu-west-1.amazonaws.com', server 192.168.0.1
[Oct 17 08:50:08.796] iParcelBox-30AEA4C6F1C8 1753 909.337 2 2|mgos_mqtt_conn.c:227    MQTT0 TCP connect error (-1)
[Oct 17 08:50:08.796] iParcelBox-30AEA4C6F1C8 1754 909.342 2 2|mgos_mqtt_conn.c:253    MQTT0 Disconnect
`

gadget-man, 2020-10-17:
[Oct 17 08:46:44.790] iParcelBox-30AEA4C6F1C8 1557 705.332 2 2|pn_ping.c:36            PING 64 bytes from 192.168.0.1 icmp_seq=69 ttl=64 time=1 ms
[Oct 17 08:46:51.780] iParcelBox-30AEA4C6F1C8 1558 712.327 2 3|mg_mqtt.c:192           Send PINGREQ
[Oct 17 08:46:51.780] iParcelBox-30AEA4C6F1C8 1559 712.366 2 3|mg_mqtt.c:174           Recv PINGRESP
[Oct 17 08:46:51.780] iParcelBox-30AEA4C6F1C8 1560 712.372 2 3|mgos_mqtt_conn.c:219    MQTT0 event: 213
[Oct 17 08:46:54.751] iParcelBox-30AEA4C6F1C8 1561 715.332 2 2|pn_ping.c:36            PING 64 bytes from 192.168.0.1 icmp_seq=70 ttl=64 time=2 ms
[Oct 17 08:47:04.783] iParcelBox-30AEA4C6F1C8 1562 725.334 2 2|pn_ping.c:36            PING 64 bytes from 192.168.0.1 icmp_seq=71 ttl=64 time=4 ms
[Oct 17 08:47:06.016] iParcelBox-30AEA4C6F1C8 1563 726.297 2 2|D (726452) wifi:obss scan is disabled
[Oct 17 08:47:06.016] iParcelBox-30AEA4C6F1C8 1564 726.302 2 2|D (726452) wifi:start obss scan: obss scan is stopped
[Oct 17 08:47:14.818] iParcelBox-30AEA4C6F1C8 1565 735.333 2 2|pn_ping.c:36            PING 64 bytes from 192.168.0.1 icmp_seq=72 ttl=64 time=3 ms
[Oct 17 08:47:25.774] iParcelBox-30AEA4C6F1C8 1566 746.327 2 2|pn_ping.c:45            PING From 192.168.0.1 icmp_seq=73 timeout
[Oct 17 08:47:25.774] iParcelBox-30AEA4C6F1C8 1567 746.346 2 2|pn_ping.c:48            MemReport: intenal_free: 150056, internal_min_free: 115312, internal_free_block: 38100, psram_free: 4156540, psram_min_free: 4126236, psram_free_block: 4127996
[Oct 17 08:47:35.718] iParcelBox-30AEA4C6F1C8 1568 756.327 2 2|pn_ping.c:45            PING From 192.168.0.1 icmp_seq=74 timeout
[Oct 17 08:47:35.735] iParcelBox-30AEA4C6F1C8 1569 756.345 2 2|pn_ping.c:48            MemReport: intenal_free: 150056, internal_min_free: 115312, internal_free_block: 38100, psram_free: 4156540, psram_min_free: 4126236, psram_free_block: 4127996
[Oct 17 08:47:45.847] iParcelBox-30AEA4C6F1C8 1570 766.327 2 2|pn_ping.c:45            PING From 192.168.0.1 icmp_seq=75 timeout
[Oct 17 08:47:45.847] iParcelBox-30AEA4C6F1C8 1571 766.346 2 2|pn_ping.c:48            MemReport: intenal_free: 150056, internal_min_free: 115312, internal_free_block: 38100, psram_free: 4156540, psram_min_free: 4126236, psram_free_block: 4127996
[Oct 17 08:47:51.784] iParcelBox-30AEA4C6F1C8 1572 772.337 2 3|mg_mqtt.c:192           Send PINGREQ
[Oct 17 08:47:55.780] iParcelBox-30AEA4C6F1C8 1573 776.327 2 2|pn_ping.c:45            PING From 192.168.0.1 icmp_seq=76 timeout
[Oct 17 08:47:55.780] iParcelBox-30AEA4C6F1C8 1574 776.346 2 2|pn_ping.c:48            MemReport: intenal_free: 150056, internal_min_free: 115312, internal_free_block: 38100, psram_free: 4154980, psram_min_free: 4126236, psram_free_block: 4127996
[Oct 17 08:48:05.814] iParcelBox-30AEA4C6F1C8 1575 786.327 2 2|pn_ping.c:45            PING From 192.168.0.1 icmp_seq=77 timeout
[Oct 17 08:48:05.814] iParcelBox-30AEA4C6F1C8 1576 786.346 2 2|pn_ping.c:48            MemReport: intenal_free: 150056, internal_min_free: 115312, internal_free_block: 38100, psram_free: 4154980, psram_min_free: 4126236, psram_free_block: 4127996
[Oct 17 08:48:15.721] iParcelBox-30AEA4C6F1C8 1577 796.327 2 2|pn_ping.c:45            PING From 192.168.0.1 icmp_seq=78 timeout
[Oct 17 08:48:15.736] iParcelBox-30AEA4C6F1C8 1578 796.345 2 2|pn_ping.c:48            MemReport: intenal_free: 150056, internal_min_free: 115312, internal_free_block: 38100, psram_free: 4154980, psram_min_free: 4126236, psram_free_block: 4127996
[Oct 17 08:48:25.785] iParcelBox-30AEA4C6F1C8 1579 806.327 2 2|pn_ping.c:45            PING From 192.168.0.1 icmp_seq=79 timeout
[Oct 17 08:48:25.986] iParcelBox-30AEA4C6F1C8 1580 806.347 2 2|pn_ping.c:48            MemReport: intenal_free: 150056, internal_min_free: 115312, internal_free_block: 38100, psram_free: 4154980, psram_min_free: 4126236, psram_free_block: 4127996
[Oct 17 08:48:35.816] iParcelBox-30AEA4C6F1C8 1581 816.327 2 2|pn_ping.c:45            PING From 192.168.0.1 icmp_seq=80 timeout
[Oct 17 08:48:35.817] iParcelBox-30AEA4C6F1C8 1582 816.346 2 2|pn_ping.c:48            MemReport: intenal_free: 150056, internal_min_free: 115312, internal_free_block: 38100, psram_free: 4154980, psram_min_free: 4126236, psram_free_block: 4127996
[Oct 17 08:48:45.750] iParcelBox-30AEA4C6F1C8 1583 826.327 2 2|pn_ping.c:45            PING From 192.168.0.1 icmp_seq=81 timeout
[Oct 17 08:48:45.750] iParcelBox-30AEA4C6F1C8 1584 826.346 2 2|pn_ping.c:48            MemReport: intenal_free: 150056, internal_min_free: 115312, internal_free_block: 38100, psram_free: 4154980, psram_min_free: 4126236, psram_free_block: 4127996
[Oct 17 08:48:52.103] iParcelBox-30AEA4C6F1C8 1585 832.612 2 0|mg_ssl_if_mbedtls.c:30  0x3ffe7b44 mbedtls_ssl_flush_output() returned -78 (-0x004e)
[Oct 17 08:48:52.103] iParcelBox-30AEA4C6F1C8 1586 832.621 2 0|mg_ssl_if_mbedtls.c:30  0x3ffe7b44 mbedtls_ssl_write_record() returned -78 (-0x004e)
[Oct 17 08:48:52.103] iParcelBox-30AEA4C6F1C8 1587 832.630 2 0|mg_ssl_if_mbedtls.c:30  0x3ffe7b44 mbedtls_ssl_send_alert_message() returned -78 (-0x004e)
[Oct 17 08:48:52.104] iParcelBox-30AEA4C6F1C8 1588 832.635 2 2|mgos_mqtt_conn.c:253    MQTT0 Disconnect
[Oct 17 08:48:52.104] iParcelBox-30AEA4C6F1C8 1589 832.641 2 3|mgos_event.c:135        ev MOS5 triggered 1 handlers
[Oct 17 08:48:52.104] iParcelBox-30AEA4C6F1C8 1590 832.647 2 3|mgos_event.c:135        ev MOS5 triggered 1 handlers
[Oct 17 08:48:52.104] iParcelBox-30AEA4C6F1C8 1591 832.662 2 2|init.js:468             MQTT Disconnect
[... cut ...]
[Oct 17 08:50:05.828] iParcelBox-30AEA4C6F1C8 1751 906.345 2 2|pn_ping.c:48            MemReport: intenal_free: 146528, internal_min_free: 115312, internal_free_block: 38100, psram_free: 4156668, psram_min_free: 4126236, psram_free_block: 4127996
[Oct 17 08:50:08.796] iParcelBox-30AEA4C6F1C8 1752 909.331 2 0|mg_resolv.c:208         Failed to resolve 'a24buxh69lukw9-ats.iot.eu-west-1.amazonaws.com', server 192.168.0.1
[Oct 17 08:50:08.796] iParcelBox-30AEA4C6F1C8 1753 909.337 2 2|mgos_mqtt_conn.c:227    MQTT0 TCP connect error (-1)
[Oct 17 08:50:08.796] iParcelBox-30AEA4C6F1C8 1754 909.342 2 2|mgos_mqtt_conn.c:253    MQTT0 Disconnect

gadget-man, 2020-10-17:
So it looks as though it's set off soon after `wifi:obss scan is disabled`

david-cermak, 2021-07-09:
@gadget-man I'm sorry for not responding. Is there any update to this issue?
Thanks for posting the new log, but this only shows what you've shared previously. It would be really helpful to provide some more detailed info with lwip debugging on, as requested [above](https://github.com/espressif/esp-idf/issues/5810#issuecomment-689514097).

> So it looks as though it's set off soon after wifi:obss scan is disabled

I think this is unrelated, but will check with the wifi team.

gadget-man, 2021-07-09:
We are still seeing this issue on a small number of devices. Interesting, in nearly every case the customer also has an LG TV connected to their same network, either via WiFi or an ethernet connection. To date we haven't been able to find a customer who is technical enough to be able to capture Wireshark updates. I'll try again to see if I can turn on lip debugging.

xueyunfei998, 2021-08-23:
hi @gadget-man 

Do you mean that if there is an LG TV device in the network, the Ping Service will fail?

If it is a problem with the LG device, you can remove the LG device from the network to see if the problem can be reproduced？

You can build a testing environment and capture packets with Wireshark to reproduce this problem.

gadget-man, 2021-08-23:
We aren't certain it's the cause, but nearly every customer who has experienced intermittent disconnections on ESP also has an LG television. Turning the power off to the TV stops the connection issues. It is only ICMP Ping and TCP that is affected - even on a 'disconnected' device I am still able to receive logging info via UDP.  Rebooting the devices causes it to immediately reconnect.

xueyunfei998, 2021-09-14:
hi @gadget-man 
Because this problem requires packet capture analysis, but the current conditions do not allow it, can you upgrade the IDF version to 4.1 to see if this problem can be solved?

xueyunfei998, 2021-09-14:
hi @gadget-man 

Provide the model of LG TV, and we purchase an environment test internally

Alvin1Zhang, 2021-11-17:
@gadget-man Thanks for reporting and sharing the updates, feel free to reopen if the issue happens again.

closed 2021-11-17 by Alvin1Zhang