espressif/esp-idf #5813: UART read putting garbage in my buffer

mzimmers, 2020-08-31:
1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [|none]

- Module or chip used:  [ESP32-WROVER]
- IDF version (run ``git describe --tags`` to find it):v4.0.1

- Build System:         [||idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
    
- Operating System:     [Windows||]
- (Windows only) environment type: [MSYS2 |ESP Command Prompt|].
- Using an IDE?: [No|]
- Power Supply:         [USB|]

## Problem Description

Very repeatable problem - after a few "correct UART reads, a read inserts several bytes of garbage in my input buffer. I've used a serial monitor to verify that it's not coming from the sender.

// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
0x0000:  3C.3F.78.6D.6C.20.76.65 72.73.69.6F.6E.3D.22.31 |  <?xml version="1
0x0010:  2E.30.22.20.65.6E.63.6F 64.69.6E.67.3D.22.55.54 |  .0" encoding="UT
0x0020:  46.2D.38.22.3F.3E.0A.3C 57.69.66.69.42.75.74.74 |  F-8"?>.<WifiButt
0x0030:  6F.6E.4D.65.73.73.61.67 65.3E.0A.3C.49.50.56.34 |  onMessage>.<IPV4
0x0040:  41.64.64.72.65.73.73.3E 31.39.32.2E.31.36.38.2E |  Address>192.168.
0x0050:  38.36.2E.38.33.3C.2F.49 50.56.34.41.64.64.72.65 |  86.83</IPV4Addre
0x0060:  73.73.3E.0A.3C.4D.41.43                         |  ss>.<MAC
I (39896) UartMgr: read(): attempting to read in 112 bytes from SIO input buffer.
I (39906) UartMgr: read(): 112 bytes read into m_buffRx.
I (39906) UartMgr: read(): buffer just read is:

0x0000:  41.64.64.72.65.73.73.3E 30.30.3A.32.30.3A.66.37 |  Address>00:20:f7
0x0010:  3A.30.34.3A.36.31.3A.64 30.3C.2F.4D.41.43.41.64 |  :04:61:d0</MACAd
0x0020:  64.72.65.73.73.3E.0A.3C 50.61.63.6B.65.74.54.79 |  dress>.<PacketTy
0x0030:  70.65.3E.52.65.73.74.61 72.74.20.52.65.71.75.65 |  pe>Restart Reque
0x0040:  73.74.3C.2F.50.61.63.6B 65.74.54.79.70.65.3E.0A |  st</PacketType>.
0x0050:  3C.50.72.6F.64.75.63.74 0E.BB.39.EE.56.CF.84.63 |  <Product..9.V..c
0x0060:  84.F9.EE.63.63.54.BD.5E 75.53.B9.CF.BD.37.42.FF |  ...ccT.^uS...7B.
I (39976) UartMgr: read(): performing hex dump on m_buffRx.

I (40096) UartMgr: uartEventTask(): read() returned 112 bytes.
E (40116) UartMgr: uartEventTask(): uart frame error.

```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

[uartmgr.cpp.txt](https://github.com/espressif/esp-idf/files/5147841/uartmgr.cpp.txt)

Alvin1Zhang, 2020-08-31:
Thanks for reporting and letting us know, we will look into.

Ethan-CYQ, 2020-08-31:
Hi, @mzimmers 
The attached code seem has some parts that is not show your configuration (e.g. uart configuration)and not relate to ESP-IDF(e.g. `cksumValid`, `CHECKSUM`). So I can't reproduce your problem with complete code(I can only use a simple demo about UART that similar as your code). Can you provide more simple code so that we can find out your problem?

mzimmers, 2020-08-31:
Hi Ethan - sorry about that; I was in a bit of a hurry to get that issue reported. Here's the header file.
[uartmgr.h.txt](https://github.com/espressif/esp-idf/files/5150663/uartmgr.h.txt)

mzimmers, 2020-09-01:
[uart.txt](https://github.com/espressif/esp-idf/files/5153597/uart.txt)

This log file shows the (correct) start of a message, and several continuations of that message that get garbled. I hope it helps.

Ethan-CYQ, 2020-09-01:
Hi, @mzimmers 
I tried to reproduce your problem the whole day, but the attached code still have a lot of part that is relate to your project and I don't know whether I could delete them or how to use them.
Can you simplify your code so that I can figure out the detail in your code? I would appreciate if the code could be build by ESP-IDF.

mzimmers, 2020-09-01:
Hi Ethan - here's a complete example.
[input.txt](https://github.com/espressif/esp-idf/files/5156739/input.txt)

You'll need to send these messages to the ESP32 - sorry that they're not decoded.
[uart_min.zip](https://github.com/espressif/esp-idf/files/5156723/uart_min.zip)

mzimmers, 2020-09-01:
I can also provide you with the utility that will automatically send those messages if you like.

Ethan-CYQ, 2020-09-02:
Hi, @mzimmers 

I had run your example and test them.  But I have not find the garbage in my local test. The following is my steps to reproduce your problem.
- Build and flash your example into my `ESP-WROVER-KIT`.
- Write a simple demo that use UART to send data.(The data is from your `input.txt`), and flash the demo into my `ESP32-DevKitC`.
- USB connect the PC with `ESP-WROVER-KIT` and monitor `ESP-WROVER-KIT`.
- `ESP-WROVER-KIT` suppport power to `ESP32-DevKitC`, and connect the `ESP-WROVER-KIT` RX with `ESP-WROVER-KIT` TX.

Here is my debug logs: 
```
0x0000:  3C.3F.78.6D.6C.20.76.65 72.73.69.6F.6E.3D.22.31 |  <?xml version="1 
0x0010:  2E.30.22.20.65.6E.63.6F 64.69.6E.67.3D.22.55.54 |  .0" encoding="UT 
0x0020:  46.2D.38.22.3F.3E.0A.3C 57.69.66.69.42.75.74.74 |  F-8"?>.<WifiButt 
0x0030:  6F.6E.4D.65.73.73.61.67 65.3E.0A.3C.49.50.56.34 |  onMessage>.<IPV4 
0x0040:  41.64.64.72.65.73.73.3E 30.2E.30.2E.30.2E.30.3C |  Address>0.0.0.0< 
0x0050:  2F.49.50.56.34.41.64.64 72.65.73.73.3E.0A.3C.4D |  /IPV4Address>.<M 
0x0060:  41.43.41.64.64.72.65.73 73.3E.30.30.3A.31.31.3A |  ACAddress>00:11: 
0x0070:  32.32.3A.33.33.3A.34.34                         |  22:33:44 
I (1182) UartMgr: read(): performing hex dump on m_buffRx.

0x0000:  3C.3F.78.6D.6C.20.76.65 72.73.69.6F.6E.3D.22.31 |  <?xml version="1 
0x0010:  2E.30.22.20.65.6E.63.6F 64.69.6E.67.3D.22.55.54 |  .0" encoding="UT 
0x0020:  46.2D.38.22.3F.3E.0A.3C 57.69.66.69.42.75.74.74 |  F-8"?>.<WifiButt 
0x0030:  6F.6E.4D.65.73.73.61.67 65.3E.0A.3C.49.50.56.34 |  onMessage>.<IPV4 
0x0040:  41.64.64.72.65.73.73.3E 30.2E.30.2E.30.2E.30.3C |  Address>0.0.0.0< 
0x0050:  2F.49.50.56.34.41.64.64 72.65.73.73.3E.0A.3C.4D |  /IPV4Address>.<M 
0x0060:  41.43.41.64.64.72.65.73 73.3E.30.30.3A.31.31.3A |  ACAddress>00:11: 
0x0070:  32.32.3A.33.33.3A.34.34                         |  22:33:44 
I (1252) UartMgr: uartEventTask(): dumping m_buffRx.

0x0000:  3C.3F.78.6D.6C.20.76.65 72.73.69.6F.6E.3D.22.31 |  <?xml version="1 
0x0010:  2E.30.22.20.65.6E.63.6F 64.69.6E.67.3D.22.55.54 |  .0" encoding="UT 
0x0020:  46.2D.38.22.3F.3E.0A.3C 57.69.66.69.42.75.74.74 |  F-8"?>.<WifiButt 
0x0030:  6F.6E.4D.65.73.73.61.67 65.3E.0A.3C.49.50.56.34 |  onMessage>.<IPV4 
0x0040:  41.64.64.72.65.73.73.3E 30.2E.30.2E.30.2E.30.3C |  Address>0.0.0.0< 
0x0050:  2F.49.50.56.34.41.64.64 72.65.73.73.3E.0A.3C.4D |  /IPV4Address>.<M 
0x0060:  41.43.41.64.64.72.65.73 73.3E.30.30.3A.31.31.3A |  ACAddress>00:11: 
0x0070:  32.32.3A.33.33.3A.34.34                         |  22:33:44 
W (1322) UartMgr: msgComplete(): message is missing XML_ROOT_TAG_END.
I (1322) UartMgr: read(): 165 bytes read into m_buffRx.
I (1332) UartMgr: read(): buffer just read is:

0x0000:  3A.35.35.3C.2F.4D.41.43 41.64.64.72.65.73.73.3E |  :55</MACAddress> 
0x0010:  0A.3C.50.61.63.6B.65.74 54.79.70.65.3E.52.65.73 |  .<PacketType>Res 
0x0020:  74.61.72.74.20.52.65.71 75.65.73.74.3C.2F.50.61 |  tart Request</Pa 
0x0030:  63.6B.65.74.54.79.70.65 3E.0A.3C.50.72.6F.64.75 |  cketType>.<Produ 
0x0040:  63.74.4E.61.6D.65.3E.57 69.46.69.20.42.75.74.74 |  ctName>WiFi Butt 
0x0050:  6F.6E.3C.2F.50.72.6F.64 75.63.74.4E.61.6D.65.3E |  on</ProductName> 
0x0060:  0A.3C.53.65.72.69.61.6C 4E.75.6D.62.65.72.3E.31 |  .<SerialNumber>1 
0x0070:  32.33.34.35.36.37.38.39 3C.2F.53.65.72.69.61.6C |  23456789</Serial 
0x0080:  4E.75.6D.62.65.72.3E.0A 3C.2F.57.69.66.69.42.75 |  Number>.</WifiBu 
0x0090:  74.74.6F.6E.4D.65.73.73 61.67.65.3E.0A.66.66.33 |  ttonMessage>.ff3 
0x00A0:  66.31.62.34.36                                  |  f1b46 
I (1422) UartMgr: read(): performing hex dump on m_buffRx.

0x0000:  3C.3F.78.6D.6C.20.76.65 72.73.69.6F.6E.3D.22.31 |  <?xml version="1 
0x0010:  2E.30.22.20.65.6E.63.6F 64.69.6E.67.3D.22.55.54 |  .0" encoding="UT 
0x0020:  46.2D.38.22.3F.3E.0A.3C 57.69.66.69.42.75.74.74 |  F-8"?>.<WifiButt 
0x0030:  6F.6E.4D.65.73.73.61.67 65.3E.0A.3C.49.50.56.34 |  onMessage>.<IPV4 
0x0040:  41.64.64.72.65.73.73.3E 30.2E.30.2E.30.2E.30.3C |  Address>0.0.0.0< 
0x0050:  2F.49.50.56.34.41.64.64 72.65.73.73.3E.0A.3C.4D |  /IPV4Address>.<M 
0x0060:  41.43.41.64.64.72.65.73 73.3E.30.30.3A.31.31.3A |  ACAddress>00:11: 
0x0070:  32.32.3A.33.33.3A.34.34 3A.35.35.3C.2F.4D.41.43 |  22:33:44:55</MAC 
0x0080:  41.64.64.72.65.73.73.3E 0A.3C.50.61.63.6B.65.74 |  Address>.<Packet 
0x0090:  54.79.70.65.3E.52.65.73 74.61.72.74.20.52.65.71 |  Type>Restart Req 
0x00A0:  75.65.73.74.3C.2F.50.61 63.6B.65.74.54.79.70.65 |  uest</PacketType 
0x00B0:  3E.0A.3C.50.72.6F.64.75 63.74.4E.61.6D.65.3E.57 |  >.<ProductName>W 
0x00C0:  69.46.69.20.42.75.74.74 6F.6E.3C.2F.50.72.6F.64 |  iFi Button</Prod 
0x00D0:  75.63.74.4E.61.6D.65.3E 0A.3C.53.65.72.69.61.6C |  uctName>.<Serial 
0x00E0:  4E.75.6D.62.65.72.3E.31 32.33.34.35.36.37.38.39 |  Number>123456789 
[... 3 lines cut ...]
I (1582) UartMgr: uartEventTask(): dumping m_buffRx.

0x0000:  3C.3F.78.6D.6C.20.76.65 72.73.69.6F.6E.3D.22.31 |  <?xml version="1 
0x0010:  2E.30.22.20.65.6E.63.6F 64.69.6E.67.3D.22.55.54 |  .0" encoding="UT 
0x0020:  46.2D.38.22.3F.3E.0A.3C 57.69.66.69.42.75.74.74 |  F-8"?>.<WifiButt 
0x0030:  6F.6E.4D.65.73.73.61.67 65.3E.0A.3C.49.50.56.34 |  onMessage>.<IPV4 
0x0040:  41.64.64.72.65.73.73.3E 30.2E.30.2E.30.2E.30.3C |  Address>0.0.0.0< 
0x0050:  2F.49.50.56.34.41.64.64 72.65.73.73.3E.0A.3C.4D |  /IPV4Address>.<M 
0x0060:  41.43.41.64.64.72.65.73 73.3E.30.30.3A.31.31.3A |  ACAddress>00:11: 
0x0070:  32.32.3A.33.33.3A.34.34 3A.35.35.3C.2F.4D.41.43 |  22:33:44:55</MAC 
[... cut ...]
0x00F0:  3C.2F.53.65.72.69.61.6C 4E.75.6D.62.65.72.3E.0A |  </SerialNumber>. 
0x0100:  3C.2F.57.69.66.69.42.75 74.74.6F.6E.4D.65.73.73 |  </WifiButtonMess 
0x0110:  61.67.65.3E.0A.66.66.33 66.31.62.34.36          |  age>.ff3f1b46 
I (1732) UartMgr: msgComplete(): message is considered complete with 285 bytes.
I (1742) UartMgr: cksumValid(): message length (without checksum) is 277.
I (1752) UartMgr: readSerial(): outgoing reply is 279 bytes.
I (1752) UartMgr: write(): sending message of 279 bytes to serial port.
```

mzimmers, 2020-09-02:
Hi Ethan -

Thank you for your hard work on this. I believe I have found the problem, and it's not an ESP32 problem -- if the host abruptly disconnects/closes the COM port prior to completing the message sending, this error occurs.

While not an ESP problem, I'm wondering whether there's a way for the ESP32 UART to detect the remote closure as something other than a frame error.

closed 2020-09-02 by mzimmers