espressif/esp-idf #5818: BLE-MESH Sensor Client Example Bug

MrCurio, 2020-09-02:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): v4.2-dev-1660-g7d7521367
- Build System:        Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): ( esp-2020r1) 8.2.0
    // 1.22.0-80-g6c4433a
- Operating System:     Linux
- Power Supply:         USB

## Problem Description
Found Bug in the BLE-MESH example of the Sensor Client Model . The problem appears when 3 (or more) mpids and values are sent from the server to the client. When trying to parse 3 values (or more), the code does not parse good property ids or anything due to a problem in the iteration through the "for loop".

The code provided in the master branch, updates the "data" pointer as follows:
 
`data += length;`

This causes to move more than 3 bytes after the first iteration causing missing data and wrong parsing. Instead that line should be:

`data += mpid_len + data_len + 1;`

This way, the data pointer moves 3 bytes per iteration while the count "length" is updated.

### Expected Behavior
Correct Parsing 
### Actual Behavior
Wrong iteration!

### Steps to reproduce

1. Modify Sensor Server example to send more than 2 values with their property ids.
2. Send those values as a Status Message to a Sensor Client Model and try to parse them.

Alvin1Zhang, 2020-09-02:
Thanks for reporting, we will look into.

Campou, 2020-09-04:
Hi @MrCurio 

The code in the example is:
```
if (data_len != ESP_BLE_MESH_SENSOR_DATA_ZERO_LEN) {
    ESP_LOG_BUFFER_HEX("Sensor Data", data + mpid_len, data_len + 1);
    length += mpid_len + data_len + 1;
} else {
    length += mpid_len;
}
data += length;
```
Code above `data += length` will check if the data length is ZERO and update the `length`value.

Thanks.

MrCurio, 2020-09-04:
Thank you!
Should i close the issue?

Campou, 2020-09-04:
Hi @MrCurio 

Yes, you can close this issue. I have submitted a MR to fix this.

Thanks.

closed 2020-09-04 by MrCurio

deepakdivakar, 2021-07-08:
Can any one please help me how to get datas in sensor mesh in group and what and things to do.
Also after getting the indoor and outdoor values in sensor mesh ,How can i change change the data to desired datas.
Advance Thanks for reply.