espressif/esp-idf #5819: I2S_BCK output with skipped fronts (or unstable pulse duration)

Serjy1710, 2020-09-02:
## Environment

- Development Kit:      ESP32-DevKitC
- Module or chip used:  ESP32-WROOM-32

## Problem Description
When I try esp-idf/examples/peripherals/i2s/main/i2s_example_main.c, I2S_BCK_IO output signal in Logic Analyzer  looks like "missed fronts", so pulse wight changes from "short" to "long", where "long"=2x"short".
I suppose, because of this, i hear no sound in I2S DAC connected.
[image]
I've try other sample frequence (48k) and bits_per_sample values (24, 32), same effect.

What's wrong with BCK? maybe, this is normal behavior?

## Platform details
`PLATFORM: Espressif 32 1.12.4 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40001.200521 (4.0.1)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.190618 (2.28.51)
 - toolchain-xtensa32 2.80200.200226 (8.2.0)`
 
example output:
`ets32mI (40) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (45) boot: SPI Speed      : 40MHz␛[0m
␛[0;32mI (49) boot: SPI Mode       : DIO␛[0m
␛[0;32mI (53) boot: SPI Flash Size : 4MB␛[0m
␛[0;32mI (57) boot: Partition Table:␛[0m
␛[0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length␛[0m
␛[0;32mI (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000␛[0m
␛[0;32mI (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000␛[0m
␛[0;32mI (83) boot:  2 factory          factory app      00 00 00010000 00100000␛[0m
␛[0;32mI (91) boot: End of partition table␛[0m
␛[0;32mI (95) boot_comm: chip revision: 1, min. application chip revision: 0␛[0m
␛[0;32mI (102) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x06ad0 ( 27344) map␛[0m
␛[0;32mI (121) esp_image: segment 1: paddr=0x00016af8 vaddr=0x3ffb0000 size=0x02150 (  8528) load␛[0m
␛[0;32mI (124) esp_image: segment 2: paddr=0x00018c50 vaddr=0x40080000 size=0x00400 (  1024) load␛[0m
␛[0;32mI (129) esp_image: segment 3: paddr=0x00019058 vaddr=0x40080400 size=0x06fb8 ( 28600) load␛[0m
␛[0;32mI (150) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x17ab0 ( 96944) map␛[0m
␛[0;32mI (184) esp_image: segment 5: paddr=0x00037ad0 vaddr=0x400873b8 size=0x02a70 ( 10864) load␛[0m
␛[0;32mI (196) boot: Loaded app from partition at offset 0x10000␛[0m
␛[0;32mI (196) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (196) cpu_start: Pro cpu up.␛[0m
␛[0;32mI (200) cpu_start: Application information:␛[0m
␛[0;32mI (205) cpu_start: Project name:     espidf-peripherals-i2s␛[0m
␛[0;32mI (211) cpu_start: App version:      1␛[0m
␛[0;32mI (215) cpu_start: Compile time:     Aug 21 2020 15:24:03␛[0m
␛[0;32mI (221) cpu_start: ELF file SHA256:  2f452bad7dd931fa...␛[0m
␛[0;32mI (227) cpu_start: ESP-IDF:          HEAD-HASH-NOTFOUND␛[0m
␛[0;32mI (233) cpu_start: Starting app cpu, entry point is 0x40081194␛[0m
␛[0;32mI (0) cpu_start: App cpu up.␛[0m
␛[0;32mI (244) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (251) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM␛[0m
␛[0;32mI (257) heap_init: At 3FFB3148 len 0002CEB8 (179 KiB): DRAM␛[0m
␛[0;32mI (263) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM␛[0m
␛[0;32mI (269) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM␛[0m
␛[0;32mI (276) heap_init: At 40089E28 len 000161D8 (88 KiB): IRAM␛[0m
␛[0;32mI (282) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (300) spi_flash: detected chip: generic␛[0m
␛[0;32mI (301) spi_flash: flash io: dio␛[0m
␛[0;32mI (301) cpu_start: Starting scheduler on PRO CPU.␛[0m
␛[0;32mI (0) cpu_start: Starting scheduler on APP CPU.␛[0m
␛[0;32mI (310) I2S: DMA Malloc info, datalen=blocksize=240, dma_buf_count=6␛[0m
␛[0;32mI (320) I2S: PLL_D2: Req RATE: 36000, real rate: 36764.000, BITS: 16, CLKM: 17, BCK: 8, MCLK: 9216921.692, SCLK: 1176448.000000, diva: 64, divb: 23␛[0m

Test bits=16 free mem=294316, written data=1440`

Alvin1Zhang, 2020-09-03:
Thanks for reporting, we will look into.

costaud, 2020-09-03:
Which sample rate did you set for the LA when  capturing the waveform ?

Serjy1710, 2020-09-03:
> Which sample rate did you set for the LA when capturing the waveform ?
> Suggest to set a higher sample rate like 20M/S or 50M/S.

Thank you for advice, you're right. I've use 4M/S. When switch to 16M/S, all looks fine. Pls. close issue

closed 2020-09-03 by Alvin1Zhang