espressif/esp-idf #5820: endless mistakes

exrector, 2020-09-03:
esp-idf installation instruction mac os doesn't work. There are a lot of errors on the installation path. This instruction is very specific. errors occur at almost every step. Python versions, access rights. Errors vary from machine to machine. Make a one-stop package. This is funny. I can't flash the device for a week. I didn't even come close to learning about the processor. I am still banging against walls with rudimentary installation procedures. The internet is littered with these errors.

exrector, 2020-09-03:
configure: error: missing required tool: gobjcopy objcopy

exrector, 2020-09-03:
brew info binutils
export PATH="/usr/local/opt/binutils/bin:$PATH"

checking for shared library run path origin... done
checking whether NLS is requested... yes
checking for CFPreferencesCopyAppValue... yes
checking for CFLocaleCopyCurrent... yes
checking for GNU gettext in libc... no
checking for iconv... yes
checking for working iconv... yes
checking how to link with libiconv... -liconv
checking for GNU gettext in libintl... yes
checking whether to use NLS... yes
checking where the gettext function comes from... external libintl
checking how to link with libintl... -lintl -Wl,-framework -Wl,CoreFoundation
checking for pkg-config... /usr/local/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for ncursesw via pkg-config... yes
checking for working ncursesw/curses.h... no
checking for working ncursesw.h... no
checking for working ncurses.h... yes
checking for Curses Panel library with ncursesw/panel.h... no
checking for Curses Panel library with panel.h... yes
checking for Curses Menu library with ncursesw/menu.h... no
checking for Curses Menu library with menu.h... no
configure: error: menu library not found

exrector, 2020-09-03:
HEAD is now at c338a656 ci: Add a Win64 build
WTF? Win64??

igrr, 2020-09-03:
Hi @exrector, sorry that you had bad experience with the getting started guide. Looks like you are trying to build the cross-compiler from scratch, which is not actually required if you simply need to get a working development environment.

If you are installing a recent version of ESP-IDF, it should be fairly straightforward on macOS.

1. The first step is to install some prerequisites: `brew install cmake ninja ccache` or `sudo port install cmake ninja ccache`, depending on whether you are using Homebrew or MacPorts.
2. Then clone ESP-IDF repository from Github. Recommend using `v4.1` tag, which is the latest stable release:
   ```
   git clone --recursive https://github.com/espressif/esp-idf.git
   ```
3. When that is done, change into the created esp-idf directory and run the install script:
   ```
   ./install.sh
   ```
4. The installation is now complete, and you can use ESP-IDF. To add ESP-IDF tools to your shell environment, go to esp-idf directory and run
   ```
   . ./export.sh
   ```
   (You can also use `direnv` tool to automatically source this export script into your environment when changing into your project directory.)

Please go through these steps, and if you meet any issue, post the command you ran and the full output you got. We'll do our best to help you.

exrector, 2020-09-04:
> Выполните эти шаги.

brew install cmake ninja ccache
cd ~/esp-idf
git clone -b v4.1 --recursive https://github.com/espressif/esp-idf.git
cd ~/esp-idf
./install.sh
cd ~/esp-idf
. ./export.sh
cd ~/esp
cp -R ~/esp-idf/examples/get-start/ hello_world ~/esp-idf/get-start/hello_world (any way copied to Finder)
cd ~/esp
cp -r $IDF_PATH/examples/get-started/hello_world .  (I'm not sure if this line is necessary)
cd ~/esp/hello_world
idf.py menuconfig
Run '/Users/exrector/esp-idf/tools/idf.py fullclean' to start again.
/Users/exrector/esp-idf/tools/idf.py fullclean
Executing action: fullclean
Done
again 
idf.py menuconfig
Q for out
idf.py build

**this part ended in success. but there are problems similar to the arduino-ide program**
Traceback (most recent call last):
  File "/Users/exrector/esp-idf/components/esptool_py/esptool/esptool.py", line 58, in <module>
    import serial.tools.list_ports as list_ports
  File "/Users/exrector/.espressif/python_env/idf4.3_py2.7_env/lib/python2.7/site-packages/serial/tools/list_ports.py", line 29, in <module>
    from serial.tools.list_ports_posix import comports
  File "/Users/exrector/.espressif/python_env/idf4.3_py2.7_env/lib/python2.7/site-packages/serial/tools/list_ports_posix.py", line 31, in <module>
    from serial.tools.list_ports_osx import comports
  File "/Users/exrector/.espressif/python_env/idf4.3_py2.7_env/lib/python2.7/site-packages/serial/tools/list_ports_osx.py", line 32, in <module>
    kIOMasterPortDefault = ctypes.c_void_p.in_dll(iokit, "kIOMasterPortDefault")
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found
ninja: build stopped: subcommand failed.
[318/952] Building C object esp-idf/wpa_supplicant/CMakeFiles/__idf_wpa_supplicant.dir/src/common/dpp.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
bash-3.2$ 

**thanks a lot for your help!
Now I know that there are not many steps indicated in the original instructions. No toolchain needed. you need to work in a clean bash shell. in the original instructions, the path is specified by another cd ~ / esp / esp-idf instead of cd ~ / esp-idf and each cloning takes place in different folders. i can't trace the logic**

I keep laughing at the line "HEAD is now at c338a656 ci: Add a Win64 build" it was very funny and strange 😂

dobairoland, 2020-09-04:
@exrector There is a workaround for that issue: https://github.com/espressif/esptool/issues/540#issuecomment-681863191

Unfortunately, this isn't included in IDF yet but you can apply the fix manually into your `~/esp-idf/components/esptool_py/esptool/esptool.py` file.

exrector, 2020-09-05:
Всем спасибо за помощь. Все получилось
Thanks everybody for help! Everything worked.

closed 2020-09-05 by exrector