espressif/esp-idf #5821: boot failure due to SOC_RESERVE_MEMORY_REGION overlap

jeremyherbert, 2020-09-03:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it): v4.1 (but I have also checked with master; same issue)
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:    Linux
- Using an IDE?: No
- Power Supply:         external 5V

## Problem Description

I am having a problem where the boot fails because something is reserving the entire memory region in the address space:

```
I (29) boot: ESP-IDF v4.1 2nd stage bootloader
I (29) boot: compile time 18:04:05
I (29) boot: chip revision: 1
I (30) qio_mode: Enabling default flash chip QIO
I (33) boot.esp32: SPI Speed      : 80MHz
I (35) boot.esp32: SPI Mode       : QIO
I (37) boot.esp32: SPI Flash Size : 16MB
I (40) boot: Enabling RNG early entropy source...
I (43) boot: Partition Table:
I (44) boot: ## Label            Usage          Type ST Offset   Length
I (48) boot:  0 nvs              WiFi data        01 02 0000d000 00004000
I (52) boot:  1 otadata          OTA data         01 00 00011000 00002000
I (55) boot:  2 phy_init         RF data          01 01 00013000 00001000
I (59) boot:  3 factory          factory app      00 00 00020000 00400000
I (63) boot:  4 ota_0            OTA app          00 10 00420000 00400000
I (67) boot:  5 ota_1            OTA app          00 11 00820000 00400000
I (70) boot:  6 storage          Unknown data     01 82 00c20000 00100000
I (74) boot:  7 auth             Unknown data     01 82 00d20000 00100000
I (78) boot:  8 nvs_key          NVS keys         01 04 00e20000 00001000
I (82) boot: End of partition table
I (84) boot: Defaulting to factory image
I (86) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x2090c (133388) map
I (137) esp_image: segment 1: paddr=0x00040934 vaddr=0x3ffbdb60 size=0x02a4c ( 10828) load
I (141) esp_image: segment 2: paddr=0x00043388 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/jeremy/esp/esp-idf-v4.1/components/freertos/xtensa_vectors.S:1778

I (142) esp_image: segment 3: paddr=0x00043794 vaddr=0x40080404 size=0x0c884 ( 51332) load
I (165) esp_image: segment 4: paddr=0x00050020 vaddr=0x400d0020 size=0x614d8 (398552) map
0x400d0020: _stext at ??:?

I (303) esp_image: segment 5: paddr=0x000b1500 vaddr=0x4008cc88 size=0x05ce0 ( 23776) load
0x4008cc88: prvAddNewTaskToReadyList at /home/jeremy/esp/esp-idf-v4.1/components/freertos/tasks.c:1062

I (324) boot: Loaded app from partition at offset 0x20000
I (324) boot: Disabling RNG early entropy source...
I (324) psram: This chip is ESP32-D0WD
I (326) spiram: Found 64MBit SPI RAM device
I (329) spiram: SPI RAM mode: flash 80m sram 80m
I (331) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (335) cpu_start: Pro cpu up.
I (337) cpu_start: Application information:
I (339) cpu_start: Project name:     esp32_christie_bbq
I (342) cpu_start: App version:      34e41ee-dirty
I (345) cpu_start: Compile time:     Sep  3 2020 18:02:32
I (348) cpu_start: ELF file SHA256:  73208bf173b92ccc...
I (351) cpu_start: ESP-IDF:          v4.1
I (353) cpu_start: Starting app cpu, entry point is 0x400816f8
0x400816f8: call_start_cpu1 at /home/jeremy/esp/esp-idf-v4.1/components/esp32/cpu_start.c:271

I (342) cpu_start: App cpu up.
I (856) spiram: SPI SRAM memory test OK
D (856) memory_layout: Checking 8 reserved memory ranges:
D (856) memory_layout: Reserved memory range 0x3f800000 - 0x40000000
E (859) memory_layout: SOC_RESERVE_MEMORY_REGION region range 0x3f800000 - 0x40000000 overlaps with 0x3ffae000 - 0x3ffae6e0
abort() was called at PC 0x400e1c2f on core 0
0x400e1c2f: s_prepare_reserved_regions at /home/jeremy/esp/esp-idf-v4.1/components/soc/src/memory_layout_utils.c:109
 (inlined by) soc_get_available_memory_regions at /home/jeremy/esp/esp-idf-v4.1/components/soc/src/memory_layout_utils.c:126

ELF file SHA256: 73208bf173b92ccc

Backtrace: 0x4008944b:0x3ffe34e0 0x400898e4:0x3ffe3510 0x400e1c2f:0x3ffe3540 0x400d5035:0x3ffe38a0 0x40081957:0x3ffe3c40 0x4007932a:0x3ffe3c80 |<-CORRUPTED
0x4008944b: invoke_abort at /home/jeremy/esp/esp-idf-v4.1/components/esp32/panic.c:159

0x400898e4: abort at /home/jeremy/esp/esp-idf-v4.1/components/esp32/panic.c:174

0x400e1c2f: s_prepare_reserved_regions at /home/jeremy/esp/esp-idf-v4.1/components/soc/src/memory_layout_utils.c:109
 (inlined by) soc_get_available_memory_regions at /home/jeremy/esp/esp-idf-v4.1/components/soc/src/memory_layout_utils.c:126

0x400d5035: heap_caps_init at /home/jeremy/esp/esp-idf-v4.1/components/heap/heap_caps_init.c:65

0x40081957: call_start_cpu0 at /home/jeremy/esp/esp-idf-v4.1/components/esp32/cpu_start.c:256
```

I seem to be having this exact problem: https://github.com/raphaelbs/esp32-cam-ai-thinker/issues/26 Even the addresses are exactly the same. But it seems there was no resolution, it just disappeared.

I unfortunately cannot attach my code, but I have attached the sdkconfig file. 
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5167211/sdkconfig.txt)

jeremyherbert, 2020-09-03:
I have been able to get this to fail with the hello world example. See attached.
[overlap_error_test.zip](https://github.com/espressif/esp-idf/files/5167532/overlap_error_test.zip)

jeremyherbert, 2020-09-03:
I have no idea how to explain this, but I have found there is a single Kconfig option causing this problem. If I set the config "Type of SPI RAM chip in use" to "Auto-detect", everything works fine. If I set it to "ESP-PSRAM64 or LY68L6400" then it crashes with the above error.

negativekelvin, 2020-09-04:
https://github.com/espressif/esp-idf/pull/5373

bill-orange, 2020-09-20:
@jeremyherbert I was not able to get your "Auto-detect" fix to work for me.  Can you post your entire kconfig file?  There could be more than a couple of settings in play here.

projectgus, 2020-09-21:
Hi folks,

Apologies for the delay in merging the linked PR from @negativekelvin . We'll merge this ASAP.

In the meantime setting SPI RAM type to "Auto-detect" as mentioned should work.

@bill-orange if you're still having problems with this, could you please post some information about your hardware and the crash you are seeing/

bill-orange, 2020-09-21:
Thank you for you offer of assistance.  It is likely that I am not implementing the SPI RAM type properly to form a fix.  I modified sdkconfig.default to include `CONFIG_SPIRAM_TYPE_AUTO=y`.  I suspect that was not what was needed,  here is the error.

```
D (1215) memory_layout: Checking 8 reserved memory ranges:␛[0m
D (1215) memory_layout: Reserved memory range 0x3f800000 - 0x40000000␛[0m
␛[0;31mE (1220) memory_layout: SOC_RESERVE_MEMORY_REGION region range 0x3f800000 - 0x40000000 overlaps with 
0x3ffae000 - 0x3ffae6e0␛[0m
abort() was called at PC 0x400dc81b on core 0

ELF file SHA256: 1ecf195c45fe00f3

Backtrace: 0x40089370:0x3ffe34a0 0x4008981c:0x3ffe34d0 0x400dc81b:0x3ffe3500 0x400dc8c0:0x3ffe3540 0x400db895:0x3ffe3890 0x40082599:0x3ffe3c40 0x4007931a:0x3ffe3c80 |<-CORRUPTED

Rebooting...
```
I have been working the an AI-Thinker ESP32-CAM.  The only code I have tried to get working is the motion detect example here:

[](https://github.com/bill-orange/esp32-cam-ai-thinker/tree/master/examples/change_detection)

Any advice you can offer would be greatly appreciated.

projectgus, 2020-09-23:
Hi @bill-orange 

It looks like your ESP32-CAM has an 8MB SPIRAM chip. I can't explain why enabling automatic detection hasn't fixed the problem though, maybe the `sdkconfig.defaults` change hasn't been applied and your project is still using a custom sdkconfig?

Could you please open menuconfig and manually check this value is set to Auto-detect in the menu?
https://docs.espressif.com/projects/esp-idf/en/stable/api-reference/kconfig.html#config-spiram-type

bill-orange, 2020-09-23:
Your insight was correct `sdkconfig.defaults` changes were apparently not being applied.  I modified `sdkconfig` directly as follows:

#CONFIG_SPIRAM_SIZE=-1
CONFIG_SPIRAM_TYPE=Auto-detect

The memory crash is gone.  Commenting out the first line above seemed the be part of the magic.

Thanks!

franzhoepfinger, 2020-10-13:
Hello, 

solved this like this: 

in soc\_memory\_layout.c

```c
#if CONFIG_SPIRAM_SIZE == -1
// Assume we need to reserve 4MB in the auto-detection case
#define RESERVE_SPIRAM_SIZE (4*1024*1024)
#elif CONFIG_SPIRAM_SIZE == (8*1024*1024)
// Assume we need to reserve 4MB in the case the 8MB SPIRAM is populated
#define RESERVE_SPIRAM_SIZE (4*1024*1024)
#else
#define RESERVE_SPIRAM_SIZE CONFIG_SPIRAM_SIZE
#endif
```

closed 2020-10-14 by espressif-bot