espressif/esp-idf #5822: WiFi error: Guru Meditation Error: Core 1 panic'ed (IllegalInstruction)

jarcos-al, 2020-09-03:
Hello everyone. ;)  I am Jorge and I'm new on this forum.

Currently, I'm developing a firmware in a ESP32 module for create a custom hub.

I have an [b]ESP32-Wrover-E module (16MB spi flash) [/b]and I have an issue with WiFi that i can't fix.
I'm using[b] ESP-IDF  v4.3-dev[/b], SPI speed: 40 MHz and SPI Mode: DIO.

The problem reported by the IDF Serial Monitor is this when the WiFi has been connected to an AP for a while:

[quote]Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013f504: 04606502 0000f01d ad004136
Core  1 register dump:
PC      : 0x4013f508  PS      : 0x00060130  A0      : 0x00040020  A1      : 0x3ffc1f90  
A2      : 0x3ffdd1a8  A3      : 0x00001770  A4      : 0x00000000  A5      : 0x3ffb1a68  
A6      : 0x3ffb6460  A7      : 0x00000001  A8      : 0x8013f508  A9      : 0x3ffc1f70  
A10     : 0x3ffdd1a8  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x005b8d80  
A14     : 0x3ffb6460  A15     : 0x3ffb5a50  SAR     : 0x00000018  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4012c180  LEND    : 0x4012c1ab  LCOUNT  : 0x00000007  

Backtrace:0x4013f505:0x3ffc1f90 0x0004001d:0x3ffc1fb0 |<-CORRUPTED

ELF file SHA256: 4c36298654fdb336

Rebooting...[/quote]

I have tried other ESP32 modules (wrover b in a ESP32-Wrover-Kit v4.1 and ESP32-WROOM-32 in ESP32- Devkit-C) and others power suply and dc regulators for discard a power suply issue.
I have also put capacitors of 2200 uf value to the 3.3 v in pin.

The error tends to appear randomly at runtime, so it is impossible for me to debug it. :? 

Can anyone help me to fix this problem?

Thanks in advance.

## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version (run ``git describe --tags`` to find it):
    

v4.3-dev-771-gc77c4ccf6-dirty
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:   [Linux]
- Using an IDE?: [Yes (Eclipse)]
- Power Supply:         [external 3.3V]

## Debug Logs

```
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7116
load:0x40078000,len:13852
ho 0 tail 12 room 4
load:0x40080400,len:3952
entry 0x40080678
I (31) boot: ESP-IDF v4.3-dev-771-gc77c4ccf6-dirty 2nd stage bootloader
I (31) boot: compile time 15:11:03
I (32) boot: chip revision: 3
I (36) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (43) boot.esp32: SPI Speed      : 40MHz
[... 102 lines cut ...]
I (00:00:20.322) MQTTS: MQTT_EVENT_DATA

**(Minutes later...)**

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013f504: 04606502 0000f01d ad004136
Core  1 register dump:
PC      : 0x4013f508  PS      : 0x00060d30  A0      : 0x00040020  A1      : 0x3ffc1f90  
A2      : 0x3ffdd1a4  A3      : 0x00001770  A4      : 0x00000000  A5      : 0x3ffb1a68  
A6      : 0x3ffb6460  A7      : 0x00000001  A8      : 0x8013f508  A9      : 0x3ffc1f70  
A10     : 0x3ffdd1a4  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x005b8d80  
A14     : 0x3ffb6460  A15     : 0x3ffb5a50  SAR     : 0x00000018  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4012c180  LEND    : 0x4012c1ab  LCOUNT  : 0x00000007  

Backtrace:0x4013f505:0x3ffc1f90 0x0004001d:0x3ffc1fb0 |<-CORRUPTED

ELF file SHA256: 4c36298654fdb336

Rebooting...
```

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder)
[attachments.zip](https://github.com/espressif/esp-idf/files/5167897/attachments.zip)

liuzfesp, 2020-09-03:
Hi @jarcos-al please help to attach the *.elf file here and then I can analyze the backtrace of the crashing. You can find the *.elf file under the project/build dir, e.g. the iperf.elf is located under $IDF_PATH/examples/wifi/iperf/build. 

Also notified that the *.elf file should be the one which is generated at the same time as the *.bin file that triggers the crashing. If you project is re-compile, then you may need take following steps to reproduce it:

- Re-build the project, and keep the *.elf/*.bin
- Reproduce the crashing and attach the log and *.elf here

igrr, 2020-09-03:
Judging by the symptoms (IllegalInstruction with A0 register containing 0x40020), this seems to be the same issue as https://github.com/espressif/esp-idf/issues/5758 — the fix for it is in review and should be merged soon.

jarcos-al, 2020-09-04:
@igrr, I confirm, the issue is due to Heap tracing in mode 'Standalone'. I have desactivated this option in the sdkconfig and the problem has not reappeared.
Thanks for the support!

closed 2020-09-04 by jarcos-al