espressif/esp-idf #5823: Nimble tutorial for custom BLE service/characteristic? (IEP-227)

freeride01, 2020-09-03:
Hi, I found [this page](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_gatts.html) with tutorials on how to make a custom service/characteristic in BLE, but they are all for the bluedroid stack. Is there one for Nimble?

dhrishi, 2020-09-04:
Hi @freeride01 , examples for NimBLE host stack in ESP-IDF are present [here](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble)
The BLE peripheral example (bleprph) specifically demonstrates using a custom (128 bit) service/characteristic. You can find the relevant code [here](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/nimble/bleprph/main/gatt_svr.c#L39)

Alvin1Zhang, 2020-09-22:
Thanks for reporting, feel free to reopen, thanks.

closed 2020-09-22 by Alvin1Zhang