espressif/esp-idf #5828: Require help with the health server model in esp ble mesh

ankit-thealchemist, 2020-09-05:
Hello,

In the esp ble mesh, I am trying to implement the health server/client model. I have a question that can we set the different health test structure in the single element, or we have to implement the different test array with different company id on the different elmenent. Or I am missing something here. 

`esp_ble_mesh_health_srv_t health_server_1 = {
    // server structure
    .health_test =
        {
            .id_count = ARRAY_SIZE(test_array),
            .test_ids = test_array,
            .company_id = CID_ESP,
        },
    // health server test struct
};`

Campou, 2020-09-21:
Hi @ankit-thealchemist 

Sorry for the late reply.

The `test_ids` is just a collection of the health test, based on the different test ids, different test operation can be performed. But for a single element, it can only contain one company id.

The user data of the health server may needs some optimization, because previously, we didn't come up with the idea that users may perform health test with different company ids.

If any implementation update is needed, I could provide a patch here.

Thanks.

ankit-thealchemist, 2020-09-21:
@Campou thanks for reply!
I was checking if I am not missing anything important.  Will let you know if I need something more!

ankit-thealchemist, 2020-09-21:
Closing the issue!

closed 2020-09-21 by ankit-thealchemist