espressif/esp-idf #5829: How to connect to a already provisioned WiFi credential?? Could you also refer me to some WiFi profile API's??

noel-ED, 2020-09-05:
I am using MQTT SSL library to connect to a broker and I have built up my application on it. In order to provision my device I have made use of the BluFi example and incorporated it to work as desired. The only issue I have now is related to connecting to an already provisioned Wifi AP.
I have used a TI wifi controller extensively and there is a provision where I can retrieve the wifi Credentials using their get_profile(.....) API's.
I am sure there will be a similar implementation on the ESP 32 device too. Could you  guide me through. I have gone through all of the available guides available online for it but could not come across relevant info.
Hope to hear soon.
Thank you.

-Noel

xiewenxiang, 2021-09-08:
@noel-ED 
Could you please describe your issues more detail? Is there any errors occur during blufi provision?

jack0c, 2022-05-20:
You can try example https://github.com/espressif/esp-idf/tree/master/examples/wifi/getting_started/station.

Alvin1Zhang, 2022-07-25:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more details, thanks.

closed 2022-07-25 by Alvin1Zhang