espressif/esp-idf #5831: DNS Not working when DHCP disabled for setting static ip

jogendar, 2020-09-06:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Alvin1Zhang, 2020-09-07:
@jogendar Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

jogendar, 2020-09-08:
@Alvin1Zhang  Thanks for looking in to it.

## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version WroverKit v3
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):
    v4.0.1
- Build System:         |idf.py
- Operating System:     Windows
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: No
- Power Supply: USB

## Problem Description
Connected to WiFi In station mode and
I have https client (get query to get weather from open weather api) + tcp server application running on two different threads.
to connect to TCP server, i made it static ip. but when static ip enbled not able to connect to weather API (with DNS error)

### Expected Behavior
static IP + DNS should work together since there is no relation b/w DHCP and DNS
### Actual Behavior

### Steps to reproduce

1. Enable wifi in Station mode
2. Disable dhcp and set static ip
3. call http url with domain name

I don't have the setup right now. I'll post the logs once I got setup in my hand

hagre, 2020-09-08:
Some days ago i mabye had the same problem. After some hours "debuging" the software, i discoverd that the problem was the new android mobile phone hotspot i have just started to use. With the change to Android pie 9.0 all ip adresses of the hotspot/ap/gateway changed automaticly to a random number each activationtime. 
On the ESP side i used a static ip setting with a correct/possible DNS and esp ip, but i missed the changing gateway ip. So all connections in the network itself worked normal, but connections to the internet incl. the DNS stopped working!
Now i am connecting using dhcp to get the actual setting. disconnect and reconnect with the same numbers as static settings (ecxept the esp ip).

YouDONG-ESP, 2020-10-21:
Hi @jogendar , did @hagre helps your problem?

freakyxue, 2020-11-10:
hi @jogendar 
Follow the steps below to test:

-  connect to ap

- disable dhcp

-  set static ip addr

- set fallback dns

- get website dns IP

freakyxue, 2020-12-23:
hi @jogendar 
Do you have any questions? If not, I will close this issue. If you have any problems, please open this question at any time. Thank you

Alvin1Zhang, 2020-12-23:
Thanks for reporting, will close due to short of feedback, feel free to reopen. Thanks.

closed 2020-12-23 by Alvin1Zhang