espressif/esp-idf #5832: Task watchdog got triggered. The following tasks did not reset the watchdog in time

stdenits, 2020-09-06:
## Environment

Development Kit: [none]
Kit version: [none]
Module or chip used: [ESP32-PICO-D4]
IDF version: v4.3
Build System: [???]
Compiler version: [???]
Operating System: [Windows]
(Windows only) environment type: [???].
Using an IDE?: [Yes (VSCode + ESP-IDF Extension)]
Power Supply: [external 3.3V]

## Problem Description

ESP32 reboots when entering to the while(1){} loop at main().
The code link lead to the blank row or to the condition with bool variable.

### Expected Behavior

It works fine with ESP IDF 4.0 and probably 4.2. Never got errors with while() cycle. I'm interested what have changed.

### Actual Behavior

### Steps to reproduce

### Code to reproduce this issue
```
void app_main()
{
    /* Some inits */

    if (TEST ? 1 : 0) {
        /* SPI send */

        ESP_LOGI(TAG3, "Start SPI transactions");
        /* Some variables */
        
        while (1) {
        **Reboots here**
        /* Some cycles with condition */
        }
    } else {
        /* SPI send */
    }
}
```

## Debug Logs
**Incorrect behavior log**
```
I (0) cpu_start: App cpu up.
I (509) cpu_start: Pro cpu start user code
I (509) cpu_start: Application information:
I (509) cpu_start: Project name:     wifi_softAP
I (514) cpu_start: App version:      1
I (519) cpu_start: Compile time:     Sep  6 2020 17:36:19
I (525) cpu_start: ELF file SHA256:  054ee918d459648a...
I (531) cpu_start: ESP-IDF:          v4.3-dev-907-g6c17e3a64-dirty
I (538) heap_init: Initializing. RAM available for dynamic allocation:
I (545) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (551) heap_init: At 3FFB8D90 len 00027270 (156 KiB): DRAM
I (557) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (563) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (570) heap_init: At 40096C8C len 00009374 (36 KiB): IRAM
I (577) spi_flash: detected chip: gd
I (580) spi_flash: flash io: dio
I (585) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (595) ESP32_Server: app_main() runned on core 0

I (635) ESP32_SPI: Initializing SPI
I (635) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (635) ESP32_SPIFFS: Initializing SPIFFS
I (735) ESP32_SPIFFS: Partition size: total: 896321, used: 529359
I (735) ESP32_Server: ESP_WIFI_MODE_AP
I (735) ESP32_mDNS: mdns hostname set to: [irsta.local]
I (755) wifi:wifi driver task: 3ffc3be0, prio:23, stack:6656, core=0
I (755) system_api: Base MAC address is not set
I (755) system_api: read default base MAC address from EFUSE
I (785) wifi:wifi firmware version: 8d94a49
I (785) wifi:wifi certification version: v7.0
I (785) wifi:config NVS flash: enabled
I (785) wifi:config nano formating: disabled
I (785) wifi:Init data frame dynamic rx buffer num: 32
I (785) wifi:Init management frame dynamic rx buffer num: 32
I (795) wifi:Init management short buffer num: 32
I (795) wifi:Init dynamic tx buffer num: 32
I (805) wifi:Init static rx buffer size: 1600
I (805) wifi:Init static rx buffer num: 10
I (815) wifi:Init dynamic rx buffer num: 32
I (815) wifi_init: rx ba win: 6
I (815) wifi_init: tcpip mbox: 32
I (825) wifi_init: udp mbox: 6
I (825) wifi_init: tcp mbox: 6
I (835) wifi_init: tcp tx win: 5744
I (835) wifi_init: tcp rx win: 5744
I (835) wifi_init: tcp mss: 1440
I (845) wifi_init: WiFi IRAM OP enabled
I (845) wifi_init: WiFi RX IRAM OP enabled
I (855) ESP32_Server: WiFi *** 
I (855) ESP32_Server: PSW *** 
I (875) phy: phy_version: 4370, 4e803b3, Aug 11 2020, 14:18:07, 0, 1
I (875) wifi:mode : sta (***)
I (1005) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:13
I (1005) wifi:state: init -> auth (b0)
I (1005) wifi:state: auth -> assoc (0)
I (1015) wifi:state: assoc -> run (10)
I (1035) wifi:connected with ***, aid = 3, channel 6, BW20, bssid = ***
I (1035) wifi:security: WPA2-PSK, phy: bgn, rssi: -86
I (1045) wifi:pm start, type: 1

I (1085) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (1635) esp_netif_handlers: sta ip: 192.168.0.248, mask: 255.255.255.0, gw: 192.168.0.1
I (1635) ESP32_Server: SYSTEM_EVENT_STA_GOT_IP 
I (1635) ESP32_Server: Login Success
I (1645) ESP32_Server: Starting server on port: '80'
I (1645) ESP32_Server: Registering URI handlers
sended {"c80":2}
I (1745) ESP32_SPI: Start SPI transactions
E (6745) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (6745) task_wdt:  - IDLE0 (CPU 0)
E (6745) task_wdt: Tasks currently running:
E (6745) task_wdt: CPU 0: main
E (6745) task_wdt: CPU 1: IDLE1
E (6745) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x40106457:0x3FFB0820 0x40082B21:0x3FFB0840 0x400D91C2:0x3FFBACE0 0x40088088:0x3FFBAD00 0x40087FF5:0x3FFBAD20
0x40106457: task_wdt_isr at C:/Users/User/esp-idf/components/esp_common/src/task_wdt.c:187

0x40082b21: _xt_lowint1 at C:/Users/User/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105

0x400d91c2: app_main at C:\Users\User\esp\softAp\softAP\build/../main/softap_example_main.c:804

0x40088088: main_task at C:/Users/User/esp-idf/components/freertos/xtensa/port.c:511

0x40087ff5: vPortTaskWrapper at C:/Users/User/esp-idf/components/freertos/xtensa/port.c:169

E (6745) task_wdt: Print CPU 1 backtrace

Backtrace:0x40084D17:0x3FFB0E20 0x40082B21:0x3FFB0E40 0x4000BFED:0x3FFBBB30 0x400882D5:0x3FFBBB40 0x401066FB:0x3FFBBB60 0x40106707:0x3FFBBB90 0x400D7915:0x3FFBBBB0 0x40088AD1:0x3FFBBBD0 0x40087FF5:0x3FFBBBF0
0x40084d17: esp_crosscore_isr at C:/Users/User/esp-idf/components/esp32/crosscore_int.c:77

0x40082b21: _xt_lowint1 at C:/Users/User/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105

0x400882d5: vPortExitCritical at C:/Users/User/esp-idf/components/freertos/xtensa/port.c:445

0x401066fb: esp_task_wdt_reset at C:/Users/User/esp-idf/components/esp_common/src/task_wdt.c:333

0x40106707: idle_hook_cb at C:/Users/User/esp-idf/components/esp_common/src/task_wdt.c:88

0x400d7915: esp_vApplicationIdleHook at C:/Users/User/esp-idf/components/esp_common/src/freertos_hooks.c:51 (discriminator 1)

0x40088ad1: prvIdleTask at C:/Users/User/esp-idf/components/freertos/tasks.c:3386 (discriminator 1)

0x40087ff5: vPortTaskWrapper at C:/Users/User/esp-idf/components/freertos/xtensa/port.c:169

E (11745) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (11745) task_wdt:  - IDLE0 (CPU 0)
E (11745) task_wdt: Tasks currently running:
E (11745) task_wdt: CPU 0: main
E (11745) task_wdt: CPU 1: IDLE1
E (11745) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x40106457:0x3FFB0820 0x40082B21:0x3FFB0840 0x400D91C2:0x3FFBACE0 0x40088088:0x3FFBAD00 0x40087FF5:0x3FFBAD20
0x40106457: task_wdt_isr at C:/Users/User/esp-idf/components/esp_common/src/task_wdt.c:187

0x40082b21: _xt_lowint1 at C:/Users/User/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105

0x400d91c2: app_main at C:\Users\User\esp\softAp\softAP\build/../main/softap_example_main.c:804

0x40088088: main_task at C:/Users/User/esp-idf/components/freertos/xtensa/port.c:511

0x40087ff5: vPortTaskWrapper at C:/Users/User/esp-idf/components/freertos/xtensa/port.c:169
```

## Other items if possible

lucastcox, 2020-09-07:
You need to add a vTaskDelay(number of ticks you want to delay); in your while loop or it will run constantly and prevent the IDLE task from servicing the watchdog. 

Either do the above or disable the task watchdog in the menuconfig.

stdenits, 2020-09-08:
@lucastcox 
Thank you. Now it works.

closed 2020-09-08 by stdenits

godofred, 2023-11-10:
For future reference, I added 10ms to my main loop and the warning was gone.
I also tried 1ms, but it did not have effect.