espressif/esp-idf #5834: Re-enable cpu cache

chegewara, 2020-09-07:
This is very rare, not possible to replicate issue. CPU cache got re-enabled just like that during OTA update over ble:
```
Re-enable cpu cache.
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Core  0 register dump:
PC      : 0x4014e35c  PS      : 0x00060634  A0      : 0x8008ba4a  A1      : 0x3ffbe3a0  
0x4014e35c: llcp_channel_map_req_pre_handler at llc_llcp.c:?

A2      : 0x4014e35c  A3      : 0x00000000  A4      : 0x3ffb2fa6  A5      : 0x00000001  
0x4014e35c: llcp_channel_map_req_pre_handler at llc_llcp.c:?

A6      : 0x00000006  A7      : 0x3ffaf7c4  A8      : 0x80044715  A9      : 0x00000001  
A10     : 0x00000000  A11     : 0x00000006  A12     : 0x00000001  A13     : 0x3ffb2fa6  
A14     : 0x0000064f  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
```

`ESP-IDF:          v4.3-dev-907-g6c17e3a64`
Using bluedroid stack.

chegewara, 2020-09-07:
This is the rest of crash backtrace, but i think its not that important as it is cause by cache re-enable:
```
0x4014e359: config_llc_hci_funcs_reset at ??:?

0x4008ba47: r_lld_pdu_rx_handler at ??:?

0x4008a7dd: r_lld_evt_end at ??:?

0x4008a089: r_lld_evt_end_isr at ??:?

0x4008c1f2: r_rwble_isr at ??:?

0x4008d123: r_rwbtdm_isr_wrapper at intc.c:?

0x40082bc5: _xt_lowint1 at /home/chegewara/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105

0x4008480d: delay_us at /home/chegewara/esp/esp-idf/components/spi_flash/spi_flash_os_func_app.c:101

0x40093f4e: spi_flash_chip_generic_wait_idle at /home/chegewara/esp/esp-idf/components/spi_flash/spi_flash_chip_generic.c:326

0x400942b5: spi_flash_chip_generic_page_program at /home/chegewara/esp/esp-idf/components/spi_flash/spi_flash_chip_generic.c:216

0x40093e8d: spi_flash_chip_generic_write at /home/chegewara/esp/esp-idf/components/spi_flash/spi_flash_chip_generic.c:237

0x400846cd: esp_flash_write at /home/chegewara/esp/esp-idf/components/spi_flash/esp_flash_api.c:649

0x4011d4f9: esp_partition_write at /home/chegewara/esp/esp-idf/components/spi_flash/partition.c:389

0x4011ce6d: esp_ota_write at /home/chegewara/esp/esp-idf/components/app_update/esp_ota_ops.c:245

0x400d7fb0: ota_chunk at /home/chegewara/upwork/ble_mesh_lighting/test_ota/build/../main/ota_write.c:70

0x400d7a5b: OTACallbacksData::onWrite(BLECharacteristic*) at /home/chegewara/upwork/ble_mesh_lighting/test_ota/build/../main/ble_ota.cpp:36

0x400d97f5: BLECharacteristic::handleGATTServerEvent(esp_gatts_cb_event_t, unsigned char, esp_ble_gatts_cb_param_t*) at /home/chegewara/upwork/ble_mesh_lighting/test_ota/build/../components/BLE/BLECharacteristic.cpp:298

0x400de359: BLECharacteristicMap::handleGATTServerEvent(esp_gatts_cb_event_t, unsigned char, esp_ble_gatts_cb_param_t*) at /home/chegewara/upwork/ble_mesh_lighting/test_ota/build/../components/BLE/BLECharacteristicMap.cpp:87

0x400dbf1d: BLEService::handleGATTServerEvent(esp_gatts_cb_event_t, unsigned char, esp_ble_gatts_cb_param_t*) at /home/chegewara/upwork/ble_mesh_lighting/test_ota/build/../components/BLE/BLEService.cpp:369

0x400dc3d9: BLEServiceMap::handleGATTServerEvent(esp_gatts_cb_event_t, unsigned char, esp_ble_gatts_cb_param_t*) at /home/chegewara/upwork/ble_mesh_lighting/test_ota/build/../components/BLE/BLEServiceMap.cpp:90

0x400dbc16: BLEServer::handleGATTServerEvent(esp_gatts_cb_event_t, unsigned char, esp_ble_gatts_cb_param_t*) at /home/chegewara/upwork/ble_mesh_lighting/test_ota/build/../components/BLE/BLEServer.cpp:279

0x400da219: BLEDevice::gattServerEventHandler(esp_gatts_cb_event_t, unsigned char, esp_ble_gatts_cb_param_t*) at /home/chegewara/upwork/ble_mesh_lighting/test_ota/build/../components/BLE/BLEDevice.cpp:127

0x400e3d4a: btc_gatts_cb_to_app at /home/chegewara/esp/esp-idf/components/bt/host/bluedroid/btc/profile/std/gatt/btc_gatts.c:46
 (inlined by) btc_gatts_cb_handler at /home/chegewara/esp/esp-idf/components/bt/host/bluedroid/btc/profile/std/gatt/btc_gatts.c:805

0x40104e13: btc_thread_handler at /home/chegewara/esp/esp-idf/components/bt/common/btc/core/btc_task.c:158

0x40106e39: osi_thread_run at /home/chegewara/esp/esp-idf/components/bt/common/osi/thread.c:68

0x4008e0f1: vPortTaskWrapper at /home/chegewara/esp/esp-idf/components/freertos/xtensa/port.c:169
```

saloid, 2020-09-18:
Same issue - getting it sometimes during OTA update from server but with ble connected at this time. Working on `release/4.1` branch, commit `445635fe45b7205497ad81289c5a808156a43539`

saloid, 2020-09-18:
Looks like BLE interrupts are not disabled during ota
My log:
```
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Core 0 register dump:
PC      : 0x40130a80  PS      : 0x00060734  A0      : 0x800899d2  A1      : 0x3ffbe3f0
0x40130a80: llcp_channel_map_req_pre_handler at llc_llcp.c:?

A2      : 0x40130a80  A3      : 0x00000000  A4      : 0x3ffe2e7e  A5      : 0x00000001
0x40130a80: llcp_channel_map_req_pre_handler at llc_llcp.c:?

A6      : 0x00000006  A7      : 0x3ffaf7c4  A8      : 0x80044715  A9      : 0x00000001
A10     : 0x00000000  A11     : 0x00000006  A12     : 0x00000001  A13     : 0x3ffe2e7e
A14     : 0x00000edf  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
Core 0 was running in ISR context:
EPC1    : 0x4009460f  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40130a80
0x4009460f: portENTER_CRITICAL_NESTED at C:/Users/username/esp/esp4/components/freertos/include/freertos/portmacro.h:339
 (inlined by) xPortInIsrContext at C:/Users/username/esp/esp4/components/freertos/port.c:338

0x40130a80: llcp_channel_map_req_pre_handler at llc_llcp.c:?

ELF file SHA256: d3bc1120388a74dd

Backtrace: 0x40130a7d:0x3ffbe3f0 0x400899cf:0x3ffbe420 0x400485a5:0x3ffbe450 0x4004868d:0x3ffbe470 0x4008a216:0x3ffbe490 0x4008b1d7:0x3ffbe4b0 0x4008513d:0x3ffbe4d0 0x4009be02:0x3ffeea00 0x4009bc09:0x3ffeea40 0x4009c0c1:0x3ffeea70 0x4008cbd4:0x3ffeea90 0x40145a11:0x3ffeeaf0 0x40144fa5:0x3ffeeb10 0x4014128f:0x3ffeeb30 0x4014242c:0x3ffeeb70 0x40142592:0x3ffeebb0 0x40142676:0x3ffeebd0 0x4013ea94:0x3ffeebf0 0x40094505:0x3ffeedf0
0x40130a7d: config_llc_hci_funcs_reset at ??:?

0x400899cf: r_lld_pdu_rx_handler at ??:?

0x4008a216: r_rwble_isr at ??:?

0x4008b1d7: r_rwbtdm_isr_wrapper at intc.c:?

0x4008513d: _xt_lowint1 at C:/Users/username/esp/esp4/components/freertos/xtensa_vectors.S:1153

0x4009be02: spi_flash_chip_generic_wait_idle at C:/Users/username/esp/esp4/components/spi_flash/spi_flash_chip_generic.c:269

0x4009bc09: spi_flash_chip_generic_set_write_protect at C:/Users/username/esp/esp4/components/spi_flash/spi_flash_chip_generic.c:216

0x4009c0c1: spi_flash_chip_generic_write at C:/Users/username/esp/esp4/components/spi_flash/spi_flash_chip_generic.c:192

0x4008cbd4: esp_flash_write at C:/Users/username/esp/esp4/components/spi_flash/esp_flash_api.c:615

0x40145a11: esp_partition_write at C:/Users/username/esp/esp4/components/spi_flash/partition.c:389

0x40144fa5: esp_ota_write at C:/Users/username/esp/esp4/components/app_update/esp_ota_ops.c:240
```

saloid, 2020-09-18:
Maybe, need to increase allowed interrupt priority level in sdkconfig option `CONFIG_ESP32_DPORT_DIS_INTERRUPT_LVL`? By default it's 5, but I've set to 1 - let's look how it would be.