espressif/esp-idf #5835: Need update on wifi security feature on esp32 with wpa/wpa2 mixed mode

Mras2an, 2020-09-07:
- [ YES] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ NO] [BUT I WILL TEST ON 4.0 SOON] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ YES] I have searched the issue tracker for a similar issue and not found a similar issue.

## Environment

- Module or chip used:  [ESP32-WROVER]
- IDF version : v3.2.2-132-g7dd492319
- Build System:         [Make]
- Compiler version:   xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     [Linux]
- Using an IDE?: [Yes]
- Power Supply:         [external 5V-500ma]

## Problem Description

Hello,

I work on ESP-IDF V3.3 on an ESP32 board and I have a router in WPA/WPA2 mixed mode.
 
During the wifi association procedure, the router informs the esp32 that it supports WPA or WPA2 security mode, but esp32 chooses WPA mode instead of WPA2 mode by default. Is it possible to update the SDK to choose the most secure mode (WPA2 instead of WPA)?

Thanks for your help.

### Expected Behavior

Choose the most secure mode. Have a wifi connection in WPA2 instead of WPA

### Actual Behavior

We have a wifi connection in WPA

### Steps to reproduce

1. Config the router in WPA+WPA2 like here : https://www.downloads.netgear.com/files/answer_media/media/images/answer_112/wpa+wpa2_security_ul.GIF

2. On GNU/Linux :
    # ifconfig wlan0 down
    # iwconfig wlan0 mode monitor
    # iwconfig wlan0 channel 1
    # ifconfig wlan0 up
    # Open wireshark on the wifi interface

3. Start an association procedure between the ESP32 and the wifi router.

4. On wireshark check the frame "probe response" and check if the router offer WPA+WPA2 (RSN info + wpa info element)

5. On wireshark check the frame "Assoc request" and check if the ESP32 choose WPA2 instead of WPA

Alvin1Zhang, 2020-09-08:
Thanks for reporting, we will look into.

sagb2015, 2020-09-09:
@Mras2an - This has been fixed in the latest version of v3.2. Also, when we experimented, we saw that even if RSN IE is missing WPA IE is using cipher-suite CCMP (AES) and there was 4-way handshake over the air. So the security level is still same.

sagb2015, 2020-09-09:
Closing this issue for now. Feel free to reopen if you see this on latest v3.2 version or any later releases.

closed 2020-09-09 by sagb2015

Mras2an, 2020-09-09:
@sagb2015 Good news, I will test this as soon as possible, Please, may you give me the commit ID of the fix?

sagb2015, 2020-10-20:
@Mras2an  The corresponding commit is https://github.com/espressif/esp-idf/commit/23b3d970997832087133878ce88d2d29eb15b0c7

Mras2an, 2020-10-20:
@all I tested and I confirm that bug is fixed