espressif/esp-idf #5839: Errors while interacting with microSD card (release/v3.3 and release/v4.1)

jacky-kooboo, 2020-09-08:
Hi all,
        Im trying to make a microSD card to work  for esp-idf ,It work fine with microSD 2,but got the following errors for misrcoSD 1 .  Code is copy from esp-idf/examples/storage/sd_card .
       
 Both microSD are work fine with arduino-esp32.
    
[image]

microSD 1:
Name: APPSD
Type: SDHC/SDXC
Speed: 400 kHz
Size: 30500MB

microSD 2:
Name: SD   
Type: SDHC/SDXC
Speed: 400 kHz
Size: 7540MB

ESP-WROOM-32 :https://www.ebay.com.au/itm/AU-ESP-WROOM-32-ESP32-ESP32S-2-4GHz-WiFi-Bluetooth-Development-Board-for-Arduino/254698392032
Micro SD shield : http://www.ebay.com.au/itm/191736124794

I (400307) example: Opening file
E (400337) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0xffffffff
E (400337) diskio_sdmmc: sdmmc_write_blocks failed (-1)
E (400347) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x108
E (400347) diskio_sdmmc: sdmmc_write_blocks failed (264)
I (400357) example: File written
I (400357) example: Renaming file
E (400377) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x108
E (400377) diskio_sdmmc: sdmmc_write_blocks failed (264)
E (400377) example: Rename failed

Anyone with any ideas?
Thanks in advance

igrr, 2020-09-09:
Hi @jacky-kooboo, could you please enable verbose log output in menuconfig (under Component config > Log output > Default log verbosity), and post the full output?

jacky-kooboo, 2020-09-10:
Hi @igrr ，
 

- LOG1 :microSD 1 was plugged in after ESP32  startup  had finished.  the information of microSD was printed,but the operations ared failed. 

- LOG2 :microSD1 was plugged in before ESP32 started.the information of microSD was not printed and the operations ared failed

- LOG3: Log of microSD2 which work fine.

**1. LOG1:**
I (344) example: Card unmounted
I (354) example: Read from file: '1'
I (3354) example: Initializing SD card
I (3354) example: Using SPI peripheral
D (3354) sdspi_host: sdspi_host_init_slot: SPI2 miso=19 mosi=23 sck=18 cs=5 cd=-1 wp=-1, dma_ch=1
D (3354) spi: SPI2 use gpio matrix.
V (3364) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (3364) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x80E
D (3374) intr_alloc: Connected src 30 to int 13 (cpu 0)
D (3384) spi_master: eff: 400, limit: 26666k(/2), 0 dummy, -1 delay
D (3384) spi_master: SPI2: New device added to CS0, effective clock: 400kHz
I (3394) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
D (3404) spi_master: SPI device changed from 3 to 0
D (3404) sdspi_host: start_command_default: poll_cmd_response returned 0x107
D (3414) sdspi_host: sdspi_host_start_command: cmd=52 error=0x107
D (3424) sdmmc_cmd: cmd=52, sdmmc_req_run returned 0x107
D (3444) sdspi_host: start_command_default: poll_cmd_response returned 0x107
D (3444) sdspi_host: sdspi_host_start_command: cmd=0 error=0x107
D (3444) sdmmc_cmd: cmd=0, sdmmc_req_run returned 0x107
D (3444) sdmmc_init: sdmmc_card_init: sdmmc_send_cmd_go_idle_state returned 0x107
D (3454) vfs_fat_sdmmc: sdmmc_card_init failed 0x(107)
I (3464) gpio: GPIO[23]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3474) gpio: GPIO[19]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3484) gpio: GPIO[18]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
V (3484) intr_alloc: esp_intr_free: Disabling int, killing handler
E (3494) example: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place.
I (3504) example: Card unmounted
I (3514) example: Read from file: '1'
I (6514) example: Initializing SD card
I (6514) example: Using SPI peripheral
D (6514) sdspi_host: sdspi_host_init_slot: SPI2 miso=19 mosi=23 sck=18 cs=5 cd=-1 wp=-1, dma_ch=1
D (6514) spi: SPI2 use gpio matrix.
V (6524) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (6524) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x80E
D (6534) intr_alloc: Connected src 30 to int 13 (cpu 0)
D (6544) spi_master: eff: 400, limit: 26666k(/2), 0 dummy, -1 delay
D (6544) spi_master: SPI2: New device added to CS0, effective clock: 400kHz
I (6554) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
D (6564) spi_master: SPI device changed from 3 to 0
D (6564) sdspi_host: start_command_default: poll_cmd_response returned 0x107
D (6574) sdspi_host: sdspi_host_start_command: cmd=52 error=0x107
D (6584) sdmmc_cmd: cmd=52, sdmmc_req_run returned 0x107
D (6624) sdmmc_sd: SDHC/SDXC card
D (6624) sdspi_transaction: cmd=5, R1 response: command not supported
D (6624) sdmmc_cmd: cmd=5, sdmmc_req_run returned 0x106
D (6624) sdmmc_io: sdmmc_init_io: io_send_op_cond (1) returned 0x106; not IO card
D (6634) sdspi_host: data CRC set=1
D (6704) sdmmc_common: host_ocr=0x40ff8000 card_ocr=0xc0ff8000
D (6704) sdmmc_common: sdmmc_card_init: host_ocr=40ff8000, card_ocr=c0ff8000
D (6704) sdmmc_init: sdmmc_card_init: card type is SD
D (6714) sdmmc_common: sdmmc_init_host_frequency: using 400 kHz bus frequency
D (6714) vfs_fat_sdmmc: using pdrv=0
Name: APPSD
Type: SDHC/SDXC
Speed: 19 MHz
Size: 30500MB_
I (6744) example: Opening file
D (6764) sdspi_host: sdspi_host_start_command: cmd=24 error=0x108
D (6764) sdmmc_cmd: cmd=24, sdmmc_req_run returned 0x108
E (6764) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x108
[... 542 lines cut ...]
D (294) sdmmc_sd: SDHC/SDXC card
D (294) sdspi_transaction: cmd=5, R1 response: command not supported
D (294) sdmmc_cmd: cmd=5, sdmmc_req_run returned 0x106
D (294) sdmmc_io: sdmmc_init_io: io_send_op_cond (1) returned 0x106; not IO card
D (304) sdspi_host: data CRC set=1
D (314) sdmmc_common: host_ocr=0x40ff8000 card_ocr=0xc0ff8000
D (314) sdmmc_common: sdmmc_card_init: host_ocr=40ff8000, card_ocr=c0ff8000
D (314) sdmmc_init: sdmmc_card_init: card type is SD
D (324) sdmmc_common: sdmmc_init_host_frequency: using 400 kHz bus frequency
D (334) vfs_fat_sdmmc: using pdrv=0
Name: SD   
Type: SDHC/SDXC
Speed: 19 MHz
Size: 7540MB
I (364) example: Opening file
E (564) example: fprintf error:ERROR
I (564) example: File written
I (644) example: Renaming file
I (744) example: Reading file
I (754) example: Read from file: 'Hello SD   !'
I (754) gpio: GPIO[23]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (764) gpio: GPIO[19]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (774) gpio: GPIO[18]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
V (784) intr_alloc: esp_intr_free: Disabling int, killing handler
I (784) example: Card unmounted

jacky-kooboo, 2020-09-11:
Hi @igrr ,
      When I turn off the CRC(CMD59 ,CRC_ON_OFF) option ,both microSD are work fine.
``` c
esp_err_t sdmmc_init_spi_crc(sdmmc_card_t* card)
{
    /* In SD mode, CRC checks of data transfers are mandatory and performed
     * by the hardware. In SPI mode, CRC16 of data transfers is optional and
     * needs to be enabled.
     */
    assert(host_is_spi(card));
    esp_err_t err = sdmmc_send_cmd_crc_on_off(card, false);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "%s: sdmmc_send_cmd_crc_on_off returned 0x%x", __func__, err);
        return err;
    }
    return ESP_OK;
}
```

igrr, 2020-09-16:
@jacky-kooboo If turning of CRC makes the error disappear, it is possible that the data you are writing to the card contains errors.

If you later read back this data, does it come back as expected, or is there any difference?

closed 2022-02-17 by Alvin1Zhang