espressif/esp-idf #5840: V4.0.1 esp -idf BLE causing crashes

sagarvegg, 2020-09-08:
----------------------------- Delete below -----------------------------

COMMIT : #5626 

Hey, We are developing one commercial system and using ESP32 as our core hardware,
We really admire how the esp32 has revolutionized the whole segment,

Problem is we are getting ISR WDT timeout ate rwble.c , ASSERT_PARAM(64 0).
 
The real problem is we need to solve this error anyhow in order to move our work forward but we can not do anything because there is no access of rwble.c file.

1) I will share sdkconfig file below. for any reason we can not share elf file.
2) We are able to solve the error by ourself. But for whatsever reason esp32 hasn't open sourced the core files yet.

Possible problem.

We are using the freertos tasks and xqueues, We have tried everyway but can not find a solution.

The only problem is errors are randoms, like they stuck random. It's been 2018 since this kind of is being introduced, i wonder the issues are still present. We can not move our product into production until we resolve this issues. And we really are out of time. 

So i am creating one thread here. Which is to solve all rwble.c random errors from bottom to scratch. We are willing to help anybody else for our expertise as well.

So do provide a possible solutions which might help us or anyone.

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5188167/sdkconfig.txt)

- Module or chip used:  ESP32 WROVER 8MB PSRAM
- IDF version (run ``git describe --tags`` to find it):
   4.0.1
- Build System:   CMAKE

```
Core 1 register dump:
PC      : 0x40086865  PS      : 0x00060034  A0      : 0x8008aa60  A1      : 0x3ffbea20  
0x40086865: r_assert_param at ??:?

A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054  
A6      : 0x3ffbdbe4  A7      : 0x60008050  A8      : 0x80086865  A9      : 0x3ffbea00  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4008679d  LEND    : 0x400867a4  LCOUNT  : 0x00000000  
0x4008679d: r_assert_param at ??:?

0x400867a4: r_assert_param at ??:?

Core 1 was running in ISR context:
EPC1    : 0x400e3046  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40086865
0x400e3046: SimpleTimer::run() at /home/sagar/Desktop/esp/esp-idf/examples/get-started/blink/build/../components/arduino/libraries/SimpleTimer/SimpleTimer.cpp:60 (discriminator 2)

0x40086865: r_assert_param at ??:?

ELF file SHA256: 903960294ddf9b36

Backtrace: 0x40086862:0x3ffbea20 0x4008aa5d:0x3ffbea40 0x4008b677:0x3ffbea60 0x4008532d:0x3ffbea80 0x400e3043:0x3ffd7360 0x400eed0d:0x3ffd7380 0x400db453:0x3ffd7410 0x400d990b:0x3ffd7440 0x400d9925:0x3ffd7460 0x40099f29:0x3ffd7480
0x40086862: r_assert_param at ??:?

0x4008aa5d: r_rwble_isr at ??:?

0x4008b677: r_rwbtdm_isr_wrapper at intc.c:?

0x4008532d: _xt_lowint1 at /home/sagar/Desktop/esp/esp-idf/components/freertos/xtensa_vectors.S:1153

0x400e3043: SimpleTimer::run() at /home/sagar/Desktop/esp/esp-idf/examples/get-started/blink/build/../components/arduino/libraries/SimpleTimer/SimpleTimer.cpp:90

0x400eed0d: blue::handle() at /home/sagar/Desktop/esp/esp-idf/examples/get-started/blink/build/../components/arduino/libraries/IOTMasterBoard/src/blue.cpp:529

0x400db453: handleLoop() at /home/sagar/Desktop/esp/esp-idf/examples/get-started/blink/build/../components/arduino/libraries/IOTMasterBoard/src/IOTMasterBoard.cpp:138

0x400d990b: loop() at /home/sagar/Desktop/esp/esp-idf/examples/get-started/blink/build/../main/blink.cpp:179

0x400d9925: loopTask(void*) at /home/sagar/Desktop/esp/esp-idf/examples/get-started/blink/build/../main/blink.cpp:190

0x40099f29: vPortTaskWrapper at /home/sagar/Desktop/esp/esp-idf/components/freertos/port.c:143

Core 0 register dump:
PC      : 0x4008432a  PS      : 0x00060f34  A0      : 0x80091a5a  A1      : 0x3ffdb4f0  
0x4008432a: esp_clk_cpu_freq at /home/sagar/Desktop/esp/esp-idf/components/esp32/clk.c:150

A2      : 0x3ffe01e0  A3      : 0x00bbb0ec  A4      : 0x00bdb6c0  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x80092049  A9      : 0x3ffdb600  
A10     : 0x3ffdb620  A11     : 0x3ffdb62e  A12     : 0x00000031  A13     : 0x00060f23  
A14     : 0x00000000  A15     : 0x3ffdb570  SAR     : 0x00000004  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4009377d  LEND    : 0x4009378d  LCOUNT  : 0xffffffff  
0x4009377d: bzero at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/string/bzero.c:31

0x4009378d: __submore at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/ungetc.c:82

ELF file SHA256: 903960294ddf9b36

Backtrace: 0x40084327:0x3ffdb4f0 0x40091a57:0x3ffdb510 0x40091686:0x3ffdb530 0x40091937:0x3ffdb550 0x40091d85:0x3ffdb580 0x400909e5:0x3ffdb5b0 0x40090b85:0x3ffdb5e0 0x400911fc:0x3ffdb600 0x40092046:0x3ffdb620 0x4009b054:0x3ffdb660 0x400dcd22:0x3ffdb6a0 0x40099f29:0x3ffdb6f0
0x40084327: esp_clk_cpu_freq at /home/sagar/Desktop/esp/esp-idf/components/esp32/clk.c:150

0x40091a57: esp_apptrace_tmo_check at /home/sagar/Desktop/esp/esp-idf/components/app_trace/app_trace_util.c:29

0x40091686: esp_apptrace_trax_down_buffer_get at /home/sagar/Desktop/esp/esp-idf/components/app_trace/app_trace.c:610

0x40091937: esp_apptrace_read at /home/sagar/Desktop/esp/esp-idf/components/app_trace/app_trace.c:940

0x40091d85: SEGGER_RTT_ReadNoLock at /home/sagar/Desktop/esp/esp-idf/components/app_trace/sys_view/esp32/SEGGER_RTT_esp32.c:120

0x400909e5: _HandleIncomingPacket at /home/sagar/Desktop/esp/esp-idf/components/app_trace/sys_view/SEGGER/SEGGER_SYSVIEW.c:461

0x40090b85: _SendPacket at /home/sagar/Desktop/esp/esp-idf/components/app_trace/sys_view/SEGGER/SEGGER_SYSVIEW.c:739

0x400911fc: SEGGER_SYSVIEW_SendPacket at /home/sagar/Desktop/esp/esp-idf/components/app_trace/sys_view/SEGGER/SEGGER_SYSVIEW.c:2198

0x40092046: SYSVIEW_RecordU32x4 at /home/sagar/Desktop/esp/esp-idf/components/app_trace/sys_view/Sample/OS/SEGGER_SYSVIEW_FreeRTOS.c:254

0x4009b054: xQueueGenericReceive at /home/sagar/Desktop/esp/esp-idf/components/freertos/queue.c:1551

0x400dcd22: vTHistLstSerEveSendPacket(void*) at /home/sagar/Desktop/esp/esp-idf/examples/get-started/blink/build/../components/arduino/libraries/IOTMasterBoard/src/tasks.cpp:187 (discriminator 1)

0x40099f29: vPortTaskWrapper at /home/sagar/Desktop/esp/esp-idf/components/freertos/port.c:143
```

Alvin1Zhang, 2020-09-09:
Thanks for reporting and letting us know, we will look into.

chhajedji, 2020-09-09:
@sagarvegg Please provide the IDF commit which you are using. It would be helpful if you can give bootup logs also.

chhajedji, 2020-09-09:
ASSERT (64, 0) has been resolved in [this](https://github.com/espressif/esp32-bt-lib/commit/0ec0b596be39d6dbd7fb765f7fbfb76b9b8f0f71) commit on master and [this](https://github.com/espressif/esp32-bt-lib/commit/567dacc973329fe34f6a48c15ffffcb28150a85a) commit on release/v4.0 branch. Please see if you have this commit in your working tree. This commit should be present in submodule `$IDF_PATH/components/bt/controller/lib/`

sagarvegg, 2020-09-09:
[bootup_log.txt](https://github.com/espressif/esp-idf/files/5196212/bootup_log.txt)

sagarvegg, 2020-09-10:
@chhajedji here our bootup logs file

[bootup_log.txt](https://github.com/espressif/esp-idf/files/5199723/bootup_log.txt)

sagarvegg, 2020-09-10:
> ASSERT (64, 0) has been resolved in [this](https://github.com/espressif/esp32-bt-lib/commit/0ec0b596be39d6dbd7fb765f7fbfb76b9b8f0f71) commit on master and [...]

It's not fixed. we have checked

chhajedji, 2020-09-10:
Do you mean you have the fixing commit in your git working tree and still you face this issue or something else?

akshar001, 2020-09-11:
Yes! Same issue @chhajedji. By the way i and @sagarvegg work in a same team

akshar001, 2020-09-11:
For your being sured,
https://github.com/espressif/esp-idf/tree/release/v4.0

This is a branch we are using.
We have re cloned this branch exactly 13 days ago. which is the latest commit i think

akshar001, 2020-09-11:
We can not share our real code, but we will make one code which has the same amount of task that we are running and let's see you can reproduce the issue

akshar001, 2020-09-11:
Can't we get an access to rwble.c code?
We can help to fix.

sagarvegg, 2020-09-15:
@chhajedji @Alvin1Zhang we have made one separate code which replicates the same behavior you guys can download it and run it from your side to find the actual problem.

[testing.zip](https://github.com/espressif/esp-idf/files/5223041/testing.zip)

blueMoods, 2020-09-16:
@akshar001 @sagarvegg  Sorry for that rw_ble.c is not open to the outside.

I'm trying to reproduce it with your code.

And could you please provide the commit of esp-idf which you used and a whole log?

blueMoods, 2020-09-16:
And how to reproduce it ?
I have fix some build errors and run it on ESP32,  it seems like init bluetooth and start adv success, than do nothing. Than what should I do? Just waiting it crash?

here is the logs
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6996
load:0x40078000,len:14724
ho 0 tail 12 room 4
load:0x40080400,len:4292
entry 0x400806e4
I (71) boot: Chip Revision: 1
I (72) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot: ESP-IDF v4.0.1-441-g6093407 2nd stage bootloader
I (40) boot: compile time 12:01:15
I (40) boot: Enabling RNG early entropy source...
I (46) boot: SPI Speed      : 40MHz
I (50) boot: SPI Mode       : DIO
I (54) boot: SPI Flash Size : 4MB
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x222d8 (139992) map
I (161) esp_image: segment 1: paddr=0x00032300 vaddr=0x3ffbdb60 size=0x034c0 ( 13504) load
I (167) esp_image: segment 2: paddr=0x000357c8 vaddr=0x40080000 size=0x00400 (  1024) load
I (168) esp_image: segment 3: paddr=0x00035bd0 vaddr=0x40080400 size=0x0a440 ( 42048) load
I (194) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x7b784 (505732) map
I (374) esp_image: segment 5: paddr=0x000bb7a4 vaddr=0x4008a840 size=0x0a508 ( 42248) load
I (405) boot: Loaded app from partition at offset 0x10000
I (405) boot: Disabling RNG early entropy source...
I (405) cpu_start: Pro cpu up.
I (409) cpu_start: Application information:
I (414) cpu_start: Project name:     testing
I (419) cpu_start: App version:      1
I (423) cpu_start: Compile time:     Sep 16 2020 12:01:14
I (429) cpu_start: ELF file SHA256:  147b82000cd757bb...
I (435) cpu_start: ESP-IDF:          v4.0.1-441-g6093407
I (441) cpu_start: Starting app cpu, entry point is 0x4008124c
I (0) cpu_start: App cpu up.
I (452) heap_init: Initializing. RAM available for dynamic allocation:
I (458) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (464) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (470) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (477) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (483) heap_init: At 3FFCA790 len 00015870 (86 KiB): DRAM
I (489) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (495) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (502) heap_init: At 40094D48 len 0000B2B8 (44 KiB): IRAM
I (508) cpu_start: Pro cpu start user code
I (526) spi_flash: detected chip: generic
I (527) spi_flash: flash io: dio
I (527) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (536) uart: queue free spaces: 20
I (536) uart: queue free spaces: 20
I (546) uart: queue free spaces: 20
Hex of _frst_data_queue 3ffce1cc
Hex of _scnd_data_queue 3ffce274
Hex of _thrd_data_queue 3ffce59c
ready to prepare all tasks
task for first data packets --> created
task for third data packets --> created
I (2046) BTDM_INIT: BT controller compile version [b1eb377]
I (2066) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (3706) phy: phy_version: 4370, 4e803b3, Aug 11 2020, 14:18:07, 0, 0
I (3946) SEC_GATTS_DEMO: app_main init bluetooth
I (4076) SEC_GATTS_DEMO: The number handle = 8
I (4106) SEC_GATTS_DEMO: advertising start success
I (4296) IBEACON_DEMO: register callback
```

akshar001, 2020-09-16:
Yes wait for 1 hour 2-hour for crash

blueMoods, 2020-09-16:
It have run more about 10 hours, and nothing happened.
could you please provide the commit of esp-idf which you used and the complete log?

akshar001, 2020-09-17:
@blueMoods we have used 4.1 and it's running fine. You guys can close this issue

Alvin1Zhang, 2020-09-18:
Thanks for the feedback.

closed 2020-09-18 by Alvin1Zhang