espressif/esp-idf #5841: ESP32-S2-Saola-1 can't flash with IDF4.1 on ESP32S2 Beta

techtoys, 2020-09-08:
# Operating system
Win7 pro 64-bit
# Python version
3.7.3

# What Chip
ESP32S2Beta
Features: Engineering Sample

# What development board or other hardware is the chip attached to
ESP32-S2-Saola-1

# Is anything else attached to the development board, except for the serial flasher connections?
Nothing is connected.

# Are you running esptool.py from an IDE such as Arduino or Eclipse?
No IDE. Running esptool.py from Windows command prompt

# Full esptool.py command line that was run:
idf.py -p COM92 flash

# Full output from esptool.py
Adding "flash"'s dependency "all" to list of commands with default set of option
s.
Executing action: all (aliases: build)
Running ninja in directory c:\users\john\esp\esp32-s2-parallel-master\build
Executing "ninja all"...
[1/4] cmd.exe /C "cd /D C:\Users\John\...*************************************"
Partition table binary generated. Contents:
*******************************************************************************
Espressif ESP32 Partition Table
Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[2/4] Performing build step for 'bootloader'
ninja: no work to do.
Executing action: flash
Running esptool.py in directory c:\users\john\esp\esp32-s2-parallel-master\build

Executing "C:\Users\John\.espressif\python_env\idf4.1_py3.7_env\Scripts\python.e
xe C:\esp-idf-4.1\components/esptool_py/esptool/esptool.py -p COM92 -b 460800 --
before default_reset --after hard_reset --chip esp32s2beta write_flash @flash_pr
oject_args"...
esptool.py -p COM92 -b 460800 --before default_reset --after hard_reset --chip e
sp32s2beta write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000
 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 pa
rallel.bin
esptool.py v2.9-dev
Serial port COM92
Connecting....
Chip is ESP32S2 Beta
Features: Engineering Sample
Crystal is 40MHz
MAC: 7c:df:a1:00:e9:28
Uploading stub...
Running stub...

A fatal error occurred: Invalid head of packet (0x50)
esptool.py failed with exit code 2

# Do you have any other information from investigating this?
Tried with hello-world example with the same flash error. The problem is not compile related.
I did press and hold BOOT key and click on RST key onboard to bring up the bootloader program but still not successful

# Is there any other information you can think of which will help us reproduce this problem?
I have no clue. Also tried `idf.py set-target esp32s2beta` and set `CONFIG_ESP32_BETA_CHIP_VERSION` under menuconfig but there is no success.

chegewara, 2020-09-09:
`idf.py settarget esp32s2`
but at first delete build folder.

igrr, 2020-09-09:
Hi @techtoys, it looks like you are using the final production version of ESP32-S2, not the engineering sample (ESP32-S2 beta).

* The development board with ESP32-S2 beta (engineering sample) looks like the one shown in https://medium.com/hacksters-blog/a-first-look-at-the-beta-esp32-s2-development-kits-3fb6bf01a55c. It has silkscreen marking "ESP32-S2 Beta Devkit-C v1.x"
* The development board with final production version of ESP32-S2 looks like this: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/_images/esp32-s2-saola-1-v1.2-isometric.png.

Another way to check the chip version is to run:
```
esptool.py --chip esp32 --no-stub --port PORT read_mem 0x40001000
```
This should report:

* 0x9540fe78 for ESP32-S2 beta
* 0x000007c6 for ESP32-S2

As mentioned in the docs, ESP-IDF v4.1 only supports ESP32-S2 beta silicon. IDF v4.2 and later will support the final production version of ESP32-S2.

If your chip is the final production version of ESP32-S2, please switch to `release/v4.2` or `master` branch, and use `esp32s2` target instead of `esp32s2beta`.

If your chip actually is ESP32-S2 beta, based on the above checks, then we will need to troubleshoot this further...

techtoys, 2020-09-09:
Hi @igrr, thank you. It turns out to be an esp32s2. I installed IDF4.2dev and get the EVB successfully programmed.

closed 2020-09-09 by techtoys