espressif/esp-idf #5842: Download error while set up the env with the install.sh

WuYingzheng, 2020-09-09:
I will get an SSL error when I try to set up the environment with the script install.sh

i use python3.
`
❯ ./install.sh
`
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Installing xtensa-esp32-elf@esp-2020r2-8.2.0
Downloading xtensa-esp32-elf-gcc8_2_0-esp-2020r2-linux-amd64.tar.gz to /home/wyz/.espressif/dist/xtensa-esp32-elf-gcc8_2_0-esp-2020r2-linux-amd64.tar.gz.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:852)>
WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp-2020r2-linux-amd64.tar.gz to /home/wyz/.espressif/dist/xtensa-esp32-elf-gcc8_2_0-esp-2020r2-linux-amd64.tar.gz.tmp

The following code will solve this issue. Can you merge it into mater? It will save a lot of time for people who also enconter this problem.
`
import ssl
ssl._create_default_https_context = ssl._create_unverified_context
`
Thank you.

igrr, 2020-09-09:
Hi @WuYingzheng, thanks for posting this solution.

While it does solve the problem, we would not like to suggest our users to ignore certificate validation issues, especially when downloading executable files from the Internet, as [nasty things may happen.](https://github.com/secretsquirrel/BDFProxy)

Usually this error is caused by an outdated set of TLS root certificates. If you share which Linux distribution (and which version) you use, and also how you installed Python 3, I can give some suggestions about updating the TLS root certificate bundle.

WuYingzheng, 2020-09-21:
Thank you for your reply!

I use window sub-system linux. Now i have migrated to ubuntu.

closed 2020-09-21 by WuYingzheng