espressif/esp-idf #5843: if app partition has encrypted flag but encryption is not enabled then OTA will fail

bill-iot-hw, 2020-09-09:
I'm using a custom partition table csv, when I have the encrypted flags set on all partitions (except NVS) but encryption is not enabled then OTA updates fail with the following messages:
`E (447898) esp_image: image at 0x120000 has invalid magic byte
E (447898) boot_comm: mismatch chip ID, expected 0, found 6987
E (447908) boot_comm: can't run on lower chip revision, expected 1, found 175
W (447918) esp_image: image at 0x120000 has invalid SPI mode 71`
 I am using ESP-IDF 3.3
Docs imply that setting the encrypted flag on an app partition is redundant
"Only one flag is currently supported, encrypted. If this field is set to encrypted, this partition will be encrypted if Flash Encryption is enabled.

(Note that app type partitions will always be encrypted, regardless of whether this flag is set or not.)"

But I find in practice that if the encrypted flag is present then OTA updates fail. 

Removing the encrypted flag resolves the issue.

I would expect that if encryption is not enabled then OTA would ignore the encrypted flags in the partition csv and OTA would succeed.

mahavirj, 2020-09-09:
@bill-iot-hw This was addressed in https://github.com/espressif/esp-idf/pull/3328 (available `release/v4.0` onwards). Can you please check if it helps in your case as well?

bill-iot-hw, 2020-09-09:
Unfortunately I won't have time in the near future to check since this project is moving to production on release 3.3. However it's not a show stopper nor do I think a fix is really needed since it's been fixed in future versions. Really just more of an oddity I ran into during some testing.

 I'll close the issue  since it's been addressed in future version and the work-around is simply to remove the encrypted flags from the partition table if not using encryption.

closed 2020-09-09 by bill-iot-hw