espressif/esp-idf #5844: Init WiFi STA instead of AP

saloid, 2020-09-09:
## Environment

- IDF version (run ``git describe --tags`` to find it): master branch

## Problem Description

Is it ok that in both STA and AP init functions ([here](https://github.com/espressif/esp-idf/blob/master/components/tcpip_adapter/tcpip_adapter_compat.c#L61) and [here](https://github.com/espressif/esp-idf/blob/master/components/tcpip_adapter/tcpip_adapter_compat.c#L73)) `esp_wifi_set_default_wifi_sta_handlers()` is used? Maybe in AP init should be used `esp_wifi_set_default_wifi_ap_handlers()`?

liuzfesp, 2020-09-09:
Thanks @saloid for your reporting, it should be our bug, will check it.

AxelLin, 2020-09-18:
It's fixed by edb8b91efc12d1adc43072018d42c6149440ca00 in master tree.

closed 2020-09-18 by saloid