espressif/esp-idf #5846: Problem with ESP32 Wrover-B with wifi TCP/IP connection - high latency and anser to cloud connection in HTTP/POST-REST

filzek, 2020-09-10:
Using release v4.1 the wifi seens to be stable and we havent seen disconnection but the TCP/IP stack seens to become lazy and with a very high latency to answer, connection to a http request takes longer and longer until connection become very unstable even with very good rssi and wifi connection.

Seens that somehow the TCP/IP stack and wifi arent working correct.

We will try to reproduce the issues and report it more complete tomorrow, but the problem exist so far.

Alvin1Zhang, 2020-10-20:
@filzek Thanks for reporting, would you please help share if more updates for this issue? Thanks.

filzek, 2020-11-08:
We will prepare a working example for this problem so can test it and check where problem relays.

Alvin1Zhang, 2020-11-09:
Thanks for the updates.

lhespress, 2020-11-26:
@filzek Does this issue still exist?

maikeriva, 2020-12-12:
I am experiencing similar problems on IDF v4.0.2. I performed various tests on my custom code and high latency seems to occur within the lwip or wifi stack. Interestingly enough, that does seem to happen only when the ESP module receives data, and not when it sends to the server.

Will try to create a working code stub in the next days.

lhespress, 2020-12-14:
@maikeriva 

Thanks for the updates, you also can check it on IDF v4.2. It's better to share the code when reproduce the issue again.

maikeriva, 2021-01-04:
Turns out in my case that turning off WiFi power saving did the trick.

`esp_wifi_set_ps(WIFI_PS_NONE)`

It was particularly hard to find out since the latency issue does not appear when stress-testing by firing multiple requests to the ESP32, but only with sparse requests every now and then.

filzek, 2021-01-05:
I can try it tomorrow to check performance and later return for the results, as far latest v4.2 release still having this problem so far, but lets test this solution idea.

filzek, 2021-01-05:
Problem was that is almost impossible to remove the code to provide one working example, but we test with IIS, NGIX and some other and the problem happen randomly.

We have found out that the answer when the problem happen from the server is correct but the esp32 reads ZERO in the answer, so is something with internal lwip and esp32, so the package was really lost and the content was unread.

lhespress, 2021-01-05:
@filzek the esp32 reads ZERO means that the connection is inactive, Could you enable the `SOCKETS_DEBUG` in `lwipopts.h` file and test again? if output the log as follows:

```
lwip_recv_tcp: p == NULL, error is "Connection closed."!
```
It means that the server close the connection actively.

filzek, 2021-01-05:
Hi @lhespress I have enable it in:
esp-idf\components\lwip\port\esp32\include\lwipopts.h

 #define SOCKETS_DEBUG                   LWIP_DBG_ON

Is that right or do I need to change it somewhere else?

filzek, 2021-01-05:
Log says:

lwip_recv_tcp: netconn_recv err=-3, pbuf=0x0
lwip_recv_tcp: p == NULL, error is "Timeout."!
lwip_recvfrom(55):  addr=xxx.xxx.xxx.xxx port=80 len=-1
lwip_close(55)
SIZE OF RECEIVED [0] DATA[] is not higher than 0 so, will restart loop

lhespress, 2021-01-07:
@filzek  It shows that recv data timeout from the log, the recvmbox maybe full, could you modify the `CONFIG_LWIP_TCP_RECVMBOX_SIZE`, `CONFIG_LWIP_TCPIP_RECVMBOX_SIZE` and check again?

lhespress, 2021-01-07:
@filzek In addition to that, what did you call `setsockopt` on `SO_RCVTIMEO` for change the recv timeout value from zero (default) to the other value.

lhespress, 2021-01-07:
In addition to that you also can modify the timeout value as follow steps:
```
esp_http_client_config_t config = {
        .host = "httpbin.org",
        .path = "/get",
        .query = "esp",
        .event_handler = _http_event_handler,
        .user_data = local_response_buffer,        // Pass address of local buffer to get response
        .disable_auto_redirect = true,
        .timeout_ms = 15000,
    };
   esp_http_client_handle_t client = esp_http_client_init(&config);
```

Alvin1Zhang, 2021-08-19:
Thanks for reporting, feel free to reopen.

closed 2021-08-19 by Alvin1Zhang

baldhead69, 2021-08-28:
So, how was this problem solved ?

filzek, 2021-08-28:
Not sure how we solve it, but it was solved

lhespress, 2021-08-30:
@filzek 

Did you modify the timeout value as follow steps:

```
struct timeval receiving_timeout;
        receiving_timeout.tv_sec = 15;
        receiving_timeout.tv_usec = 0;
        if (setsockopt(s, SOL_SOCKET, SO_RCVTIMEO, &receiving_timeout,
                sizeof(receiving_timeout)) < 0) {
            ESP_LOGE(TAG, "... failed to set socket receiving timeout");
            close(s);
            vTaskDelay(4000 / portTICK_PERIOD_MS);
            continue;
        }
```

@baldhead69 Could you check it again?

baldhead69, 2021-08-30:
Where to put this ?