espressif/esp-idf #5848: [ESP-BLE-MESH] Battery Server displays blank in nRF MESH

jpaulruiz, 2020-09-10:
Is battery model in ESP-BLE-MESH functional?

I created a server and client for battery model and the display from nRF MESH is blank for the battery server.

//=============================================
ESP_BLE_MESH_MODEL_PUB_DEFINE(battery_srv_pub,2 + 3, ROLE_NODE);  /*battery server model*/
static esp_ble_mesh_gen_battery_srv_t gen_battery_srv = {
	.rsp_ctrl.get_auto_rsp = ESP_BLE_MESH_SERVER_AUTO_RSP,
	.rsp_ctrl.set_auto_rsp = ESP_BLE_MESH_SERVER_AUTO_RSP,
};

ESP_BLE_MESH_MODEL_PUB_DEFINE(battery_cli_pub, 2 + 1, ROLE_NODE); /*battery client model*/
//=============================================

static esp_ble_mesh_model_t root_models[] = {
    ESP_BLE_MESH_MODEL_CFG_SRV(&config_server),
	ESP_BLE_MESH_MODEL_GEN_ONOFF_SRV(&onoff_pub_0, &onoff_server_0),
	ESP_BLE_MESH_MODEL_GEN_ONOFF_CLI(&onoff_cli_pub_0, &client),
	ESP_BLE_MESH_MODEL_GEN_BATTERY_SRV(&battery_srv_pub,&gen_battery_srv),
	ESP_BLE_MESH_MODEL_GEN_BATTERY_CLI(&battery_cli_pub, &client),
};

Campou, 2020-09-11:
Hi @jpaulruiz 

Please define two different `esp_ble_mesh_client_t` in the example, one for onoff client model, and the other for battery client model, like:
```
static esp_ble_mesh_client_t onoff_client;
static esp_ble_mesh_client_t battery_client;
```

I have tried with the nRF Mesh App (Android, version 2.4.1), it could display both battery client & battery server models.

Thanks.

jpaulruiz, 2020-09-11:
Hi @Campou 

I can display both the battery server and client on nRF. However, when I clicked the battery server it displays nothing unlike the generic Onoff model that has OnOff toggle on nRF.

I have assumed that the battery server displays data like battery indicator.

Thanks.

Campou, 2020-09-21:
Hi @jpaulruiz 

Thanks for the update, I think it's just the behavior of the App. So about this issue, can we close it for now? And for any other issues, please feel free to create other issues for them.

Thanks.

closed 2020-09-21 by jpaulruiz