espressif/esp-idf #5850: BT Classic Bond not added to list when two ESP32 modules are connected

JHedgesBartec, 2020-09-10:
## Environment

    Module or chip used: ESP32-WROOM-32
    IDF version: v4.0.1 Release
    Build System: CMake
    Compiler version SysGcc 8.2.0
    Operating System: Windows
    (Windows only) environment type: IDE below for build process
    IDE: VisualStudio and VisualGDB
    Power Supply: external 3.3V

## Problem Description

Bond not added to BT Classic GAP list when two ESP32 modules are connected with a SPP profile.

When connecting a ESP32 to ANY other device the bond is created, this is regardless of who initiates the connection.

We are using the modules in dual master/slave mode where they have a graphical interface to connect to other devices. Therfore both the server and client is used for the SPP profile.

### Expected Behavior

Bond would be returned from esp_bt_gap_get_bond_device_list()

### Actual Behavior

No bond is added to the list

### Steps to reproduce

1. Create a project which contains both a acceptor and initiator
2. Put the same code on both ESP32 modules (with different names)
3. Using one ESP32 trigger a discovery and SPP connect under BT classic.
4. Modules connect and data can be transmitted
5. Break the connection
6. Request the bond list from both devices - no bond added

### Code to reproduce this issue
Unable to provide source.

Alvin1Zhang, 2020-09-11:
Thanks for reporting, we will look into.

Wth-Esp, 2020-09-11:
Hi,

Have you initialize the first variable of this `esp_bt_gap_get_bond_device_list`? 

`dev_num` should be initialized before calling this API, like `int dev_num =10;`.

Thanks
Wei Tianhua

JHedgesBartec, 2020-09-11:
> 
> 
> Hi,
> 
> Have you initialize the first variable of this `esp_bt_gap_get_bond_device_list`?
> 
> `dev_num` should be initialized before calling this API, like `int dev_num =10;`.
> 
> Thanks
> Wei Tianhua

Hi @Wth-Esp,

Yes, as stated in the issue description `esp_bt_gap_get_bond_device_list()` works correctly with other devices.
For example pairings with Android devices, WT12 bluegiga modules, baremetal linux devices and windows PCs all pair 'OK' (https://github.com/espressif/esp-idf/issues/5384) and are shown in the bond list.

Regards,
Jack.

JHedgesBartec, 2020-09-11:
Further to the above I have now got a debug trace from both ESP32 modules from the process up until the `ESP_SPP_DISCOVERY_COMP_EVT` where the device bond is checked and is usually present.

## Device initiating connection (still running as both client and server)
```
I (17431) BT_BTM: BTM_StartInquiry: mode: 1, dur: 8, rsps: 0, flt: 0

D (17431) BT_BTM: BTM_StartInquiry: p_inq->inq_active = 0x01

D (17433) BT_HCI: HCI Enqueue Comamnd opcode=0xc05

D (17441) BT_HCI: Receive packet event_code=0xe

D (17443) BT_HCI: btu_hcif_command_complete_evt

D (17448) BT_BTM: btm_acl_update_busy_level

D (17452) BT_BTM: BTM_BLI_INQ_EVT

D (17456) BT_HCI: HCI Enqueue Comamnd opcode=0x401

I (17461) CL_GAP_CB: event 1
D (17463) BT_HCI: Receive packet event_code=0xf

I (17469) CL_GAP_CB: Discovery started.
D (22470) BT_HCI: Receive packet event_code=0x2f

I (22470) BT_BTM: BTM_CheckEirData type=0x03

I (22470) BT_BTM: btm_set_eir_uuid eir_complete_list=0x01

I (22475) BT_BTM: BTM_CheckEirData type=0x05

I (22480) BT_BTM: BTM_CheckEirData type=0x07

I (22485) BT_BTM: BTM_InqDbRead: bd addr [c44f336053bf]

I (22492) BT_BTM: BTM_CheckEirData type=0x09

I (22497) CL_GAP_CB: event 0
I (22500) BT_BTM: BTM_CheckEirData type=0x09

D (22821) BT_HCI: Receive packet event_code=0x2f

D (22822) BT_BTM: BDA seen before [c44f 3360 53bf]

D (22822) BT_BTM: update RSSI new:-45, old:-44

I (22825) BT_BTM: BTM_CheckEirData type=0x03

I (22830) BT_BTM: btm_set_eir_uuid eir_complete_list=0x01

I (22836) BT_BTM: BTM_CheckEirData type=0x05

I (22841) BT_BTM: BTM_CheckEirData type=0x07

I (22846) BT_BTM: BTM_InqDbRead: bd addr [c44f336053bf]

I (22853) BT_BTM: BTM_CheckEirData type=0x09

I (22858) CL_GAP_CB: event 0
I (22861) BT_BTM: BTM_CheckEirData type=0x09

D (22883) BT_HCI: Receive packet event_code=0x2f

D (22883) BT_BTM: BDA seen before [c44f 3360 53bf]

D (22883) BT_BTM: update RSSI new:-45, old:-45

I (22886) BT_BTM: BTM_CheckEirData type=0x03

[... 1444 lines cut ...]

D (40977) BT_BTM: btm_acl_report_role_change

D (40982) BT_BTM: btm_acl_update_busy_level

D (40986) BT_BTM: BTM_BLI_ACL_DOWN_EVT

D (40990) BT_BTM: acl hci_handle=128 transport=1 connectable_mode=0x0 link_role=1

D (40998) BT_BTM: before update p_dev_rec->sec_flags=0x88

D (41003) BT_BTM: Bletooth link down

D (41007) BT_BTM: after update p_dev_rec->sec_flags=0x88

D (41013) BT_BTM: btm_acl_resubmit_page

D (41017) BT_BTM: btm_sec_disconnected before update sec_flags=0x88

D (41023) BT_BTM: btm_sec_disconnected after update sec_flags=0x88

D (41030) BT_BTM: btm_get_acl_disc_reason_code

I (41035) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
```

boblane1, 2020-10-28:
HI, @JHedgesBartec 

According to your trace debug and description, it seems that these two devices only initiated the SDP procedure, so it were not added to the bond list. After these two devices establish the SPP connection for the first time, they will be paired and bonded.

Alvin1Zhang, 2021-01-21:
Thanks for reporting, feel free to reopen.

JHedgesBartec, 2021-03-19:
I have now recreated this issue reliably with source code based upon the initiator/acceptor demos.

The following source configures both the SPP server and allows for initiating connections. In my source (not disclosed) I have a UART interface where we use a command set similar to AT to initaite the connections. So for the purposes of the demo the #define for ROLE can be set to ACCEPTOR or INITIATOR, both configurations will run the server, the INITIATOR will form the connection.

Attached is a .ZIP which contains the sdkconfig, source file and the IDF-Monitor output of both devices proving that a connection can be formed and terminated without a bond being created. The zip also contains both the elf and binary files which can be programmed onto two ESP32 modules. The demo was built using ESP-IDF release v4.0, the IDF source can be obtained from the link below, selecting the “esp32-gcc8.2.0-r2.exe” download:
https://visualgdb.com/support/esp32/

In the demo, the device configured with the role initiator will connect to the secondary device, send "Test Data", wait for the "ACK" response and then close the connection. At which point the CLOSE event triggers a query f the bond list, which is empty.

```
/*
   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include <stdint.h>
#include <string.h>
#include <stdbool.h>
#include <stdio.h>
#include "nvs.h"
#include "nvs_flash.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "esp_bt.h"
#include "esp_bt_main.h"
#include "esp_gap_bt_api.h"
#include "esp_bt_device.h"
#include "esp_spp_api.h"

#include "time.h"
#include "sys/time.h"

#define ACCEPTOR (0)
#define INITIATOR (1)

#define ROLE (INITIATOR)

#define SPP_TAG "SPP_SERVER_CLIENT"
#if ROLE == ACCEPTOR
#define EXCAMPLE_DEVICE_NAME "ESP_SPP_ACCEPTOR"
#else
#define EXCAMPLE_DEVICE_NAME "ESP_SPP_INITIATOR"
#endif
#define SPP_SERVER_NAME "SPP_SERVER"

static const esp_spp_mode_t esp_spp_mode = ESP_SPP_MODE_CB;

static const esp_spp_sec_t sec_mask = ESP_SPP_SEC_NONE;

static esp_bd_addr_t peer_bd_addr;
static uint8_t peer_bdname_len;
static char peer_bdname[ESP_BT_GAP_MAX_BDNAME_LEN + 1];
static const char remote_device_name[] = "ESP_SPP_ACCEPTOR";
#if ROLE == INITIATOR
static const esp_bt_inq_mode_t inq_mode = ESP_BT_INQ_MODE_GENERAL_INQUIRY;
static const uint8_t inq_len = 30;
static const uint8_t inq_num_rsps = 0;
#endif

static bool get_name_from_eir(uint8_t *eir, char *bdname, uint8_t *bdname_len)
{
    uint8_t *rmt_bdname = NULL;
    uint8_t rmt_bdname_len = 0;

    if (!eir) {
        return false;
[... 240 lines cut ...]
        return;
    }

    if ((ret = esp_spp_register_callback(esp_spp_cb)) != ESP_OK) {
        ESP_LOGE(SPP_TAG, "%s spp register failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }

    if ((ret = esp_spp_init(esp_spp_mode)) != ESP_OK) {
        ESP_LOGE(SPP_TAG, "%s spp init failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }

   esp_bt_pin_type_t pin_type = ESP_BT_PIN_TYPE_FIXED;
   esp_bt_pin_code_t pin_code;
   pin_code[0] = '1';
   pin_code[1] = '2';
   pin_code[2] = '3';
   pin_code[3] = '4';
   if (ESP_OK != esp_bt_gap_set_pin(pin_type, 4, pin_code))
   {
      ESP_LOGE(SPP_TAG, "Failed to set PIN");
   }
}
```

JHedgesBartec, 2021-03-19:
@Alvin1Zhang please could you re-open this issue.

Alvin1Zhang, 2021-03-22:
@JHedgesBartec Thanks for sharing the updates, reopen now.

boblane1, 2021-03-22:
Hi @JHedgesBartec,

There are two reasons to result in this issue:
1.  According to the Bluetooth Core Specification v4.2, the paring is not allowed if the initiator and the acceptor both have fixed PIN type.
[image]

2.  If PIN input is needed, you shall change `sec_mask` from `ESP_SPP_SEC_NONE` to `ESP_SPP_SEC_AUTHENTICATE`.

Thanks!

JHedgesBartec, 2021-03-22:
Hi @boblane1,

Regarding point 1: please could you reference your source for that information? We have historically used WT12 Bluegiga modules (BT Classic only) and these support pairings when both the Initiator and the Acceptor have the same fixed pins. I would like to see if this is something only supported in the earlier Bluetooth Core Specification.

Regarding point 2: I tried setting `sec_mask` to `ESP_SPP_SEC_AUTHENTICATE` however authentications still fails when both devices require a fixed pin.

In our use case the ESP32 modules are used with a command set interface, controlled by a host MCU. The host MCU provides a graphical interface for scanning for devices, creating pairings, initiating connections as well as acting as a Acceptor. In this case both modules would require a fixed pin to be able to form the bonds, the user cannot input a pin.
As both modules must operate with the Acceptor and Initiator roles, how would you suggest we proceed using a fixed PIN?

Many Thanks!

JHedgesBartec, 2021-03-22:
Hi @boblane1,

I have managed to find a copy of the Bluetooth Core Specificaton v2.1 + EDR, of which the WT12-A Bluegiga module conforms to and the limitation of both devices having a fixed PIN is there. So it appears to be a conformance issue on this module or they handle the dual Acceptor/Initiator role differently.

Is there a way to use the ESP-IDF in such a way that the modules can both have a fixed pin? Your suggestion of changing `sec_mask` fails the authentication when both devices have a fixed PIN.

Thanks
@JHedgesBartec

boblane1, 2021-03-22:
Hi @JHedgesBartec,

> Regarding point 1: please could you reference your source for that information? We have historically used WT12 Bluegiga modules (BT Classic only) and these support pairings when both the Initiator [...]

Bluetooth Core Specification v4.2 [Vol2, Part C] chapter 4.2.2.2 on page 278. And the ESP32 conform the procedure.

> In this case both modules would require a fixed pin to be able to form the bonds, the user cannot input a pin.
As both modules must operate with the Acceptor and Initiator roles, how would you suggest we proceed using a fixed PIN?

After setting `sec_mask` to `ESP_SPP_SEC_AUTHENTICATE`, you have two options:
1. If you still want to use fixed PIN, you can set fixed PIN for SPP acceptor, and set variable PIN for SPP initiator to avoid the limitation. When SPP initiator receives `ESP_BT_GAP_PIN_REQ_EVT`, it can directly reply PIN code in the callback like the following code without user input.
```C
static void esp_bt_gap_cb(esp_bt_gap_cb_event_t event, esp_bt_gap_cb_param_t *param)
{
...
    case ESP_BT_GAP_PIN_REQ_EVT:{
        ESP_LOGI(SPP_TAG, "ESP_BT_GAP_PIN_REQ_EVT min_16_digit:%d", param->pin_req.min_16_digit);
        if (param->pin_req.min_16_digit) {
            ESP_LOGI(SPP_TAG, "Input pin code: 0000 0000 0000 0000");
            esp_bt_pin_code_t pin_code = {0};
            esp_bt_gap_pin_reply(param->pin_req.bda, true, 16, pin_code);
        } else {
            ESP_LOGI(SPP_TAG, "Input pin code: 1234");
            esp_bt_pin_code_t pin_code;
            pin_code[0] = '1';
            pin_code[1] = '2';
            pin_code[2] = '3';
            pin_code[3] = '4';
            esp_bt_gap_pin_reply(param->pin_req.bda, true, 4, pin_code);
        }
        break;
    }
...
}

void app_main()
{
...
#if ROLE == ACCEPTOR
   esp_bt_pin_type_t pin_type = ESP_BT_PIN_TYPE_FIXED;
   esp_bt_pin_code_t pin_code;
   pin_code[0] = '1';
   pin_code[1] = '2';
   pin_code[2] = '3';
   pin_code[3] = '4';
   if (ESP_OK != esp_bt_gap_set_pin(pin_type, 4, pin_code))
   {
      ESP_LOGE(SPP_TAG, "Failed to set PIN");
   }
#else
    esp_bt_pin_type_t pin_type = ESP_BT_PIN_TYPE_VARIABLE;
    esp_bt_pin_code_t pin_code;
    esp_bt_gap_set_pin(pin_type, 0, pin_code);
#endif
...
}

```
2. If you use fixed PIN to avoid user input, I suggest you to use SSP(Secure Simple Pairing).  Before using it, you need to config the sdkconfig(Component config -> Bluetooth -> Bluedroid Options -> Secure Simple Pairing) and set the IO capabilities. You can reference it from the original SPP acceptor demo on ESP32.

Thanks.

JHedgesBartec, 2021-03-23:
Hi @boblane1,

Thank you for your help, it is very much appreciated.

I have made the changes and we now suceessfully see a bond created, however after the initial connection, a reconnection is failing with
`W (2811) BT_RFCOMM: PORT_StartCnf failed result:3` at 'ESP_SPP_CL_INIT_EVT'

The steps for this are as follows:
1. Run the acceptor demo on one module, attached is a modified one which will handshake data.
2. Run the initiator demo on the second module, clearing all pairings on the SPP INIT event, forming a connection and bond, sending data, safely disconnecting.
3. Run the initiator demo, this time so it doesn't clear the pairings in the SPP INIT event and the issue will be seen.

In this case I would expect the bond to still be valid and allow a connection to be formed. If the bond list is cleared on the acceptor then the initiator will start working again.

As you can see on the boot of the initiator the bond is still present, a GAP and SPP discovery is still used before connecting, I have tried without and no change.

I initially saw this in my software, and again returned to the demo, where we have the command interface so this can be recreated without changing the software or rebooting the modules.

Demos attached, here is the log from the initiator, no events are seen at the acceptor:
```
I (1181) SPP_INITIATOR_DEMO: ESP_SPP_INIT_EVT
I (1201) SPP_INITIATOR_DEMO: event: 10
I (1211) SPP_INITIATOR_DEMO: ESP_SPP_START_EVT
I (1211) SPP_INITIATOR_DEMO: ESP_BT_GAP_DISC_STATE_CHANGED_EVT
I (1211) SPP_INITIATOR_DEMO: Set Variable PIN
I (1661) SPP_INITIATOR_DEMO: ESP_BT_GAP_DISC_RES_EVT
I (1661) SPP_INITIATOR_DEMO: 24 0a c4 00 36 88
I (1661) SPP_INITIATOR_DEMO: ESP_SPP_ACCEPTOR
I (1671) SPP_INITIATOR_DEMO: ESP_BT_GAP_DISC_STATE_CHANGED_EVT
I (2761) SPP_INITIATOR_DEMO: ESP_SPP_DISCOVERY_COMP_EVT status=0 scn_num=1
I (2771) SPP_INITIATOR_DEMO: ESP_SPP_CL_INIT_EVT
W (2811) BT_RFCOMM: PORT_StartCnf failed result:3
I (2811) SPP_INITIATOR_DEMO: ESP_SPP_CLOSE_EVT
I (2811) SPP_INITIATOR_DEMO: Module has 1 bonds
I (2811) SPP_ACCEPTOR_DEMO: Bond : 24 0a c4 00 36 88
```

Many Thanks.

[demo_files.zip](https://github.com/espressif/esp-idf/files/6188661/demo_files.zip)

boblane1, 2021-03-24:
Hi @JHedgesBartec,

>W (2811) BT_RFCOMM: PORT_StartCnf failed result:3

This log indicates the authentication is failed. These two devices probably do not have the same link key. Please clear bond on both side, and try to re-pair to see if the problem is still exist. The following log is what I used the demo you supplied.
[image]

BTW,  `esp_spp_start_discovery` does not mean to reconnect. If you want to release the previous connection, you shall call `esp_spp_disconnect`.

Thanks.

JHedgesBartec, 2021-03-24:
Hi @boblane1 

Did you try rebooting the initiator only without clearing the bond?

The behaviour I see is, a bond will be formed between the acceptor and intitator, I then boot the initiator, this time with the code which removes the bond commented out, therefore simulating a fresh connection, reusing a existing bond and the connecton is rejected wih the warning.

If I clear the bond on the acceptor it all starts working again, however the ESP32 should be able to handle this as bonds can be removed without informing the other device.

As i said, in the application I am using it for, I see this exact behaviour with the following steps, however as I use a command interface it is not ideal for providng demo software:

1. With the initiator set with a variable PIN and the acceptor with a fixed PIN, form a bond between the devices.
2. Reboot the intiator and check the bond is still there - do not remove the bond
3. Attempt to connect to the acceptor using either a SPP discovery leading to connect, or GAP discovery, SPP discovery, then SPP connect.
4. Connection is rejected with the reported error.

In this software bonds are only cleared on command, so they should still be valid.
The acceptor can be rebooted several times, this does not help with the issue, again bonds are not being cleared on this device.

One thing you could try is clearing the bond on the initiator but not the acceptor and then working though my steps above?

Thanks.

JHedgesBartec, 2021-03-24:
Hi @boblane1 ,

I have been doing some further work to prove this issue. I now have a 100% repeatable scenario.
Please find attached new demo files (source and binaries).
In these demos the operation is now controlled by a basic command interface for the UART, the premise of completing a GAP scan, SPP scan, SPP connect, handshake some data, disconnect is the same.
The UART is configured to be the same as the console, `UART_NUM_0` and is controlled by the following commands:
- `b` - lists the bonds to the console
- `d` - deletes the bonds
- `r` - restarts the module using `esp_restart()`
- `c` - connects to the acceptor (only available on the initiator

To replicate the issue please follow these steps carefully:
1. Clear the bonds on both the acceptor and initiator, reboot and list the bonds to be sure its clear
2. Initiate the connection and wait for the disconnect event
3. Waiting 2-3 seconds from the disconnect event, list and then clear the bonds. Doing this any quicker will not remove the bond.
4. Reboot the initiator 
5. Initiate a connection, again waiting for the disconnect.
6. Wait minimum 2-3 seconds and initiate the connection again

At step 6 is where the issue occurs every single time for me. 
I have included the binary files and elf files in the zip in case this is limited to the IDF version I am using (in the first post).
I am running the acceptor on ESP32 Core Board V2 and the Initiator on a ESP WROVER KIT v4.1, though I have seen this issue when running both the acceptor and initiator on ESP-WROOM-32 on proprietary hardware.

To allow the connection to be successful again, the bonds must be cleared from the acceptor, so as you suggest it might be some issue with the link key. Though I think it is a case where it is not handled correctly.
As far as I am aware this will be a bug, as you should be able to re-pair to a Bluetooth device numerous times and form a connection each time.
I have another Bluetooth Module (WT12-A) which when used as the acceptor with the ESP32 as the initiator works fine, there are no issues at all. It is similar with other BT devices, a factory reset is not normally needed to be able to re-pair and connect.

I hope using the attached demos you are able to reproduce this behaviour yourself.

Many thanks.

[uart_spp_demo.zip](https://github.com/espressif/esp-idf/files/6198112/uart_spp_demo.zip)

munho-espressif, 2021-04-08:
Hi @JHedgesBartec,

Now I'll take over this problem, but I'm not free right now. I suggest that you use variable pin code for both devices. I'll solve this problem later when I have time.

Thanks.

JHedgesBartec, 2021-04-08:
Hi @munho-espressif,

Thankyou for posting the update, unfortunately the issue is still present when both devices are using variable pins. There is no known work around for this issue.

Thanks

xiongweichao, 2021-06-17:
Hi @JHedgesBartec 

Can you try the bt-lib attached below and check if it solves your issue? Let me know if it solves the issue. 
Replace the file at $IDF_PATH/components/bt/controller/libbtdm_app.a with this file. 

Thanks

[bt-lib.zip](https://github.com/espressif/esp-idf/files/6670097/bt-lib.zip)

JHedgesBartec, 2021-06-22:
Hi @xiongweichao,

I have tried applying the new library to our current IDF version and the example projects no longer build. Many of the errors are undefined references, so I presume the library was built from a newer version.
There is also a slight difference to the location of the library:
$IDF_PATH/components/bt/controller/**lib/**libbtdm_app.a

Please see the first post for the IDF version used.

Thanks

xiongweichao, 2021-06-22:
Hi @JHedgesBartec ,

Sorry, please retry the bt-lib attached below and check if it solves your issue.

Thanks
[bt-lib.zip](https://github.com/espressif/esp-idf/files/6693931/bt-lib.zip)

JHedgesBartec, 2021-06-23:
Hi @xiongweichao,

I am still seeing the undefined reference issues as with the first version.

I have double checked the version of the IDF framework and despite our IDE saying it is v4.0.1, I have checked esp_idf_version.h which shows:
`/** Major version number (X.x.x) */
#define ESP_IDF_VERSION_MAJOR   4
/** Minor version number (x.X.x) */
#define ESP_IDF_VERSION_MINOR   0
/** Patch version number (x.x.X) */
#define ESP_IDF_VERSION_PATCH   0`

Apologies, I will update the bug report to reflect this.

Jack

xiongweichao, 2021-06-30:
Hi, @JHedgesBartec 

Which idf version(**commit id**) are you using ?

Thanks

JHedgesBartec, 2021-07-14:
Hi @xiongweichao ,

We use VisualStudio with the VisualGDB plugin so the ESP-IDF is provided through a toolchain download, not seperately checked out by us.
We are using ESP-IDF version v4.0 as provided here:
https://visualgdb.com/support/esp32/

The folder $IDF_PATH/.git contains a packed-refs files which suggests it could actually be v4.1-dev (see attached file, .txt extension added to allow upload).

I hope this helps, if there is anything that is retained on the filesystem to reference the commit ID please indicate where I can find this.

Thanks

[packed-refs.txt](https://github.com/espressif/esp-idf/files/6814534/packed-refs.txt)

JHedgesBartec, 2021-07-14:
Hi @xiongweichao ,

On boot of the module the monitor prints 'IDF version used: v4.0-beta2-174-g99fb9a3f7-dirty' to the console, not sure if this will help identify the commit id.

xiongweichao, 2021-07-15:
Hi, @JHedgesBartec 

Please retry the bt-lib attached below and check if it solves your issue.

Thanks

[libbtdm_app.zip](https://github.com/espressif/esp-idf/files/6820166/libbtdm_app.zip)

JHedgesBartec, 2021-07-15:
Hi @xiongweichao,

Regarding the quoted problem below the library change has fixed the issue.

However the original problem, a bond not being created between the two when both devices are set with a fixed PIN still persists. We require one of the modules to act as both an initiator and an acceptor, how is it possible to use a fixed PIN in this case?

Thanks

> 
> 
> Hi @boblane1 ,
> 
> I have been doing some further work to prove this issue. I now have a 100% repeatable scenario.
> Please find attached new demo files (source and binaries).
> In these demos the operation is now controlled by a basic command interface for the UART, the premise of completing a GAP scan, SPP scan, SPP connect, handshake some data, disconnect is the same.
> The UART is configured to be the same as the console, `UART_NUM_0` and is controlled by the following commands:
> 
>     * `b` - lists the bonds to the console
> 
>     * `d` - deletes the bonds
> 
>     * `r` - restarts the module using `esp_restart()`
> 
>     * `c` - connects to the acceptor (only available on the initiator
> 
> 
> To replicate the issue please follow these steps carefully:
> 
>     1. Clear the bonds on both the acceptor and initiator, reboot and list the bonds to be sure its clear
> 
>     2. Initiate the connection and wait for the disconnect event
> 
>     3. Waiting 2-3 seconds from the disconnect event, list and then clear the bonds from the initiator only. Doing this any quicker will not remove the bond.
> 
>     4. Reboot the initiator
> 
>     5. Initiate a connection, again waiting for the disconnect.
> 
>     6. Wait minimum 2-3 seconds and initiate the connection again
> 
> 
> At step 6 is where the issue occurs every single time for me.
> I have included the binary files and elf files in the zip in case this is limited to the IDF version I am using (in the first post).
> I am running the acceptor on ESP32 Core Board V2 and the Initiator on a ESP WROVER KIT v4.1, though I have seen this issue when running both the acceptor and initiator on ESP-WROOM-32 on proprietary hardware.
> 
> To allow the connection to be successful again, the bonds must be cleared from the acceptor, so as you suggest it might be some issue with the link key. Though I think it is a case where it is not handled correctly.
> As far as I am aware this will be a bug, as you should be able to re-pair to a Bluetooth device numerous times and form a connection each time.
> I have another Bluetooth Module (WT12-A) which when used as the acceptor with the ESP32 as the initiator works fine, there are no issues at all. It is similar with other BT devices, a factory reset is not normally needed to be able to re-pair and connect.
> 
> I hope using the attached demos you are able to reproduce this behaviour yourself.
> 
> Many thanks.
> 
> [uart_spp_demo.zip](https://github.com/espressif/esp-idf/files/6198112/uart_spp_demo.zip)

xiongweichao, 2021-07-19:
Hi @JHedgesBartec ,

According to the Bluetooth Core Specification v4.2, the paring is not allowed if the initiator and the acceptor both have fixed PIN type.  For this module that is both the initiator and the acceptor, you can use variable PIN.

Thanks
[image]

closed 2022-08-04 by espressif-bot