espressif/esp-idf #5851: smartconfig dont working when esp_restart()

hnhkj, 2020-09-11:
running smartconfig overtime and esp_restart(). but when esp32 restart, smartconfig can not working. 

source code

```c
/* Common functions for protocol examples, to establish Wi-Fi or Ethernet connection.

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
 */

#include <string.h>
#include "protocol_examples_common.h"
#include "sdkconfig.h"
#include "esp_event.h"
#include "esp_wifi.h"
#include "esp_wifi_default.h"
#if CONFIG_EXAMPLE_CONNECT_ETHERNET
#include "esp_eth.h"
#endif
#include "esp_log.h"
#include "esp_netif.h"
//#include "panic_funcs.h"
#include "driver/gpio.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "lwip/err.h"
#include "lwip/sys.h"
#include "esp_smartconfig.h"
#include "led_custom.h"

#define GOT_IPV4_BIT BIT(0)
#define GOT_IPV6_BIT BIT(1)
#define ESPTOUCH_DONE_BIT BIT(2)
#define ESPTOUCH_FAIL_BIT BIT(3)

#ifdef CONFIG_EXAMPLE_CONNECT_IPV6
#define CONNECTED_BITS (GOT_IPV4_BIT | GOT_IPV6_BIT)

#if defined(CONFIG_EXAMPLE_CONNECT_IPV6_PREF_LOCAL_LINK)
#define EXAMPLE_CONNECT_PREFERRED_IPV6_TYPE ESP_IP6_ADDR_IS_LINK_LOCAL
#elif defined(CONFIG_EXAMPLE_CONNECT_IPV6_PREF_GLOBAL)
#define EXAMPLE_CONNECT_PREFERRED_IPV6_TYPE ESP_IP6_ADDR_IS_GLOBAL
#elif defined(CONFIG_EXAMPLE_CONNECT_IPV6_PREF_SITE_LOCAL)
#define EXAMPLE_CONNECT_PREFERRED_IPV6_TYPE ESP_IP6_ADDR_IS_SITE_LOCAL
#elif defined(CONFIG_EXAMPLE_CONNECT_IPV6_PREF_UNIQUE_LOCAL)
#define EXAMPLE_CONNECT_PREFERRED_IPV6_TYPE ESP_IP6_ADDR_IS_UNIQUE_LOCAL
#endif // if-elif CONFIG_EXAMPLE_CONNECT_IPV6_PREF_...

#else
#define CONNECTED_BITS (GOT_IPV4_BIT)
#endif

static EventGroupHandle_t s_connect_event_group;
//static EventGroupHandle_t s_smartconfig_event_group;

static esp_ip4_addr_t s_ip_addr;
static const char *s_connection_name;
static esp_netif_t *s_example_esp_netif = NULL;
static uint8_t s_example_esp_netif_status = 0;

#ifdef CONFIG_EXAMPLE_CONNECT_IPV6
static esp_ip6_addr_t s_ipv6_addr;

/* types of ipv6 addresses to be displayed on ipv6 events */
static const char *s_ipv6_addr_types[] = {
[... 652 lines cut ...]
I (6782) wifi:wifi firmware version: cc57434
I (6782) wifi:wifi certification version: v7.0
I (6782) wifi:config NVS flash: enabled
I (6782) wifi:config nano formating: disabled
I (6782) wifi:Init dynamic tx buffer num: 32
I (6782) wifi:Init data frame dynamic rx buffer num: 32
I (6792) wifi:Init management frame dynamic rx buffer num: 32
I (6802) wifi:Init management short buffer num: 32
I (6802) wifi:Init static rx buffer size: 1600
I (6802) wifi:Init static rx buffer num: 10
I (6812) wifi:Init dynamic rx buffer num: 32
I (6912) phy: phy_version: 4181, c9a03e2, Mar 10 2020, 12:45:06, 0, 0
I (6922) wifi:mode : sta (10:52:1c:50:bf:d8)
I (6922) example_connect: esp wifi connect LinkIt_Smart_7688_1B1855, 123467890
W (8972) example_connect: reason: fail 201, LinkIt_Smart_7688_1B1855
W (8972) example_connect: retry to connect to the AP 1
W (11032) example_connect: reason: fail 201, LinkIt_Smart_7688_1B1855
W (11032) example_connect: retry to connect to the AP 2
W (13082) example_connect: reason: fail 201, LinkIt_Smart_7688_1B1855
W (13082) example_connect: retry to connect to the AP 3
W (15132) example_connect: reason: fail 201, LinkIt_Smart_7688_1B1855
W (15132) example_connect: retry to connect to the AP 4
E (16922) example_connect: connect to the AP fail, start smartconfig
I (16972) smartconfig: SC version: V2.9.0
```

zhangyanjiaoesp, 2020-09-14:
@hnhkj 

The reason code `201` indicates that the STA can't find the AP, please check the AP's state. When this device can't connect to AP, can other devices  connect successfully ?

hnhkj, 2020-09-19:
other device connect to right.
It's sure. when router and esp32 power off and power on. esp32 reboot faster and router reboot slow. then esp32 wifi connect fail. then esp32 enter to smartconfig mode. esp32 restart when overtime 1min. 
1. if router AP normal running in 1min. then esp32 working normal. it's will get username/password and connect to wifi.
2. if router AP normal running over 1min. then esp32 dont working normal. then enter loop (see debug information). error dont fixed when I restart router(router working. other device can connect to router). if I press esp32 hardware restart. it's working.

I think that smartconfig have bugs and esp_restart().
1. if smartconfig fail then restart. smartconfig will keep some information in esp32.
2. esp_restart() can not reset function of wifi if smartconfig fail restart.

zhangyanjiaoesp, 2020-09-21:
@hnhkj 

I found your IDF version is `ESP-IDF:  v4.2-dev-1303-gcd4fa46f5`, we have solved an `esp_restart()` bug in the later version.
Can you please update your IDF version to the latest and test again ?

hnhkj, 2020-10-12:
thanks. running.

closed 2020-10-12 by hnhkj