espressif/esp-idf #5853: docker run -it espressif/idf /bin/bash fail

guolxtank, 2020-09-11:
when I run docker on wsl(ubuntu 20 lts),an error shows up:
```shell
cccd@DESKTOP-D0PC:~/esp32$ docker run --rm -v $PWD:/project -w /project espressif/idf idf.py build
Unable to find image 'espressif/idf:latest' locally
latest: Pulling from espressif/idf
23884877105a: Pull complete
bc38caa0f5b9: Pull complete
2910811b6c42: Pull complete
36505266dcc6: Pull complete
618f883824a6: Pull complete
5cab8898c194: Pull complete
b66ebc3d3ffb: Pull complete
9f08e4000fa8: Pull complete
7ce64652eb1a: Pull complete
Digest: sha256:9ad3ab4e04f317a425dbad3ae4818e2ab6852685d8a1188502db66f07d03d4b0
Status: Downloaded newer image for espressif/idf:latest
Adding ESP-IDF tools to PATH...
Using Python interpreter in /opt/esp/python_env/idf4.2_py3.6_env/bin/python
Checking if Python packages are up to date...
Python requirements from /opt/esp/idf/requirements.txt are satisfied.
Added the following directories to PATH:
  /opt/esp/idf/components/esptool_py/esptool
  /opt/esp/idf/components/espcoredump
  /opt/esp/idf/components/partition_table
  /opt/esp/idf/components/app_update
  /opt/esp/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin
  /opt/esp/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin
  /opt/esp/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin
  /opt/esp/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin
  /opt/esp/tools/cmake/3.16.4/bin
  /opt/esp/tools/openocd-esp32/v0.10.0-esp32-20200420/openocd-esp32/bin
  /opt/esp/python_env/idf4.2_py3.6_env/bin
  /opt/esp/idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

Executing action: all (aliases: build)
CMakeLists.txt not found in project directory /project
Your environment is not configured to handle unicode filenames outside of ASCII range. Environment variable LC_ALL is temporary set to C.UTF-8 for unicode support.

```
Would you please tell me what should I do?
Thanks!

dobairoland, 2020-09-11:
Hi @guolxtank. 

> CMakeLists.txt not found in project directory /project

> -v $PWD:/project: mounts the current directory on the host ($PWD) as /project directory in the container

You should change directory into a valid ESP-IDF project before running the docker command, For example:

```
cd $IDF_PATH/examples/get-started/hello_world
```

guolxtank, 2020-09-11:
it works!thanks a lot!

dobairoland, 2020-09-11:
Great! I glad to hear it.

closed 2020-09-11 by dobairoland

QGB, 2022-04-19:
`docker run --rm -v /home/qgb/github/esp-idf/examples/get-
started/hello_world:/project -w /proj
ect espressif/idf idf.py build`