espressif/esp-idf #5854: LWIP crashes at xQueueGenericSend

sebirdman, 2020-09-11:
## Environment
- IDF version 4.0.1
- Build System:  cmake

## Problem Description
After some time, i'm seeing the following stack trace from my devices:

```
#0  0x4000bff0 in ?? ()
#1  0x40093979 in vTaskExitCritical (mux=<optimized out>) at ../dependencies/espressif/esp-idf/components/freertos/tasks.c:4274
#2  0x40092baf in xQueueGenericSend (xQueue=0x3fff9dd0, pvItemToQueue=<optimized out>, xTicksToWait=<optimized out>, xCopyPosition=0) at ../dependencies/espressif/esp-idf/components/freertos/queue.c:834
#3  0x40155765 in sys_sem_signal (sem=0x3f802b60) at ../dependencies/espressif/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c:133
#4  0x401577ed in lwip_netconn_do_writemore (conn=0x3f802f7c) at ../dependencies/espressif/esp-idf/components/lwip/lwip/src/api/api_msg.c:1865
#5  0x401585d4 in lwip_netconn_do_write (m=0x3ffbaac0) at ../dependencies/espressif/esp-idf/components/lwip/lwip/src/api/api_msg.c:1910
#6  0x4014024c in tcpip_thread_handle_msg (msg=0x3ffbaa70) at ../dependencies/espressif/esp-idf/components/lwip/lwip/src/api/tcpip.c:168
#7  0x401402f7 in tcpip_thread (arg=0x0) at ../dependencies/espressif/esp-idf/components/lwip/lwip/src/api/tcpip.c:154
#8  0x400920c8 in vPortTaskWrapper (pxCode=0x401402c4 <tcpip_thread>, pvParameters=0x0) at ../dependencies/espressif/esp-idf/components/freertos/port.c:143
```

### Expected Behavior

Doesn't crash

### Actual Behavior

Crashes

I can provide elf file and coredump file if necessary.

david-cermak, 2020-09-15:
Hi @sebirdman 

Could you please try to decode the coredump with `--rom-elf` parameter as described [here](https://docs.espressif.com/projects/esp-idf/en/v4.0.1/api-guides/core_dump.html#rom-functions-in-backtraces) or just share the elf file and the coredump output as you've suggested?

Is it be possible to share a very simple application sample where this crash could be reproducible?

sebirdman, 2020-09-15:
@david-cermak decoding with --rom-elf is way more interesting. I'm seeing one of my threads:

```
==================== THREAD 23 (TCB: 0x3ffbb304, name: 'main') =====================
#0  0x4009c194 in invoke_abort () at ../dependencies/espressif/esp-idf/components/esp32/panic.c:159
#1  0x4009c542 in abort () at ../dependencies/espressif/esp-idf/components/esp32/panic.c:175
#2  0x4009c561 in vApplicationStackOverflowHook (xTask=0x3ffbb304 <rwip_heap_msg+6372>, pcTaskName=0x3ffbb33c <rwip_heap_msg+6428> \"main\") at ../dependencies/espressif/esp-idf/components/esp32/panic.c:125
#3  0x40094703 in vTaskSwitchContext () at ../dependencies/espressif/esp-idf/components/freertos/include/freertos/portable.h:214
#4  0x4009259b in _frxt_dispatch () at ../dependencies/espressif/esp-idf/components/freertos/portasm.S:406
Backtrace stopped: Cannot access memory at address 0x3ff000d0
```

does this mean that this error i'm getting is likely from a stack overflow on my main task?

david-cermak, 2020-09-17:
@sebirdman Yes, this looks like the stack overflow hook triggered. 
Have you tried to update the main task stack size? Note that, it is possible to evaluate actual stack space of your tasks using this API https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos.html#_CPPv427uxTaskGetStackHighWaterMark12TaskHandle_t

sebirdman, 2020-09-18:
@david-cermak thanks for the help, after pushing an update to my devices in the field they no longer report this issue. 

that --rom-elf param is very helpful.

closed 2020-09-18 by sebirdman

igrr, 2020-09-18:
@sebirdman In the core dumps you were getting, was `main` task identified as the task where the crash has happened? Normally the backtrace of this task should be printed at the very top of the core dump.

If that wasn't the case, perhaps you could share the core dump with us to investigate this? If the task which triggered the crash is not listed at the top, this is obviously not very helpful...

sebirdman, 2020-09-18:
@igrr it doesn't look like the main task was identified here's the full dump:

```
233M001494000178-2020.254.07fa50be-1600026400.coredump
espcoredump.py v0.4-dev
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffd6eb4, name: 'tiT', GDB name: 'process 1073573556'

================== CURRENT THREAD REGISTERS ===================
exccause       0x0 (IllegalInstructionCause)
excvaddr       0x0
epc1           0x0
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0xffff
epc7           0x73
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
eps7           0x0
pc             0x40082a5e       0x40082a5e <esp_crosscore_int_send+78>
lbeg           0x400967e4       1074358244
lend           0x40096800       1074358272
lcount         0xffffffff       4294967295
sar            0x10     16
ps             0x60b20  396064
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x80082ac0       -2146948416
a1             0x3ffd6d20       1073573152
a2             0x3ff000dc       1072693468
a3             0x1      1
a4             0x3ffbfbb8       1073478584
a5             0xfe     254
a6             0x1      1
a7             0x0      0
a8             0x80082a40       -2146948544
a9             0x3ffd6d00       1073573120
a10            0x0      0
a11            0x3ffd6eb4       1073573556
a12            0x60b20  396064
a13            0xfe     254
a14            0x1      1
a15            0x0      0

==================== CURRENT THREAD STACK =====================
#0  0x40082a5e in esp_crosscore_int_send (core_id=<optimized out>, reason_mask=<optimized out>) at ../dependencies/espressif/esp-idf/components/esp32/crosscore_int.c:105
#1  0x40082ac0 in esp_crosscore_int_send_yield (core_id=0) at ../dependencies/espressif/esp-idf/components/esp32/crosscore_int.c:111
#2  0x40092fa8 in xQueueGenericReceive (xQueue=0x3ffbd8dc <rwip_heap_env+8368>, pvBuffer=0x3ffd6df0, xTicksToWait=<optimized out>, xJustPeeking=0) at ../dependencies/espressif/esp-idf/components/freertos/include/freertos/portable.h:214
#3  0x401558d4 in sys_arch_mbox_fetch (mbox=<optimized out>, msg=0x3ffd6df0, timeout=<optimized out>) at ../dependencies/espressif/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c:297
#4  0x40140223 in tcpip_timeouts_mbox_fetch (mbox=0x3ffc6f14 <tcpip_mbox>, msg=0x3ffd6df0) at ../dependencies/espressif/esp-idf/components/lwip/lwip/src/api/tcpip.c:110
[... 276 lines cut ...]
.coredump.tasks.data 0x3ffb5630 0x1f4 RW 
.coredump.tasks.data 0x3ffd4988 0x170 RW 
.coredump.tasks.data 0x3ffe27f0 0x1f0 RW 
.coredump.tasks.data 0x3ffdb7ac 0x170 RW 
.coredump.tasks.data 0x3ffdc410 0x214 RW 
.coredump.rom.text 0x40000000 0x170 R X
.coredump.rom.text 0x40000180 0x8 R X
.coredump.rom.text 0x400001c0 0x8 R X
.coredump.rom.text 0x40000200 0x8 R X
.coredump.rom.text 0x40000240 0x8 R X
.coredump.rom.text 0x40000280 0xc R X
.coredump.rom.text 0x400002c0 0x4 R X
.coredump.rom.text 0x40000300 0x8 R X
.coredump.rom.text 0x40000340 0x18 R X
.coredump.rom.text 0x400003c0 0x8 R X
.coredump.rom.text 0x40000400 0x160 R X
.coredump.rom.text 0x40000560 0xd10c R X
.coredump.rom.text 0x40010000 0x54ef4 R X

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
```

Do you have an email i can use to send the coredump and elf?

igrr, 2020-09-18:
ivan at %company% dot com, much appreciated!