espressif/esp-idf #5859: install.sh sets  -e, so my shell closes and I cannot see the errors

crgarcia12, 2020-09-13:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      ESP32-DevKitC|
- Kit version DevKitC): v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): v4.3-dev-907-g6c17e3a64
- Build System: idf.py
- Operating System: Linux
- Using an IDE?: Yes (VsCode)
- Power Supply: USB

## Problem Description
I am new to Bash and linux. I am having problems with install.sh and export sh, but I cannot see the errors because the console closes inmediatly. 
Either I am executing the scripts wrong (. ./install.sh) or based on my search install.sh maybe should not have `set -e` at the beggining.

//Detailed problem description goes here.
After running export, the shell closes

### Expected Behavior
When there is an error, I see the error message and my shell reminds open

igrr, 2020-09-14:
> Either I am executing the scripts wrong (. ./install.sh) 

Hi @crgarcia12, indeed there are two ways to run scripts in bash:

* execute it (in a subshell) — `./install.sh` or `/path/to/install.sh`.
* "source" it — `. ./install.sh` or `source ./install.sh` — this runs the script in the current shell, and primarily used for setting environment variables.

For install.sh, you need to do the former (execute) and for export.sh you need latter (source).

See the relevant part of the Getting Started guide: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#id3

crgarcia12, 2020-09-14:
Such a beginner question :S Sorry for that
I had some global settings in my bash that were interfering with this

Thank you @igrr for your help!

closed 2020-09-14 by crgarcia12