espressif/esp-idf #5861: Using aws-iot shadow along with publish on topic causing tls error -0x4e

rhr407, 2020-09-13:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version:  v4.1
- Build System:         [idf.py]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

I am using aws-iot component from [here](https://duckduckgo.comhttps://github.com/espressif/esp-aws-iot) and added to the components folder of esp-idf. I want to use both aws-iot shadow service along with publishing some data to another topic. THe problem is that I am facing an mbedtls error -0x4e. I would be grateful if you kindly help me with this issues.

### Expected Behavior

These should be no errors in publishing the message to aws topic along with running shadow in parallel. Both threads should work smoothly.

### Actual Behavior
Th

### Steps to reproduce

Just use the two components of aws-iot i.e. aws-iot shadow and pubsub in parallel as separate threads.

### Code to reproduce this issue

## Debug Logs

```
E (340173) aws_iot: failed! mbedtls_ssl_write returned -0x4e
Published: {"DT":2,"T":53,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}I (340273) mbedtls: ssl_tls.c:2721 in_left: 0, nb_want: 5

I (340273) mbedtls: ssl_tls.c:2722 ssl->f_recv(_timeout)() returned -26624 (-0x6800)

W (340283) mbedtls: ssl_tls.c:4973 mbedtls_ssl_fetch_input() returned -26624 (-0x6800)

W (340293) mbedtls: ssl_tls.c:4344 ssl_get_next_record() returned -26624 (-0x6800)

W (340303) mbedtls: ssl_tls.c:8335 mbedtls_ssl_read_record() returned -26624 (-0x6800)

I (340323) mbedtls: ssl_tls.c:8270 => read

I (340323) mbedtls: ssl_tls.c:4311 => read record

I (340323) mbedtls: ssl_tls.c:2536 => fetch input

I (340333) mbedtls: ssl_tls.c:2697 in_left: 0, nb_want: 5

I (340433) mbedtls: ssl_tls.c:2721 in_left: 0, nb_want: 5

I (340433) mbedtls: ssl_tls.c:2722 ssl->f_recv(_timeout)() returned -26624 (-0x6800)

W (340433) mbedtls: ssl_tls.c:4973 mbedtls_ssl_fetch_input() returned -26624 (-0x6800)

W (340443) mbedtls: ssl_tls.c:4344 ssl_get_next_record() returned -26624 (-0x6800)

W (340453) mbedtls: ssl_tls.c:8335 mbedtls_ssl_read_record() returned -26624 (-0x6800)

I (340473) mbedtls: ssl_tls.c:8270 => read

I (340473) mbedtls: ssl_tls.c:4311 => read record

I (340473) mbedtls: ssl_tls.c:2536 => fetch input

I (340483) mbedtls: ssl_tls.c:2697 in_left: 0, nb_want: 5

I (340583) mbedtls: ssl_tls.c:2721 in_left: 0, nb_want: 5
[... 319 lines cut ...]
I (343173) mbedtls: ssl_tls.c:4311 => read record

I (343173) mbedtls: ssl_tls.c:2536 => fetch input

I (343183) mbedtls: ssl_tls.c:2697 in_left: 0, nb_want: 5

I (343193) mbedtls: ssl_tls.c:8682 => write

I (343193) mbedtls: ssl_tls.c:2755 => flush output

I (343193) mbedtls: ssl_tls.c:2774 message length: 168, out_left: 168

I (343203) mbedtls: ssl_tls.c:2779 ssl->f_send() returned -78 (-0x004e)

W (343213) mbedtls: ssl_tls.c:8612 mbedtls_ssl_flush_output() returned -78 (-0x004e)

I (343223) mbedtls: ssl_tls.c:8710 <= write

```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

shahpiyushv, 2020-09-14:
@rhr407 , your thing-shadow and subscribe handling should be from a single thread which should also keep calling aws_iot_shadow_yield() in a loop. Publish calls could be made from other threads, but that's about it, no yield or other calls. Does your code ensure this?

rhr407, 2020-09-14:
yes I am doing that in shadow code only as follows:
```
while (NETWORK_ATTEMPTING_RECONNECT == rc || NETWORK_RECONNECTED == rc || SUCCESS == rc) {

        rc = aws_iot_shadow_yield(&mqttClient, 100) ;

        if (NETWORK_ATTEMPTING_RECONNECT == rc || shadowUpdateInProgress) {
            rc = aws_iot_shadow_yield(&mqttClient, 1000) ;

            // If the client is attempting to reconnect, or already waiting on a shadow update,
            // we will skip the rest of the loop.
            continue;
        }

```

but not in the pubsub code. I am just publishing in another thread as you mentioned.