espressif/esp-idf #5863: unable to replicate sntp example in v4.0.1

mzimmers, 2020-09-13:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROVER]
- IDF version (run ``git describe --tags`` to find it):
    // v4.0.1
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. build the example; it runs fine.
2. Build my app with the example code; doesn't connect.
3. call sntpInit() once - after WiFi is connected and IP address assigned.
4. call sntpSync() periodically - again, after WiFi connected.
5. I left out the callback because it's never called.
6. this is the output, over and over:

> I (31597) Wifi: sntpSync(): attempt 1 to sync with SNTP server; syncStatus is 0.
I (33597) Wifi: sntpSync(): attempt 2 to sync with SNTP server; syncStatus is 0.
I (35597) Wifi: sntpSync(): attempt 3 to sync with SNTP server; syncStatus is 0.
I (37597) Wifi: sntpSync(): attempt 4 to sync with SNTP server; syncStatus is 0.
I (39597) Wifi: sntpSync(): attempt 5 to sync with SNTP server; syncStatus is 0.
I (41597) Wifi: sntpSync(): attempt 6 to sync with SNTP server; syncStatus is 0.
I (43597) Wifi: sntpSync(): attempt 7 to sync with SNTP server; syncStatus is 0.
I (45597) Wifi: sntpSync(): attempt 8 to sync with SNTP server; syncStatus is 0.
I (47597) Wifi: sntpSync(): attempt 9 to sync with SNTP server; syncStatus is 0.
W

### Code to reproduce this issue

```
void Wifi::sntpInit()
{
    server = m_tasks->flash->get(NVS_KEY_NTP_SERVER);
    sntp_setoperatingmode(SNTP_OPMODE_POLL);
    sntp_setservername(0, server.c_str());
    sntp_set_time_sync_notification_cb(sntpCallback);
#ifdef CONFIG_SNTP_TIME_SYNC_METHOD_SMOOTH
    sntp_set_sync_mode(SNTP_SYNC_MODE_SMOOTH);
#endif
    sntp_init();
}

void Wifi::sntpSync(tm &timeinfo)
{
    time_t now = 0;
    int retry = 0;
    int syncStatus;

    do
    {
        while ((syncStatus = sntp_get_sync_status()) == SNTP_SYNC_STATUS_RESET && ++retry < 10)
        {
            ESP_LOGI(TAG, "sntpSync(): attempt %d to sync with SNTP server; syncStatus is %d.", retry, syncStatus);
            vTaskDelay(2000 / portTICK_PERIOD_MS);
        }
            time(&now);
            localtime_r(&now, &timeinfo);
            ESP_LOGI(TAG, "sntpSync(): now is %ld.", now);
    } while (false);
}
```
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5215342/sdkconfig.txt)

[wifibutton.zip](https://github.com/espressif/esp-idf/files/5215346/wifibutton.zip)

Alvin1Zhang, 2020-09-14:
Thanks for reporting and letting us know, we will look into.

mzimmers, 2020-09-18:
Anything?

KonstantinKondrashov, 2020-09-21:
Hi @mzimmers!
You can try to use a little another approach for updating time. Please look at the [sntp_set_sync_interval(uint32_t interval_ms)](https://github.com/espressif/esp-idf/blob/master/components/lwip/include/apps/sntp/sntp.h#L123-L133) function, it will sync time periodically.

Using sntpSync->sntp_get_sync_status() does not initiate anything inside the sntp functionality. You should not call sntpSync() periodically.

Use `sntpCallback` to get a notification that time was updated.

mzimmers, 2020-09-21:
Hi Konstantin - thank you for the suggestion. I believe sntp_set_sync_interval() was introduced in v4.1 of the IDF, and I'm still on v4.0.1.

Alvin1Zhang, 2020-10-21:
Thanks for reporting, feel free to reopen. Thanks.

closed 2020-10-21 by Alvin1Zhang