espressif/esp-idf #5864: Example of BLE (or simple BT) to PC, connected as a Serial COM Port

thomasnugent, 2020-09-14:
Hi all, I've been fairly lost in searching for a simple example of how to use the ESP32 as a Bluetooth device that can send & receive serial data to/from a computer. [This example using Arduino](https://www.techcoil.com/blog/how-to-connect-to-an-esp32-development-board-via-bluetooth-on-windows-10/) works well for me and I imagine is a very common application of ESP32's Bluetooth capability. Some others online have suggested the `gatt_server` example, but this looks like you need to specify other devices to connect to specifically I think?

I'm using ESP-IDF v4.1 rc on Windows 64, with a SparkFun ESP32 Thing. After trying a number of examples, I've attempted to port the Arduino's library to ESP-IDF manually and by using a [lib-builder](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md).

Have I completely missed an ESP-IDF example/tutorial of how to do this? Or is it possible that a simple example could be added (simpler than `Gatt_Server_Service_Table_Example_Walkthrough.md` ideally)?

From what I've seen, most examples so far work out-of-the-box when you connect 2 or more ESP32 devices, but not for 1 ESP + Windows PC. The functionality I'm looking for is to be able to connect any PC to the ESP device as an "Outgoing" COM port and be able to send and receive data just like `uart_read_bytes` and `uart_send_bytes` works.

xiewenxiang, 2020-10-29:
Does this example [bt_spp_acceptor](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/bt_spp_acceptor) meet your needs?

thomasnugent, 2020-10-29:
Hi @xiewenxiang, from memory you have to run another ESP with spp_initiator for this example to work, is that correct? Or is it possible to run an ESP with [bt_spp_acceptor](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/bt_spp_acceptor) and connect to from a desktop by pairing (as you would BlueTooth speakers for example)?

xiewenxiang, 2020-10-30:
Yes, you can run ESP32 with `bt_spp_acceptor` and connect to a PC.

twlawrenceko, 2021-10-08:
> Yes, you can run ESP32 with `bt_spp_acceptor` and connect to a PC.

Hello, this is only for classic Bluetooth but not BLE?