espressif/esp-idf #5865: The wifi re-init when the esp32 start to OTA

NOA-Vernon, 2020-09-14:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): 
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    ESP-IDF v4.3-dev-472-gcf056a7d0-dirty
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
  xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [Yes (exlipse)]
- Power Supply:         [USB]

## Problem Description

the wifi re-init when the esp32 start to write the data to flash

### Steps to reproduce

1. download the esp-idf to your PC;
2. compile the example/system/ota/advanced_https_ota and flash to the esp32 module with 8KByte SPRAM and 8KByte Flash

Following is the log

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7036
load:0x40078000,len:13732
load:0x40080400,len:4000
entry 0x40080684
[0;32mI (29) boot: ESP-IDF v4.3-dev-472-gcf056a7d0-dirty 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 13:58:01[0m
[0;32mI (30) boot: chip revision: 1[0m
[0;32mI (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (50) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (50) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (50) boot.esp32: SPI Flash Size : 4MB[0m
[0;32mI (54) boot: Enabling RNG early entropy source...[0m
[0;32mI (60) boot: Partition Table:[0m
[0;32mI (63) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (71) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (78) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (86) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (93) boot:  3 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (101) boot:  4 ota_0            OTA app          00 10 00110000 00100000[0m
[0;32mI (108) boot:  5 ota_1            OTA app          00 11 00210000 00100000[0m
[0;32mI (116) boot: End of partition table[0m
[0;32mI (120) boot: Defaulting to factory image[0m
[0;32mI (125) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (132) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1f390 (127888) map[0m
[0;32mI (190) esp_image: segment 1: paddr=0x0002f3b8 vaddr=0x3ffb0000 size=0x00c60 (  3168) load[0m
[0;32mI (191) esp_image: segment 2: paddr=0x00030020 vaddr=0x400d0020 size=0x8f5bc (587196) map[0m
[0;32mI (420) esp_image: segment 3: paddr=0x000bf5e4 vaddr=0x3ffb0c60 size=0x02e40 ( 11840) load[0m
[0;32mI (425) esp_image: segment 4: paddr=0x000c242c vaddr=0x40080000 size=0x00404 (  1028) load[0m
[0;32mI (427) esp_image: segment 5: paddr=0x000c2838 vaddr=0x40080404 size=0x15d8c ( 89484) load[0m
[0;32mI (486) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (486) boot: Disabling RNG early entropy source...[0m
[0;32mI (498) cpu_start: Pro cpu up.[0m
[0;32mI (498) cpu_start: Starting app cpu, entry point is 0x40081424[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (514) cpu_start: Pro cpu start user code[0m
[0;32mI (514) cpu_start: Application information:[0m
[0;32mI (514) cpu_start: Project name:     advanced_https_ota[0m
[... 94 lines cut ...]
I (12882) wifi:state: init -> auth (b0)
I (13032) wifi:state: auth -> assoc (0)
I (13062) wifi:state: assoc -> run (10)
[0;31mE (13122) TRANS_SSL: ssl_poll_read select error 113, errno = Software caused connection abort, fd = 54[0m
I (13122) wifi:state: run -> init (cf00)
[0;31mE (13132) TRANS_SSL: ssl_poll_read select error 0, errno = Success, fd = 54[0m
I (13132) wifi:[0;31mE (13132) advanced_https_ota_example: Complete data was not received.[0m
new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,1>, prof:1
[0;32mI (13152) esp_image: segment 0: paddr=0x00110020 vaddr=0x3f400020 size=0x1f3f8 (127992) map[0m
[0;32mI (13152) example_connect: Wi-Fi disconnected, trying to reconnect...[0m
[0;31mE (13202) esp_image: invalid segment length 0xffffffff[0m
[0;31mE (13212) advanced_https_ota_example: Image validation failed, image is corrupted[0m
[0;31mE (13212) advanced_https_ota_example: ESP_HTTPS_OTA upgrade failed 5379[0m
I (13292) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (13292) wifi:state: init -> auth (b0)
I (13292) wifi:state: auth -> assoc (0)
I (13302) wifi:state: assoc -> run (10)
I (13312) wifi:connected with TP-LINK_3250, aid = 2, channel 1, 40U, bssid = 94:d9:b3:ee:32:50
I (13312) wifi:security: WPA2-PSK, phy: bgn, rssi: -45
I (13312) wifi:pm start, type: 0

I (13322) wifi:AP's beacon interval = 102400 us, DTIM period = 1
[0;32mI (14202) esp_netif_handlers: example_connect: sta ip: 192.168.1.100, mask: 255.255.255.0, gw: 192.168.1.1[0m
[0;32mI (14202) example_connect: Got IPv4 event: Interface "example_connect: sta" address: 192.168.1.100[0m
[0;32mI (14702) example_connect: Got IPv6 event: Interface "example_connect: sta" address: fe80:0000:0000:0000:ae67:b2ff:fe24:1824, type: ESP_IP6_ADDR_IS_LINK_LOCAL[0m

NOA-Vernon, 2020-09-14:
[config.zip](https://github.com/espressif/esp-idf/files/5217485/config.zip)
attached the config file for the project

NOA-Vernon, 2020-09-14:
[config.zip](https://github.com/espressif/esp-idf/files/5217510/config.zip)
attached the config file for the project

shubhamkulkarni97, 2020-09-14:
Hi @NOA-Vernon,

I checked running OTA example on specified ESP-IDF version and found no issues.
From the shared logs, I think there is some issue with wifi_lib submodule at your end.

The issue you are facing was fixed in https://github.com/espressif/esp-idf/commit/b9b82659dd8c3807c6ff28354f98fb721bf708c6.
https://github.com/espressif/esp32-wifi-lib/commit/19d4b32cb56373bcb341f54fee7ceaf0c6358f85 is the corresponding commit in wifi lib.
Please check if both the changes are included in your branch.

Running `git submodule update --init --recursive` should fix the issue.
If it does not fix your issue, try pulling latest branch from GitHub and run `git submodule update --init --recursive`.

Thanks,
Shubham

NOA-Vernon, 2020-09-15:
@shubhamkulkarni97  Thank you so much for the reply and your answer, yes, I have rolled back the wifi lib to the commit ID 511dd9e12 for the coexistence of BLE and wifi mesh, see this link [https://github.com/espressif/esp-mdf/issues/141].
Are you sure the newest wifi lib submodule can also fix the coexistence issue of BLE and wifi mesh?

NOA-Vernon, 2020-09-15:
@shubhamkulkarni97 I have updated to the latest branch and updated all the submodule, however, when the esp32 is going to update the firmware, it unexpedtedly reset, please see the following log. 

[0;32mI (49820) USER_OTA: Starting Advanced OTA example[0m
[0;32mI (53445) MAIN: System information, free heap: 4046884[0m
[0;32mI (53445) MAIN: free internal heap: 66536[0m
[0;32mI (53445) MAIN: uxTaskGetNumberOfTasks(): 21[0m
[0;32mI (53761) esp_https_ota: Starting OTA...[0m
[0;32mI (53764) esp_https_ota: Writing to partition subtype 16 at offset 0x290000[0m
[0;32mI (53765) USER_OTA: Running firmware version: b1edf47-dirty[0m
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x3b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7032
load:0x40078000,len:14736
ho 0 tail 12 room 4
load:0x40080400,len:4112
entry 0x40080678
[0;32mI (183) boot: ESP-IDF v4.3-dev-907-g6c17e3a64 2nd stage bootloader[0m
[0;32mI (183) boot: compile time 17:21:18[0m
[0;32mI (184) boot: chip revision: 1[0m
[0;32mI (189) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (196) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (200) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (205) boot.esp32: SPI Flash Size : 8MB[0m
[0;33mW (210) boot.esp32: PRO CPU has been reset by WDT.[0m
[0;33mW (215) boot.esp32: WDT reset info: PRO CPU PC=0x40094381[0m
[0;33mW (221) boot.esp32: WDT reset info: APP CPU PC=0x5dda020d[0m
[0;32mI (227) boot: Enabling RNG early entropy source...[0m
[0;32mI (233) boot: Partition Table:[0m
[0;32mI (237) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (244) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (251) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (259) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (267) boot:  3 factory          factory app      00 00 00010000 00280000[0m
[0;32mI (274) boot:  4 ota_0            OTA app          00 10 00290000 00280000[0m
[0;32mI (282) boot:  5 ota_1            OTA app          00 11 00510000 00280000[0m
[0;32mI (289) boot: End of partition table[0m
[0;32mI (294) boot: Defaulting to factory image[0m
[0;32mI (298) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (305) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x67888 (424072) map[0m
[0;32mI (498) esp_image: segment 1: paddr=0x000778b0 vaddr=0x3ffbdb60 size=0x04aec ( 19180) load[0m
[0;32mI (507) esp_image: segment 2: paddr=0x0007c3a4 vaddr=0x40080000 size=0x00404 (  1028) load[0m
[0;32mI (508) esp_image: segment 3: paddr=0x0007c7b0 vaddr=0x40080404 size=0x03868 ( 14440) load[0m
[0;32mI (523) esp_image: segment 4: paddr=0x00080020 vaddr=0x400d0020 size=0x15896c (1411436) map[0m
[0;32mI (1136) esp_image: segment 5: paddr=0x001d8994 vaddr=0x40083c6c size=0x17cf0 ( 97520) load[0m
[0;32mI (1199) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (1199) boot: Disabling RNG early entropy source...[0m
[0;32mI (1211) psram: This chip is ESP32-D0WD[0m
[0;32mI (1211) spiram: Found 64MBit SPI RAM device[0m
[0;32mI (1212) spiram: SPI RAM mode: flash 40m sram 40m[0m
[0;32mI (1217) spiram: PSRAM initialized, cache is in normal (1-core) mode.[0m
[0;32mI (1224) cpu_start: cpu freq: 240[0m
[0;32mI (1228) cpu_start: Pro cpu up.[0m
[0;32mI (1232) cpu_start: Single core mode[0m

saloid, 2020-09-16:
I am having same issue. It happens during `mbedtls_md()` full image check. Even interrupt watchdog timeout set for 10 seconds (maximum allowed) is not enough for it.

saloid, 2020-09-16:
Sorry, this was my fault. mbedtls works as it should, I've just accidentally passed too big size, which overlaps next memory region. So it works correctly in both cases - with SHA256 HW acceleration and without.

liuzfesp, 2020-09-21:
Hi @NOA-Vernon, have you solved your issue? I just can't reproduce the issue.

NOA-Vernon, 2020-09-21:
@liuzfesp not totally solved, I just trigger the OTA function in another task and it can run expectedly.

HarveyRong-Esp, 2020-11-09:
Hi @NOA-Vernon,
`I (12562) wifi:state: run -> init (cf00))`

cf00 should be the WiFi disconnection caused by tbtt error. Can you update the latest IDF version for testing?

NOA-Vernon, 2020-11-09:
@HarveyRong-Esp I have updated to the latest version of IDF, this issue didn't occur again. thank you so much.

HarveyRong-Esp, 2020-11-09:
Fix version link 34e6a17c09e7839f74e51b76e8a1afd238091d40

Alvin1Zhang, 2020-11-09:
Thanks for reporting and updates, feel free to reopen. Thanks.

closed 2020-11-09 by Alvin1Zhang